openrazer/openrazer #2378: Support for Razer Audio Mixer

ghostklart, 2025-01-06:
### Device Name

Razer Audio Mixer

### Product Number

RZ19-03860100

### Model Number

RZ19-0386

### VID:PID

1532:053e

### USB Details

```shell
Bus 005 Device 005: ID 1532:053e Razer USA, Ltd Razer Audio Mixer
Couldn't open device, some information will be missing
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass          239 Miscellaneous Device
  bDeviceSubClass         2 
  bDeviceProtocol         1 Interface Association
  bMaxPacketSize0        64
  idVendor           0x1532 Razer USA, Ltd
  idProduct          0x053e 
  bcdDevice            1.06
  iManufacturer           1 Razer
  iProduct                2 Razer Audio Mixer
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0277
    bNumInterfaces          7
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0x80
      (Bus Powered)
    MaxPower              500mA
    Interface Association:
      bLength                 8
      bDescriptorType        11
      bFirstInterface         0
      bInterfaceCount         6
      bFunctionClass          1 Audio
      bFunctionSubClass       0 
      bFunctionProtocol      32 
      iFunction               0 
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         1 Audio
      bInterfaceSubClass      1 Control Device
      bInterfaceProtocol     32 
      iInterface              0 
      AudioControl Interface Descriptor:
        bLength                 9
        bDescriptorType        36
        bDescriptorSubtype      1 (HEADER)
        bcdADC               2.00
[... 610 lines cut ...]
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x84  EP 4 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0010  1x 16 bytes
        bInterval               4
```

### Device Input Paths

```shell
lrwxrwxrwx 1 root root  10 Jan  6 19:41 usb-Razer_Razer_Audio_Mixer-event-if06 -> ../event14
```

### Synapse Screenshots

[image]
[image]
[image]
[image]
[image]

z3ntu, 2025-02-15:
Oh, another completely new device from Razer.

From a first glance I'd say this is likely out of scope for OpenRazer since it probably uses a completely custom protocol and doesn't have any overlap with existing functionality inside OpenRazer.

That said, it would be interesting to get USB packet captures from this device, could you provide that? https://github.com/openrazer/openrazer/wiki/Reverse-Engineering-USB-Protocol

CatOPanda, 2025-09-25:
hii, i need this device working on Linux to keep my work away from windows, im kinda new at Linux and well i guess i am the one of the 3 crazy people of the world that has the Razer audio mixer, so, is there any way that i can help making this device working on linux ?

DevNvll, 2026-09-13:
if anyone still interested, I managed to reverse engineer the protocol and built an app to control my mixer. it's not really well packaged for anyone else other than me to use but it might be useful if anyone wants to add support to openrazer.
https://github.com/DevNvll/razer-mixer-linux-driver