openrazer/openrazer #2381: Backlight off after sleep

cptn-cosmo, 2025-01-17:
### Affected Device / PID

Razer Huntsman V2, 1532:026C

### Operating System / Version

Archlinux

### Kernel Version

6.12.9-arch1-1 

### OpenRazer Version

3.8.0

### Support Checklist

- [x] My user was added to the plugdev group.
- [x] I have rebooted and the problem persists.
- [x] I looked at the [Troubleshooting](https://github.com/openrazer/openrazer/wiki/Troubleshooting) page and it did not solve the problem.

### Symptoms

- [x] Uninstalling OpenRazer fixes the problem
- [ ] Re-plugging the device fixes the problem
- [ ] It worked before a system/kernel update.
- [x] Only a problem if the daemon is running.
- [ ] Only a problem from power up or a reboot.
- [ ] I'm dual booting with Windows.
- [ ] Intermittent or erratic problem.
- [ ] My device is missing a feature.

### Please describe the issue

I have installed stealth caps on my keyboard and whenever i put the PC to sleep the backlight on the keyboard turns off (which is good). However, when i wake the PC from sleep the backlight does not turn back on, which means all my keys are black which makes logging back in unnecessarily difficult. I haven't found a setting anywhere where i could change this.

For reference i am using SDDM/plasma 6.

lah7, 2025-01-17:
Does this also happen when you lock the PC? 

If so, you likely have the `devices_off_on_screensaver` turned on in the config file in your home directory: `~/.config/openrazer/razer.conf`.  Try setting this to `False`:

```
[Startup]
devices_off_on_screensaver = False
```

_(Changes take effect on next login, or run `openrazer-daemon -r`)_

---

If that doesn't work, or was already `False`, we need to see the daemon and driver logs to understand what is happening before/after the sleep:
```
cat ~/.local/share/openrazer/logs/razer.log
```
```
sudo dmesg
```
Maybe also:
```
cat ~/.config/openrazer/persistence.conf | grep UNKWN
```

cptn-cosmo, 2025-01-21:
```
[Startup]
devices_off_on_screensaver = False

```

while technically this does "fix" the issue, i would generally want to keep the lights off during screen lock as well.

closed 2025-06-21 by github-actions[bot]