openrazer/openrazer #2382: Support for Razer Laptop Cooling Pad

MulesGaming, 2025-01-19:
### Device Name

Razer Laptop Cooling Pad

### Product Number

Not on label. Possibly RC21-02300

### Model Number

RC21-0230

### VID:PID

1532:0f43

### USB Details

```shell
Bus 001 Device 010: ID 1532:0f43 Razer USA, Ltd Razer Laptop Cooling Pad
Negotiated speed: Full Speed (12Mbps)
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 [unknown]
  bDeviceSubClass         0 [unknown]
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1532 Razer USA, Ltd
  idProduct          0x0f43 Razer Laptop Cooling Pad
  bcdDevice            2.00
  iManufacturer           1 Razer
  iProduct                2 Razer Laptop Cooling Pad
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x003b
    bNumInterfaces          2
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0x80
      (Bus Powered)
    MaxPower              500mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 [unknown]
      bInterfaceProtocol      0 
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 (null)
          wDescriptorLength      22
          Report Descriptors: 
            ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 [unknown]
      bInterfaceProtocol      0 
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 (null)
          wDescriptorLength     159
          Report Descriptors: 
            ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0010  1x 16 bytes
        bInterval               1
Device Status:     0x0000
  (Bus Powered)
```

### Device Input Paths

```shell
drwxr-xr-x 2 root root 160 Jan 19 16:40 .
drwxr-xr-x 4 root root 780 Jan 19 16:40 ..
lrwxrwxrwx 1 root root  10 Jan 19 14:54 usb-MOSART_Semi._2.4G_INPUT_DEVICE-event-if01 -> ../event11
lrwxrwxrwx 1 root root   9 Jan 19 13:55 usb-MOSART_Semi._2.4G_INPUT_DEVICE-event-kbd -> ../event4
lrwxrwxrwx 1 root root   9 Jan 19 13:55 usb-MOSART_Semi._2.4G_INPUT_DEVICE-if01-event-mouse -> ../event7
lrwxrwxrwx 1 root root   9 Jan 19 13:55 usb-MOSART_Semi._2.4G_INPUT_DEVICE-if01-mouse -> ../mouse0
lrwxrwxrwx 1 root root   9 Jan 19 16:40 usb-Razer_Razer_Laptop_Cooling_Pad-event-if01 -> ../event8
lrwxrwxrwx 1 root root   9 Jan 19 16:40 usb-Razer_Razer_Laptop_Cooling_Pad-if01-event-kbd -> ../event5
```

### USB HID Dump

```shell
001:010:001:DESCRIPTOR         1737326402.651125
 05 0C 09 01 A1 01 85 01 05 07 19 E0 29 E7 15 00
 25 01 75 01 95 08 81 02 19 00 2A FF 00 15 00 26
 FF 00 75 08 95 0E 81 00 C0 05 0C 09 01 A1 01 85
 02 19 00 2A 3C 02 15 00 26 3C 02 95 01 75 10 81
 00 75 08 95 0D 81 01 C0 05 01 09 80 A1 01 85 03
 19 81 29 83 15 00 25 01 75 01 95 03 81 02 95 05
 81 01 75 08 95 0E 81 01 C0 05 01 09 00 A1 01 85
 04 09 03 15 00 26 FF 00 35 00 46 FF 00 75 08 95
 0F 81 00 C0 05 01 09 00 A1 01 85 05 09 03 15 00
 26 FF 00 35 00 46 FF 00 75 08 95 0F 81 00 C0

001:010:000:DESCRIPTOR         1737326402.651456
 05 0C 09 01 A1 01 06 00 FF 09 02 15 00 25 01 75
 08 95 5A B1 01 C0
```

z3ntu, 2025-02-15:
Thanks for opening the issue! You can also take a look at https://github.com/openrazer/openrazer/blob/master/DEVELOPMENT.md how you could add support for this device yourself.

Also I'd be interested in what Synapse can actually control with this.

Benson5650, 2025-07-24:
> Thanks for opening the issue! You can also take a look at https://github.com/openrazer/openrazer/blob/master/DEVELOPMENT.md how you could add support for this device yourself.
> 
> Also I'd be interested in what Synapse can actually control with this.

I'm working on this. Synapse allows user to control the speed of fan or even using a fan curve.

SimonGallien, 2026-01-11:
Hi, I am on Linux Mint and I have the Razer Laptop Cooling Pad (ID 1532:0f43). Please let me know if I can be of any help with this issue.

ehog, 2026-08-28:
LED control update: the pad answers the **standard Chroma extended-matrix commands (class 0x0F)** — verified on hardware (firmware on a RC21-0230, macOS via IOKit, but this is protocol-level so it should apply to Linux directly).

Details for anyone implementing:

- Standard 90-byte Razer report, XOR checksum over bytes 2..87 at byte 88 (same framing the fan commands in @Benson5650's plugin use — the pad speaks the normal Razer wire protocol throughout)
- **Transaction id `0x02`, LED id `0x00` (ZERO_LED)** — the device returns status `0x02` (success) for these
- Working commands (openrazer's existing `razer_chroma_extended_matrix_effect_*` builders as-is, VARSTORE): none `0x00`, static `0x01`, breathing `0x02`, spectrum `0x03`, wave `0x04` (direction 1/2, speed 0x28), and brightness via class `0x0F` cmd `0x04`
- Fan quirk worth knowing: the pad has no tachometer — reading back the feature report just echoes the last response, so an RPM "read" is only meaningful right after a fan command

Working implementation (fan curve + all the lighting effects above) here if useful as a reference: https://github.com/ehog/razerpad-mac — see `razerpad.c` for the exact packet construction. Happy to run captures/tests on my unit if that helps an openrazer driver land.

christophehenry, 2026-08-29:
The LEDs are implemented by #2798 that, sadly hasn't been reviewed yet. I was waiting approval before moving on to continue the work started by #1227. Although I see someone opened #2815 for that too.