openrazer/openrazer #2383: Kernel issues across Polychromatic and Openrazer, + issue with Snake.

Fl1pSyde, 2025-01-21:
### Affected Device / PID

Razer Blackwidow Elite + Razer Basilisk V2

### Operating System / Version

Ubuntu 24.04.1 LTS

### Kernel Version

6.8.0-51-generic

### OpenRazer Version

3.9.0

### Support Checklist

- [x] My user was added to the plugdev group.
- [x] I have rebooted and the problem persists.
- [x] I looked at the [Troubleshooting](https://github.com/openrazer/openrazer/wiki/Troubleshooting) page and it did not solve the problem.

### Symptoms

- [ ] Uninstalling OpenRazer fixes the problem
- [ ] Re-plugging the device fixes the problem
- [ ] It worked before a system/kernel update.
- [ ] Only a problem if the daemon is running.
- [ ] Only a problem from power up or a reboot.
- [x] I'm dual booting with Windows.
- [ ] Intermittent or erratic problem.
- [ ] My device is missing a feature.

### Please describe the issue

Hey howdy hey Razerites. Upon trying to load into Polychromatic, I am greeted with a screen saying that, while both of my devices have a PID + friendly name, "This supported device wasn't detected by the OpenRazer daemon." Okay. I run the troubleshooter. Troubleshooter says that I don't have any of the kernel modules installed either. It doesn't help that I can't find any links to directly download any of the kernels.

So. I try Snake from the suggested software 'tab' of the [openrazer site](https://openrazer.github.io/#download). It proceeds to tell me that "The name org.ayatana.bmf was not provided by any .service files." 

Did I download something incorrectly? I've scoured the forums but I can't find an issue similar to mine. Help!!

lah7, 2025-01-21:
Sounds like a partial installation. Or perhaps the system updated recently and missed out the kernel modules.

**The way it works is:** The driver is provided as a DKMS module, which should be automatically built when your system installs or updates the Linux kernel. It's like compiling code to provide OpenRazer functionality for that version of the kernel.

The kernel module for OpenRazer is provided in the `openrazer-driver-dkms` package. Installing OpenRazer [with the Ubuntu instructions](https://openrazer.github.io/#ubuntu) should have installed this. If not:

    sudo apt install openrazer-driver-dkms

The troubleshooting guide does cover this:
https://github.com/openrazer/openrazer/wiki/Troubleshooting#kernel-modules-are-they-built-properly

Is the driver installed? Check the status from a terminal:

    sudo dkms status

You should see input like this:

    openrazer-driver/3.9.0, 6.12.10-arch1-1, x86_64: installed

If not, run this and reboot:

    sudo dkms install openrazer-driver/3.9.0

Fl1pSyde, 2025-01-22:
Unfortunately, the installation and status check were good (returned that all was well) but it still does not show my devices on Snake or Polychromatic.

lah7, 2025-01-22:
Please could you show a screenshot of what Polychromatic's troubleshooter said? 

If the modules are installed, [are they able to load?](https://github.com/openrazer/openrazer/wiki/Troubleshooting#kernel-modules-do-they-load-properly)

    sudo modprobe razerkbd
    sudo modprobe razermouse

Fl1pSyde, 2025-01-22:
[image]
[image]

`modprobe: ERROR: could not insert 'razerkbd': Key was rejected by service`

I have disabled secure boot in my HP laptop's UEFI/BIOS options. I uninstalled and reinstalled according to the linked troubleshooter. I don't know where I could find kernel lockdown, would that be in my computer's BIOS or somewhere in ubuntu?

Also.

[image]

Fl1pSyde, 2025-01-22:
Thought you may want to see this as well.

`Building module:
Cleaning build area...
KERNELDIR=/lib/modules/6.8.0-51-generic/build make driver.......
Signing module /var/lib/dkms/openrazer-driver/3.9.0/build/driver/razerkbd.ko
Signing module /var/lib/dkms/openrazer-driver/3.9.0/build/driver/razermouse.ko
Signing module /var/lib/dkms/openrazer-driver/3.9.0/build/driver/razerkraken.ko
Signing module /var/lib/dkms/openrazer-driver/3.9.0/build/driver/razeraccessory.ko
Cleaning build area...`

lah7, 2025-01-22:
Thanks for the outputs. Things should just start working if secure boot is disabled, is it definitely off? Check with:

    sudo apt install mokutil
    sudo mokutil --sb-state

From what I know, [kernel lockdown](https://www.man7.org/linux/man-pages/man7/kernel_lockdown.7.html) is only enabled on a standard Ubuntu installation when EFI secure boot is active.

Fl1pSyde, 2025-01-23:
Well, looks like my machine reinstated secure boot after turning it off.  I ensured that secure boot wouldn't auto-start. Thanks for the insight! I really appreciate the patience 😆

closed 2025-01-23 by Fl1pSyde