openrazer/openrazer #2386: [MISC] New openrazer example script

Urufusan, 2025-01-25:
Hi! I've recently started playing around with my Razer Mouse Dock Chroma. After some tinkering, I've recreated a feature present in Razer Synapse, the ability for the dock to reflect the current battery status of the mouse using its LEDs.

[You can find my project here](https://github.com/Urufusan/openrazer-dock-sync), accepting comments and critique!

Posting this as an issue so that I can get proper feedback on what I should do next (perhaps turn my POC into a PR for openrazer or polychromatic).

lah7, 2025-02-01:
Nice to see the Python library in action! 

**From a quick look:**

* Instead of looking for device name, you could look for `device.type == "mouse"`? 
   * _Saves the user needing to modify the script._
   * _Assuming there's only one `mouse` device type connected. Not sure what the mouse dock type is._
* Haven't seen the ["walrus operator" (PEP 0572)](https://peps.python.org/pep-0572/) till now. Personally, I don't like that syntax, to me it makes the code [less easy to read](https://peps.python.org/pep-0020/). For example, not 100% sure [what this part is doing.](https://github.com/Urufusan/openrazer-dock-sync/blob/73c3fbc28f541d2079e3d57a81dc93f0f85312bb/razrdocksync.py#L126-L129)
* Not sure about hardcoding the [retry counters.](https://github.com/Urufusan/openrazer-dock-sync/blob/73c3fbc28f541d2079e3d57a81dc93f0f85312bb/razrdocksync.py#L121)

**Regarding next steps:**

* Polychromatic isn't ready for scripted/software effects, it's too early to PR there.
* You can definitely polish it up and contribute it to OpenRazer's [examples folder](https://github.com/openrazer/openrazer/tree/master/examples). That's the best place right now.
* If you do contribute your addition:
   * This repo license is [GPL-2.0-or-later,](https://github.com/openrazer/openrazer/blob/master/LICENSES/GPL-2.0-or-later.txt) not AGPL-3.0-or-later. 
   * Remove the debug `IPython` parts. 
   * `configparser` might be overkill, since it's only one environment variable, which is simpler. Perhaps add a comment at the top of the file. Some desktop environments can include env variables in the launch command.
   * Less dependencies is better (i.e. don't add `click`, it wouldn't be an out-of-the-box experience since it's not installed on everyone's system)

Ideally, if an example is added, it won't need constant updates but stands the test of time. If you need flexibility, it would be worth keeping your project in a separate repository (add an `openrazer` tag so others can discover it, maybe?)

Urufusan, 2025-02-01:
> Nice to see the Python library in action!

Thanks!

> **From a quick look:**
> 
>     * Instead of looking for device name, you could look for `device.type == "mouse"`?
>       
>       * _Saves the user needing to modify the script._
>       * _Assuming there's only one `mouse` device type connected. Not sure what the mouse dock type is._

I think if I just hardcoded all supported mice for my specific dock, it would work better IMO

[image]

> * Haven't seen the ["walrus operator" (PEP 0572)](https://peps.python.org/pep-0572/) till now. Personally, I don't like that syntax, to me it makes the code [less easy to [...]

Oh, that's just what I used to set the color of the dock using a text prompt

ie ``>> 255 0 0`` sets the color to ``#ff0000`` (full red); the walrus operator is there so that I can easily check if the input was empty (to exit the loop).

>     * Not sure about hardcoding the [retry counters.](https://github.com/Urufusan/openrazer-dock-sync/blob/73c3fbc28f541d2079e3d57a81dc93f0f85312bb/razrdocksync.py#L121)

That was me just being lazy

>     * If you do contribute your addition:
>       
>       * This repo license is [GPL-2.0-or-later,](https://github.com/openrazer/openrazer/blob/master/LICENSES/GPL-2.0-or-later.txt) not AGPL-3.0-or-later.

ofc!

Thanks for responding!

Urufusan, 2025-02-01:
# UPDATE

Cleaned up the script according to what you've said, I think it should be fine as an example script

https://github.com/Urufusan/openrazer-dock-sync/commit/30a0d39bad2c2ce891450e4e4cf3bdac621729df

Will re-license ofc

Urufusan, 2025-02-04:
Added PR

closed 2025-02-15 by z3ntu