openrazer/openrazer #2388: Razer Viper V2 Pro; Middle click event missing

hackcr1258, 2025-02-01:
### Affected Device / PID

Razer Viper V2 Pro

### Operating System / Version

"Arch Linux" 6.12.10

### Kernel Version

6.12.10.arch1-1

### OpenRazer Version

3.9.0-2

### Support Checklist

- [x] My user was added to the plugdev group.
- [x] I have rebooted and the problem persists.
- [x] I looked at the [Troubleshooting](https://github.com/openrazer/openrazer/wiki/Troubleshooting) page and it did not solve the problem.

### Symptoms

- [ ] Uninstalling OpenRazer fixes the problem
- [ ] Re-plugging the device fixes the problem
- [ ] It worked before a system/kernel update.
- [ ] Only a problem if the daemon is running.
- [ ] Only a problem from power up or a reboot.
- [x] I'm dual booting with Windows.
- [ ] Intermittent or erratic problem.
- [x] My device is missing a feature.

### Please describe the issue

sudo libinput debug | grep button provides no output whatsoever. 

Additional information:

DE: KDE Plasma

The button works on my Windows, but not on Linux.

z3ntu, 2025-02-15:
Does the same problem also occur when OpenRazer is not installed?

hackcr1258, 2025-02-16:
> Does the same problem also occur when OpenRazer is not installed?

Just double checked, it does not work even without the drivers

z3ntu, 2025-02-16:
Then I'm going to say it's either a hardware problem or some other thing that we cannot fix in OpenRazer. Sorry.

closed 2025-02-16 by z3ntu