openrazer/openrazer #2390: Disable Volume Wheel on BlackWidow Elite

SoulRaven, 2025-02-01:
### Affected Device / PID

BlackWidow Elite, 1532:0228

### Operating System / Version

PopOS!

### Kernel Version

6.9.3-76060903-generic

### OpenRazer Version

3.9.0

### Support Checklist

- [x] My user was added to the plugdev group.
- [x] I have rebooted and the problem persists.
- [x] I looked at the [Troubleshooting](https://github.com/openrazer/openrazer/wiki/Troubleshooting) page and it did not solve the problem.

### Symptoms

- [ ] Uninstalling OpenRazer fixes the problem
- [ ] Re-plugging the device fixes the problem
- [ ] It worked before a system/kernel update.
- [ ] Only a problem if the daemon is running.
- [ ] Only a problem from power up or a reboot.
- [ ] I'm dual booting with Windows.
- [ ] Intermittent or erratic problem.
- [x] My device is missing a feature.

### Please describe the issue

hello,
somebody has any fix to disable the volume knob using openrazer driver?

z3ntu, 2025-02-15:
Why do you want to disable the volume wheel on the keyboard?

SoulRaven, 2025-02-15:
Because is not working and all the time I get false positive on moving the volume

z3ntu, 2025-02-15:
Maybe try input-remapper and remap to events to nothing so they get ignored?

SoulRaven, 2025-02-15:
Is not working. The events of volume up and down are not visible inside the mapper

z3ntu, 2025-02-15:
But the events do show up with e.g. `sudo libinput debug-events`?

In any case, having the option to disable/ignore various pieces of the hardware is not in scope for OpenRazer, so using some layer between the kernel input subsystem and your UI is probably your best shot - like input-remapper.

SoulRaven, 2025-02-15:
with libinput i get:

> -event4   KEYBOARD_KEY                +3.215s	KEY_VOLUMEUP (115) pressed
>  event4   KEYBOARD_KEY                +3.215s	KEY_VOLUMEUP (115) released
>  event4   KEYBOARD_KEY                +7.780s	KEY_VOLUMEDOWN (114) pressed
>  event4   KEYBOARD_KEY                +7.780s	KEY_VOLUMEDOWN (114) released
>  event4   KEYBOARD_KEY                +7.792s	KEY_VOLUMEUP (115) pressed
>  event4   KEYBOARD_KEY                +7.792s	KEY_VOLUMEUP (115) released
>  event4   KEYBOARD_KEY                +7.854s	KEY_VOLUMEDOWN (114) pressed
>  event4   KEYBOARD_KEY                +7.854s	KEY_VOLUMEDOWN (114) released
>  event4   KEYBOARD_KEY                +7.942s	KEY_VOLUMEUP (115) pressed
>  event4   KEYBOARD_KEY                +7.942s	KEY_VOLUMEUP (115) released
>  event4   KEYBOARD_KEY                +8.772s	KEY_VOLUMEUP (115) pressed
>  event4   KEYBOARD_KEY                +8.772s	KEY_VOLUMEUP (115) released

but input-remaper still not detecting events, like volum up and down