openrazer/openrazer #2392: Downloading successful, but checksum doesn't match.

TheFailureOne, 2025-02-09:
### Affected Device / PID

Razer Viper Mini

### Operating System / Version

Nobara 41

### Kernel Version

6.12.11-204.nobara.fc41.x86_64

### OpenRazer Version

3.9.0

### Support Checklist

- [x] My user was added to the plugdev group.
- [x] I have rebooted and the problem persists.
- [x] I looked at the [Troubleshooting](https://github.com/openrazer/openrazer/wiki/Troubleshooting) page and it did not solve the problem.

### Symptoms

- [ ] Uninstalling OpenRazer fixes the problem
- [ ] Re-plugging the device fixes the problem
- [ ] It worked before a system/kernel update.
- [ ] Only a problem if the daemon is running.
- [ ] Only a problem from power up or a reboot.
- [x] I'm dual booting with Windows.
- [ ] Intermittent or erratic problem.
- [ ] My device is missing a feature.

### Please describe the issue

So basically, I tried installing openrazer from the package manager, as shown in the openrazer website, I followed all the steps for Fedora 41 (which Nobara is based on) and, upon running `sudo dnf install openrazer-meta` the following gets spat out:
```shell
Updating and loading repositories:
Repositories loaded.
Package                            Arch     Version                             Repository                Size
Installing:
 openrazer-meta                    noarch   3.9.0-2.2                           nobara-updates         0.0   B
Installing dependencies:
 openrazer-daemon                  noarch   3.9.0-2.2                           nobara-updates       976.7 KiB
 openrazer-kernel-modules-dkms     noarch   3.9.0-2.2                           nobara-updates       768.7 KiB
 python3-daemonize                 noarch   2.5.0-21.fc41                       nobara                25.0 KiB
 python3-openrazer                 noarch   3.9.0-2.2                           nobara-updates       262.3 KiB
 xautomation                       x86_64   1.09-11.fc41                        nobara               183.5 KiB

Transaction Summary:
 Installing:         6 packages

Total size of inbound packages is 398 KiB. Need to download 318 KiB.
After this operation, 2 MiB extra will be used (install 2 MiB, remove 0 B).
Is this ok [y/N]: y
[1/3] python3-daemonize-0:2.5.0-21.fc41.noarch                        100% |   0.0   B/s |   0.0   B |  00m00s
>>> Already downloaded
[2/4] xautomation-0:1.09-11.fc41.x86_64                               100% |   0.0   B/s |   0.0   B |  00m00s
>>> Already downloaded                             0% [<=>               ] |   0.0   B/s |  -2.0   B | -00m00s
[3/6] openrazer-kernel-modules-dkms-0:3.9.0-2.2.noarch                100% |  37.4 KiB/s |  63.9 KiB |  00m02s
>>> Downloading successful, but checksum doesn't match. Calculated: f38b194b5ec03c2cde4add5d6be9cad16f94127e79
>>> Downloading successful, but checksum doesn't match. Calculated: f38b194b5ec03c2cde4add5d6be9cad16f94127e79
>>> No more mirrors to try - All mirrors were already tried without success|  63.4 KiB/s | 164.1 KiB | -00m00s
[4/6] openrazer-daemon-0:3.9.0-2.2.noarch                              87% |  96.1 KiB/s | 164.1 KiB |  00m02s
>>> Not finished - interrupted by error: Cannot download 08538024-openrazer-meta/openrazer-kernel-modules-dkms
[5/6] openrazer-meta-0:3.9.0-2.2.noarch                                 0% |   0.0   B/s |   0.0   B |  00m02s
>>> Downloading successful, but checksum doesn't match. Calculated: d6dc21d9c771216c73e7149fc577d660c76520b982
>>> Downloading successful, but checksum doesn't match. Calculated: d6dc21d9c771216c73e7149fc577d660c76520b982
>>> Not finished - interrupted by error: Cannot download 08538024-openrazer-meta/openrazer-kernel-modules-dkms
[6/6] python3-openrazer-0:3.9.0-2.2.noarch                              0% |   0.0   B/s |  -1.0   B |  ?     
>>> Interrupted                                   79% [==============    ] | 112.6 KiB/s | 228.0 KiB | -00m00s
--------------------------------------------------------------------------------------------------------------
[6/6] Total                                                           100% |  86.6 KiB/s | 228.0 KiB |  00m03s
Failed to download packages
 Librepo error: Cannot download 08538024-openrazer-meta/openrazer-kernel-modules-dkms-3.9.0-2.2.noarch.rpm: All mirrors were tried

``` 
as you can see, the checksum doesn't match, and thus it refuses to install. Do we know what is up with that?

z3ntu, 2025-02-15:
Is this still an issue today? I can try re-trigger the package build on build.opensuse.org where the packages are getting built.

TheFailureOne, 2025-02-23:
Turns out this was an issue on my side, shortly after I noticed that I was getting the same error on most packages I tried installing.
Fixed it with `dnf clean all`

closed 2025-02-23 by TheFailureOne

z3ntu, 2025-02-24:
Good to hear it's working for you now!