openrazer/openrazer #2395: Support for Razer Atrox Arcade Stick Xbox One

SpoodySTK, 2025-02-13:
### Device Name

Razer Atrox

### Product Number

RZ06-01150100

### Model Number

RZ06-0115

### VID:PID

1532:0a00

### USB Details

```shell
Bus 001 Device 017: ID 1532:0a00 Razer USA, Ltd Atrox Arcade Stick for Xbox One
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass          255 Vendor Specific Class
  bDeviceSubClass       255 Vendor Specific Subclass
  bDeviceProtocol       255 Vendor Specific Protocol
  bMaxPacketSize0        64
  idVendor           0x1532 Razer USA, Ltd
  idProduct          0x0a00 Atrox Arcade Stick for Xbox One
  bcdDevice            1.00
  iManufacturer           1 Razer
  iProduct                2 Razer Atrox Arcade Stick
  iSerial                 3 0000CF133A02C964
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0020
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              200mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass       255 Vendor Specific Class
      bInterfaceSubClass     71 [unknown]
      bInterfaceProtocol    208 
      iInterface              0 
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x01  EP 1 OUT
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               4
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               4
Device Status:     0x0002
  (Bus Powered)
  Remote Wakeup Enabled
```

### Device Input Paths

```shell
total 0
drwxr-xr-x 2 root root 100 Feb 13 00:15 .
drwxr-xr-x 4 root root 500 Feb 13 00:15 ..
lrwxrwxrwx 1 root root   9 Feb 13 00:00 usb-Valve_Software_Steam_Controller_123456789ABCDEF-event-mouse -> ../event3
lrwxrwxrwx 1 root root   9 Feb 13 00:00 usb-Valve_Software_Steam_Controller_123456789ABCDEF-if01-event-kbd -> ../event7
lrwxrwxrwx 1 root root   9 Feb 13 00:00 usb-Valve_Software_Steam_Controller_123456789ABCDEF-mouse -> ../mouse0
```

### USB HID Dump

```shell
No matching HID interfaces
```

z3ntu, 2025-02-15:
Is there any lighting or other settings on this device that can be controlled with Razer Synapse on Windows?

SpoodySTK, 2025-02-16:
> Is there any lighting or other settings on this device that can be controlled with Razer Synapse on Windows?

I've check and synapse wont detect the Razer Atrox XB ONE. The problem that I'm dealing that the RT and LT button doesn't work despite detecting it, it is also an issue with windows that in order to work, you need steam and enable Steam Input for xbox controllers. Its a driver issue for this controller. A regular xbox controllers works fine.

z3ntu, 2025-02-16:
This is out of scope for OpenRazer then. We generally only deal with RGB lighting and things like that, not with input for controllers or other devices which use a different way of communicating in the firmware.

closed 2025-02-16 by z3ntu