openrazer/openrazer #2397: FN keys not working with Razer DeathStalker V2 Pro TKL

benoitjpnet, 2025-02-16:
### Affected Device / PID

Razer DeathStalker V2 Pro TKL, 1532:0296

### Operating System / Version

Arch

### Kernel Version

6.13.2-arch1-1

### OpenRazer Version

3.9.0

### Support Checklist

- [x] My user was added to the plugdev group.
- [x] I have rebooted and the problem persists.
- [x] I looked at the [Troubleshooting](https://github.com/openrazer/openrazer/wiki/Troubleshooting) page and it did not solve the problem.

### Symptoms

- [ ] Uninstalling OpenRazer fixes the problem
- [ ] Re-plugging the device fixes the problem
- [ ] It worked before a system/kernel update.
- [ ] Only a problem if the daemon is running.
- [ ] Only a problem from power up or a reboot.
- [ ] I'm dual booting with Windows.
- [ ] Intermittent or erratic problem.
- [ ] My device is missing a feature.

### Please describe the issue

FN + INS = Print Screen, don't work
FN + PAGE DOWN = Sleep, don't work

When running `wev`, the fin key press is not detected.

z3ntu, 2025-02-16:
Do you see the same behavior still after stopping the OpenRazer daemon? `systemctl --user stop openrazer-daemon.service`

benoitjpnet, 2025-02-16:
Yep. I saw other bug reports and stopped it, but same behavior.

z3ntu, 2025-02-18:
Then it's not a matter of "device mode vs driver mode" from the keyboard. Do the keys work as expected without OpenRazer installed at all?

In other words, did this ever work on Linux?

cometkim, 2025-02-26:
Same here.

> In other words, did this ever work on Linux?

This was the same when I used my keyboard via a HyperSpeed ​​multi-device connection and the keyboard profile was not properly detected. So I think it doesn't work even w/o OpenRazer.

But it does work on Windows w/ Synapse

z3ntu, 2025-02-26:
Then I think our best shot would be following something like I've written in https://github.com/openrazer/openrazer/issues/2228#issuecomment-2037981701

I should probably start making a wiki page for this procedure at some point, it comes in useful more often than I'd have thought.

closed 2025-07-27 by github-actions[bot]