openrazer/openrazer #2403: Persisted config is reset constantly with USB reconnections

quietvoid, 2025-02-27:
### Affected Device / PID

Razer Basilisk V3, 0003:1532:0099.0031

### Operating System / Version

Arch Linux

### Kernel Version

6.13.4-arch1-1

### OpenRazer Version

3.10.0

### Support Checklist

- [x] My user was added to the plugdev group.
- [x] I have rebooted and the problem persists.
- [x] I looked at the [Troubleshooting](https://github.com/openrazer/openrazer/wiki/Troubleshooting) page and it did not solve the problem.

### Symptoms

- [ ] Uninstalling OpenRazer fixes the problem
- [ ] Re-plugging the device fixes the problem
- [ ] It worked before a system/kernel update.
- [x] Only a problem if the daemon is running.
- [ ] Only a problem from power up or a reboot.
- [ ] I'm dual booting with Windows.
- [ ] Intermittent or erratic problem.
- [ ] My device is missing a feature.

### Please describe the issue

I'm having issues with the latest version, the persistence config keeps adding generated serial numbers.
I'm my setup I use a USB hub and switcher, making the mouse reconnect multiple times over a day.

When the generated serial number is new (incremented), the settings are completely reset.
If I do delete `persistence.conf`, then the settings persist until it's been X number of times disconnected again, at which point the settings are wrong again.

The logs are always like this:
```
razer                          | INFO     | Found valid device.0: 0003:1532:0099.0035
razer.device0                  | INFO     | Initialising device.0 RazerBasiliskV3
razer.device0                  | WARNING  | Invalid serial number found, using a generated one.
razer.device0                  | WARNING  | Original value: 632141H21304575
```

The `persistence.conf` file is filled with incrementing `[UNKNOWN_15320099_0001]` configs.

quietvoid, 2025-02-27:
I tried downgrading to version `3.9.0` and the serial number is never detected as unknown.
So something must have caused a regression in whatever checks the validity of the serial number.

quietvoid, 2025-02-27:
Yes, it works. There is no more warning.

z3ntu, 2025-02-28:
Thanks for reporting!

closed 2025-02-28 by z3ntu

zquestz, 2025-03-06:
I am so glad I saw this. For now I can just the latest git build, but please put out a release with this fix ASAP. =)

commit a09f59b5 (2025-02-28): daemon: Relax serial number check a bit

While many serial numbers have two uppercase letters and then a mix of
numbers and letters, there seems to be also valid serial numbers which
start with a number. Relax the check to allow any mix of numbers and
uppercase letters. Let's see if that's good enough or we need to tweak
it further.

Closes: #2403
Fixes: a32dc5ff ("daemon: Be more strict in accepting serial numbers")