openrazer/openrazer #2405: Support for Razer Leviathan 2 X - Volume 50% max when Wired, Wireless gets full output

ff7freddy, 2025-03-03:
### Device Name

Razer Leviathan V2 X

### Product Number

RZ05-04280100

### Model Number

RZ02-0428

### VID:PID

1532:054a

### USB Details

```shell
lsusb -d 1532: -v

Bus 005 Device 037: ID 1532:054a Razer USA, Ltd Razer Leviathan V2 X
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               1.10
  bDeviceClass            0 [unknown]
  bDeviceSubClass         0 [unknown]
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1532 Razer USA, Ltd
  idProduct          0x054a Razer Leviathan V2 X
  bcdDevice            0.00
  iManufacturer           1 Razer
  iProduct                2 Razer Leviathan V2 X
  iSerial                 3 000000000000000
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x008f
    bNumInterfaces          3
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0x80
      (Bus Powered)
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 [unknown]
      bInterfaceProtocol      0 
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.10
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      74
          Report Descriptors: 
            ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               9
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x01  EP 1 OUT
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               9
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           0
      bInterfaceClass         1 Audio
      bInterfaceSubClass      1 Control Device
      bInterfaceProtocol      0 
      iInterface              0 
      AudioControl Interface Descriptor:
        bLength                 9
        bDescriptorType        36
        bDescriptorSubtype      1 (HEADER)
        bcdADC               1.00
        wTotalLength       0x0028
        bInCollection           1
        baInterfaceNr(0)        2
      AudioControl Interface Descriptor:
        bLength                12
        bDescriptorType        36
        bDescriptorSubtype      2 (INPUT_TERMINAL)
        bTerminalID             1
        wTerminalType      0x0101 USB Streaming
        bAssocTerminal          0
        bNrChannels             2
        wChannelConfig     0x0003
          Left Front (L)
          Right Front (R)
        iChannelNames           0 
        iTerminal               0 
      AudioControl Interface Descriptor:
        bLength                 9
        bDescriptorType        36
        bDescriptorSubtype      3 (OUTPUT_TERMINAL)
        bTerminalID             3
        wTerminalType      0x0301 Speaker
        bAssocTerminal          0
        bSourceID               9
        iTerminal               0 
      AudioControl Interface Descriptor:
        bLength                10
        bDescriptorType        36
        bDescriptorSubtype      6 (FEATURE_UNIT)
        bUnitID                 9
        bSourceID               1
        bControlSize            1
        bmaControls(0)       0x03
          Mute Control
          Volume Control
        bmaControls(1)       0x02
          Volume Control
        bmaControls(2)       0x02
          Volume Control
        iFeature                0 
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        2
      bAlternateSetting       0
      bNumEndpoints           0
      bInterfaceClass         1 Audio
      bInterfaceSubClass      2 Streaming
      bInterfaceProtocol      0 
      iInterface              0 
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        2
      bAlternateSetting       1
      bNumEndpoints           1
      bInterfaceClass         1 Audio
      bInterfaceSubClass      2 Streaming
      bInterfaceProtocol      0 
      iInterface              0 
      AudioStreaming Interface Descriptor:
        bLength                 7
        bDescriptorType        36
        bDescriptorSubtype      1 (AS_GENERAL)
        bTerminalLink           1
        bDelay                  1 frames
        wFormatTag         0x0001 PCM
      AudioStreaming Interface Descriptor:
        bLength                14
        bDescriptorType        36
        bDescriptorSubtype      2 (FORMAT_TYPE)
        bFormatType             1 (FORMAT_TYPE_I)
        bNrChannels             2
        bSubframeSize           2
        bBitResolution         16
        bSamFreqType            1 Discrete
        tSamFreq[ 0]        48000
        junk at descriptor end: 00 00 00
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x02  EP 2 OUT
        bmAttributes            9
          Transfer Type            Isochronous
          Synch Type               Adaptive
          Usage Type               Data
        wMaxPacketSize     0x00c0  1x 192 bytes
        bInterval               1
        AudioStreaming Endpoint Descriptor:
          bLength                 7
          bDescriptorType        37
          bDescriptorSubtype      1 (EP_GENERAL)
          bmAttributes         0x01
            Sampling Frequency
          bLockDelayUnits         1 Milliseconds
          wLockDelay         0x0001
Device Status:     0x0000
  (Bus Powered)
```

### Device Input Paths

```shell
ls -lah /dev/input/by-id/
total 0
drwxr-xr-x. 2 root root  340 Mar  3 15:03 .
drwxr-xr-x. 4 root root 1.2K Mar  3 15:05 ..
lrwxrwxrwx. 1 root root   10 Mar  3 13:04 usb-BY_Tech_Gaming_Keyboard-event-if01 -> ../event13
lrwxrwxrwx. 1 root root   10 Mar  3 13:03 usb-BY_Tech_Gaming_Keyboard-event-kbd -> ../event10
lrwxrwxrwx. 1 root root   10 Mar  3 13:03 usb-BY_Tech_Gaming_Keyboard-if01-event-mouse -> ../event14
lrwxrwxrwx. 1 root root    9 Mar  3 13:03 usb-BY_Tech_Gaming_Keyboard-if01-mouse -> ../mouse4
lrwxrwxrwx. 1 root root   10 Mar  3 13:04 usb-Generic_LG_Monitor_Headset_MIC-event-if00 -> ../event16
lrwxrwxrwx. 1 root root   10 Mar  3 13:03 usb-Logitech_USB_Receiver_3957386D3232-event-mouse -> ../event18
lrwxrwxrwx. 1 root root   10 Mar  3 13:03 usb-Logitech_USB_Receiver_3957386D3232-if01-event-kbd -> ../event19
lrwxrwxrwx. 1 root root    9 Mar  3 13:03 usb-Logitech_USB_Receiver_3957386D3232-mouse -> ../mouse5
lrwxrwxrwx. 1 root root    9 Mar  3 13:03 usb-_Mouse_for_Windows-event-mouse -> ../event3
lrwxrwxrwx. 1 root root    9 Mar  3 13:03 usb-_Mouse_for_Windows-if01-event-kbd -> ../event4
lrwxrwxrwx. 1 root root    9 Mar  3 13:03 usb-_Mouse_for_Windows-mouse -> ../mouse0
lrwxrwxrwx. 1 root root   11 Mar  3 15:03 usb-Razer_Razer_Leviathan_V2_X_000000000000000-event-if00 -> ../event263
lrwxrwxrwx. 1 root root   10 Mar  3 14:19 usb-Sony_PLAYSTATION_R_3_Controller-event-if00 -> ../event17
lrwxrwxrwx. 1 root root   11 Mar  3 14:19 usb-Sony_PLAYSTATION_R_3_Controller-event-joystick -> ../event261
lrwxrwxrwx. 1 root root    6 Mar  3 14:19 usb-Sony_PLAYSTATION_R_3_Controller-joystick -> ../js0
```

### USB HID Dump

```shell
usbhid-dump -m 1532 -ed
005:037:000:DESCRIPTOR         1741043184.782759
 05 0C 09 01 A1 01 85 01 15 00 25 01 09 E9 09 EA
 09 E2 09 CD 09 B5 09 B6 09 B3 09 B7 75 01 95 08
 81 42 85 03 26 FF 00 09 00 75 08 95 5B 91 02 85
 05 26 FF 00 09 00 75 08 95 0F 81 02 85 07 26 FF
 00 09 00 75 08 95 5A B1 01 C0
```

z3ntu, 2025-03-15:
First of all, this is a duplicate of #1978 just for device support.

But for the issue written in the title, this is unlikely to change with OpenRazer support since we don't touch the audio parts.

closed 2025-03-15 by z3ntu

aloerch, 2025-12-27:
FYI, this is almost certainly caused by alsamixer. If you run alsamixer from terminal, switch to the Leviathan, you'll see 2 PCM devices. In my case, with OpenSuse Tumbleweed and KDE Plasma 6, with the volume slider for the device in the GUI set to 100%, the speaker was still barely audible. The solution was to use alsamixer to raise both PCM levels for the Leviathan V2 X from 19 to 100. Then, setting the volume with the GUI slider and on the bar itself worked just fine and the audio levels are good.