openrazer/openrazer #2407: Razer Basilisk V3 Pro 35K is not recognized throug bluetooth

emouty, 2025-03-05:
### Affected Device / PID

068E:00CE

### Operating System / Version

Fedora 41

### Kernel Version

6.13.5-200.fc41.x86_64

### OpenRazer Version

3.10.0

### Support Checklist

- [x] My user was added to the plugdev group.
- [x] I have rebooted and the problem persists.
- [x] I looked at the [Troubleshooting](https://github.com/openrazer/openrazer/wiki/Troubleshooting) page and it did not solve the problem.

### Symptoms

- [ ] Uninstalling OpenRazer fixes the problem
- [ ] Re-plugging the device fixes the problem
- [ ] It worked before a system/kernel update.
- [ ] Only a problem if the daemon is running.
- [ ] Only a problem from power up or a reboot.
- [ ] I'm dual booting with Windows.
- [ ] Intermittent or erratic problem.
- [ ] My device is missing a feature.

### Please describe the issue

My Razer Basilisk V3 Pro 35K doesn't seems to be detected through Bluetooth even after official support from 3.10.0 .

I seems that it get another PID from one expected (1532:00CC or 1532:00CD) when connected through Bluetooth :

```log
[  390.182736] input: BSKV3P 35K Mouse as /devices/virtual/misc/uhid/0005:068E:00CE.0006/input/input33
[  390.183152] input: BSKV3P 35K Consumer Control as /devices/virtual/misc/uhid/0005:068E:00CE.0006/input/input34
[  390.183214] input: BSKV3P 35K System Control as /devices/virtual/misc/uhid/0005:068E:00CE.0006/input/input35
[  390.183258] input: BSKV3P 35K as /devices/virtual/misc/uhid/0005:068E:00CE.0006/input/input36
[  390.183304] input: BSKV3P 35K Keyboard as /devices/virtual/misc/uhid/0005:068E:00CE.0006/input/input37
[  390.183469] hid-generic 0005:068E:00CE.0006: input,hidraw5: BLUETOOTH HID v0.01 Mouse [BSKV3P 35K] on a0:59:50:d4:0e:5d
```

lah7, 2025-03-13:
Sorry for the delay. Does the device show up as a USB device while connected via Bluetooth?

    lsusb

If not, unfortunately it's a limitation of the driver right now. A few other https://github.com/openrazer/openrazer/labels/Bluetooth devices are affected.

The Wireless (1532:00CD) variant will be for the 2.4 GHz dongle connection.

emouty, 2025-03-16:
No it doesn't show in the output of lsub :/

LaptopDev, 2025-03-27:
Not sure if relevant to feb 14th issue; my razor basilisk v3 becomes recognized by my arch system by holding in the two side buttons and the mouse wheel button for 3 seconds. I read this tip about a year ago, not sure where. But it still works for me.

emouty, 2025-11-25:
Still relevant