openrazer/openrazer #2408: openrazer-daemon not setting the correct polling rate

Edu4rdSHL, 2025-03-06:
### Affected Device / PID

Razer Blackwidow V4 Pro

### Operating System / Version

ArchLinux

### Kernel Version

6.13.5-2-cachyos

### OpenRazer Version

3.10.0-1

### Support Checklist

- [x] My user was added to the plugdev group.
- [x] I have rebooted and the problem persists.
- [x] I looked at the [Troubleshooting](https://github.com/openrazer/openrazer/wiki/Troubleshooting) page and it did not solve the problem.

### Symptoms

- [ ] Uninstalling OpenRazer fixes the problem
- [ ] Re-plugging the device fixes the problem
- [ ] It worked before a system/kernel update.
- [ ] Only a problem if the daemon is running.
- [ ] Only a problem from power up or a reboot.
- [ ] I'm dual booting with Windows.
- [ ] Intermittent or erratic problem.
- [ ] My device is missing a feature.

### Please describe the issue

I do have the following `~/.config/openrazer/persistence.conf`

```
[REDACTED]
poll_rate = 8000
backlight_active = True
backlight_brightness = 100
backlight_effect = static
backlight_colors = 0 255 0 0 255 255 0 0 255
backlight_speed = 2
backlight_wave_dir = 1
```
and `openrazer-daemon.service` is running. However, when I check, I get this instead of the expected 8000:
```
cat /sys/module/razerkbd/drivers/hid:razerkbd/0003:1532:028D.000A/poll_rate
1000
```
If I manually do `busctl --user call org.razer /org/razer/device/REDACTED razer.device.misc setPollRate i 8000`, it works.

I have ran `openrazer-daemon` with verbose enabled and everything seems normal:

```
...
2025-03-06 04:39:18 | razer.device0                  | DEBUG    | Adding razer.device.led.macromode.getMacroEffect method to DBus
2025-03-06 04:39:18 | razer.device0                  | DEBUG    | Adding razer.device.led.macromode.setMacroEffect method to DBus
2025-03-06 04:39:18 | razer.device0                  | DEBUG    | Adding razer.device.macro.getMacros method to DBus
2025-03-06 04:39:18 | razer.device0                  | DEBUG    | Adding razer.device.macro.deleteMacro method to DBus
2025-03-06 04:39:18 | razer.device0                  | DEBUG    | Adding razer.device.macro.addMacro method to DBus
2025-03-06 04:39:18 | razer.device0                  | DEBUG    | Adding razer.device.lighting.chroma.setStarlightRandom method to DBus
2025-03-06 04:39:18 | razer.device0                  | DEBUG    | Adding razer.device.lighting.chroma.setStarlightSingle method to DBus
2025-03-06 04:39:18 | razer.device0                  | DEBUG    | Adding razer.device.lighting.chroma.setStarlightDual method to DBus
2025-03-06 04:39:18 | razer.device0                  | DEBUG    | Adding razer.device.misc.getPollRate method to DBus
2025-03-06 04:39:18 | razer.device0                  | DEBUG    | Adding razer.device.misc.setPollRate method to DBus
2025-03-06 04:39:18 | razer.device0                  | DEBUG    | Adding razer.device.misc.getSupportedPollRates method to DBus
2025-03-06 04:39:18 | razer.device0                  | DEBUG    | Adding razer.device.lighting.custom.setRipple method to DBus
2025-03-06 04:39:18 | razer.device0                  | DEBUG    | Adding razer.device.lighting.custom.setRippleRandomColour method to DBus
2025-03-06 04:39:18 | razer.device0                  | DEBUG    | DBus call set_poll_rate
2025-03-06 04:39:18 | razer.device0                  | DEBUG    | DBus call set_brightness
2025-03-06 04:39:18 | razer.device0                  | DEBUG    | Set persistence (backlight, brightness, 100)
2025-03-06 04:39:18 | razer.device0                  | DEBUG    | Sending observer message: ('effect', RazerBlackWidowV4Pro:IO2312F83100455, 'setBrightness', 255)
2025-03-06 04:39:18 | razer.device0.keymanager       | DEBUG    | Starting KeyWatcher
2025-03-06 04:39:18 | razer.device0                  | INFO     | Putting device into driver mode. Daemon will handle special functionality
2025-03-06 04:39:18 | razer                          | DEBUG    | Adding razer.devices.getOffOnScreensaver method to DBus
2025-03-06 04:39:18 | razer                          | DEBUG    | Adding razer.devices.getDevices method to DBus
2025-03-06 04:39:18 | razer                          | DEBUG    | Adding razer.devices.syncEffects method to DBus
2025-03-06 04:39:18 | razer                          | DEBUG    | Adding razer.daemon.stop method to DBus
2025-03-06 04:39:18 | razer                          | DEBUG    | Adding razer.devices.getSyncEffects method to DBus
2025-03-06 04:39:18 | razer                          | DEBUG    | Adding razer.daemon.version method to DBus
2025-03-06 04:39:18 | razer                          | DEBUG    | Adding razer.devices.enableTurnOffOnScreensaver method to DBus
2025-03-06 04:39:18 | razer                          | DEBUG    | Adding razer.devices.supportedDevices method to DBus
2025-03-06 04:39:18 | razer                          | INFO     | Serving DBus
2025-03-06 04:39:28 | razer                          | DEBUG    | State recently changed, writing to disk
2025-03-06 04:39:28 | razer                          | DEBUG    | Writing persistence config
```

So, the issue seems to be with how `openrazer-daemon` is changing the value, any ideas how to further debug? Other values seems to be set correctly.

Edu4rdSHL, 2025-03-13:
I have added this user service to solve the issue for the moment, in case anyone else is having the same problem:

```
[Unit]
Description=Set Razer Keyboard Polling Rate
After=openrazer-daemon.service
Requires=openrazer-daemon.service

[Service]
Type=oneshot
ExecStart=busctl --user call org.razer /org/razer/device/IO2312F83100455 razer.device.misc setPollRate i 8000
RemainAfterExit=yes

[Install]
WantedBy=default.target
```

Edu4rdSHL, 2025-03-13:
Hi, @lah7, thanks for your prompt reply. I have tried what you suggested, but the result is the same, the poll rate doesn't change.

Here's the log of `openrazer-daemon -v --config /home/edu4rdshl/.config/openrazer/razer.conf --persistence /home/edu4rdshl/.config/openrazer/persistence.conf -F`

```
2025-03-13 14:46:12 | razer                          | DEBUG    | Loaded device specification: RazerViperV3ProWireless ----------------- (1532:00C1)
2025-03-13 14:46:12 | razer                          | INFO     | Found device.0: 0003:1532:028D.0003
2025-03-13 14:46:12 | razer.device0                  | INFO     | Initialising device.0 RazerBlackWidowV4Pro
2025-03-13 14:46:12 | razer.device0                  | DEBUG    | Adding razer.device.misc.getVidPid method to DBus
2025-03-13 14:46:12 | razer.device0                  | DEBUG    | Adding razer.device.misc.resumeDevice method to DBus
2025-03-13 14:46:12 | razer.device0                  | DEBUG    | Adding razer.device.misc.suspendDevice method to DBus
2025-03-13 14:46:12 | razer.device0                  | DEBUG    | Adding razer.device.misc.getDeviceImage method to DBus
2025-03-13 14:46:12 | razer.device0                  | DEBUG    | Adding razer.device.misc.getDriverVersion method to DBus
2025-03-13 14:46:12 | razer.device0                  | DEBUG    | Adding razer.device.misc.getRazerUrls method to DBus
2025-03-13 14:46:12 | razer.device0                  | DEBUG    | Adding razer.device.misc.getSerial method to DBus
2025-03-13 14:46:12 | razer.device0                  | DEBUG    | Adding razer.device.misc.setDeviceMode method to DBus
2025-03-13 14:46:12 | razer.device0                  | DEBUG    | Adding razer.device.lighting.chroma.restoreLastEffect method to DBus
2025-03-13 14:46:12 | razer.device0                  | DEBUG    | Adding razer.device.misc.hasDedicatedMacroKeys method to DBus
2025-03-13 14:46:12 | razer.device0                  | DEBUG    | Adding razer.device.misc.getDeviceMode method to DBus
2025-03-13 14:46:12 | razer.device0                  | DEBUG    | Adding razer.device.lighting.chroma.getEffectSpeed method to DBus
2025-03-13 14:46:12 | razer.device0                  | DEBUG    | Adding razer.device.lighting.chroma.getEffect method to DBus
2025-03-13 14:46:12 | razer.device0                  | DEBUG    | Adding razer.device.lighting.chroma.getWaveDir method to DBus
2025-03-13 14:46:12 | razer.device0                  | DEBUG    | Adding razer.device.lighting.chroma.getEffectColors method to DBus
2025-03-13 14:46:12 | razer.device0                  | DEBUG    | Adding razer.device.misc.getFirmware method to DBus
2025-03-13 14:46:12 | razer.device0                  | DEBUG    | Adding razer.device.misc.getMatrixDimensions method to DBus
2025-03-13 14:46:12 | razer.device0                  | DEBUG    | Adding razer.device.misc.hasMatrix method to DBus
2025-03-13 14:46:12 | razer.device0                  | DEBUG    | Adding razer.device.misc.getDeviceName method to DBus
2025-03-13 14:46:12 | razer.device0                  | DEBUG    | Adding razer.device.misc.getKeyboardLayout method to DBus
2025-03-13 14:46:12 | razer.device0                  | DEBUG    | Adding razer.device.lighting.brightness.getBrightness method to DBus
2025-03-13 14:46:12 | razer.device0                  | DEBUG    | Adding razer.device.lighting.brightness.setBrightness method to DBus
2025-03-13 14:46:12 | razer.device0                  | DEBUG    | Adding razer.device.misc.getDeviceType method to DBus
2025-03-13 14:46:12 | razer.device0                  | DEBUG    | Adding razer.device.lighting.chroma.setWave method to DBus
2025-03-13 14:46:12 | razer.device0                  | DEBUG    | Adding razer.device.lighting.chroma.setWheel method to DBus
2025-03-13 14:46:12 | razer.device0                  | DEBUG    | Adding razer.device.lighting.chroma.setStatic method to DBus
2025-03-13 14:46:12 | razer.device0                  | DEBUG    | Adding razer.device.lighting.chroma.setSpectrum method to DBus
2025-03-13 14:46:12 | razer.device0                  | DEBUG    | Adding razer.device.lighting.chroma.setReactive method to DBus
2025-03-13 14:46:12 | razer.device0                  | DEBUG    | Adding razer.device.lighting.chroma.setNone method to DBus
2025-03-13 14:46:12 | razer.device0                  | DEBUG    | Adding razer.device.lighting.chroma.setBreathRandom method to DBus
2025-03-13 14:46:12 | razer.device0                  | DEBUG    | Adding razer.device.lighting.chroma.setBreathSingle method to DBus
2025-03-13 14:46:12 | razer.device0                  | DEBUG    | Adding razer.device.lighting.chroma.setBreathDual method to DBus
2025-03-13 14:46:12 | razer.device0                  | DEBUG    | Adding razer.device.lighting.chroma.setCustom method to DBus
2025-03-13 14:46:12 | razer.device0                  | DEBUG    | Adding razer.device.lighting.chroma.setKeyRow method to DBus
2025-03-13 14:46:12 | razer.device0                  | DEBUG    | Adding razer.device.led.gamemode.getGameMode method to DBus
2025-03-13 14:46:12 | razer.device0                  | DEBUG    | Adding razer.device.led.gamemode.setGameMode method to DBus
2025-03-13 14:46:12 | razer.device0                  | DEBUG    | Adding razer.device.led.macromode.getMacroMode method to DBus
2025-03-13 14:46:12 | razer.device0                  | DEBUG    | Adding razer.device.led.macromode.setMacroMode method to DBus
[... cut ...]
2025-03-13 14:46:13 | razer                          | DEBUG    | Adding razer.devices.getDevices method to DBus
2025-03-13 14:46:13 | razer                          | INFO     | Serving DBus
2025-03-13 14:46:23 | razer                          | DEBUG    | State recently changed, writing to disk
2025-03-13 14:46:23 | razer                          | DEBUG    | Writing persistence config
```

The quirk is triggered, but the `poll_rate` is still 1000:

```
$ cat /sys/module/razerkbd/drivers/hid:razerkbd/0003:1532:028D.000*/poll_rate
1000
```

lah7, 2025-03-13:
We need to rule out if it's definitely a device quirk or if there's a persistence logic bug when the daemon starts.

In that file, `device_base.py`, do you mind adding a `print()` line?

```diff
    def restore_dpi_poll_rate(self):

        ...

        poll_rate_func = getattr(self, "setPollRate", None)
        if poll_rate_func is not None:
            # Constrain value in case the available values have changed, e.g. wired/wireless might different values available
            if self.poll_rate not in self.POLL_RATES:
                self.logger.warning("Constraining poll rate because stored value " + str(self.poll_rate) + " is not available.")
                self.poll_rate = min(self.POLL_RATES, key=lambda x: abs(x - self.poll_rate))

+           print(type(poll_rate_func), self.poll_rate, self.POLL_RATES)
            poll_rate_func(self.poll_rate)
```
_(Indentation matters, Python will be picky if it's not consistent)_

By the way, do you get any errors from the driver in the kernel log when this happens? Like `Command timed out`?

    sudo dmesg | grep razer

Edu4rdSHL, 2025-03-13:
It's what I'm getting:

```
2025-03-13 16:49:21 | razer.device0                  | DEBUG    | Adding razer.device.lighting.chroma.setStarlightSingle method to DBus
2025-03-13 16:49:21 | razer.device0                  | DEBUG    | Adding razer.device.lighting.chroma.setStarlightDual method to DBus
2025-03-13 16:49:21 | razer.device0                  | DEBUG    | Adding razer.device.misc.getPollRate method to DBus
2025-03-13 16:49:21 | razer.device0                  | DEBUG    | Adding razer.device.misc.setPollRate method to DBus
2025-03-13 16:49:21 | razer.device0                  | DEBUG    | Adding razer.device.misc.getSupportedPollRates method to DBus
2025-03-13 16:49:21 | razer.device0                  | DEBUG    | Adding razer.device.lighting.custom.setRipple method to DBus
2025-03-13 16:49:21 | razer.device0                  | DEBUG    | Adding razer.device.lighting.custom.setRippleRandomColour method to DBus
<class 'method'> 8000 [125, 250, 500, 1000, 2000, 4000, 8000]
2025-03-13 16:49:21 | razer.device0                  | DEBUG    | DBus call set_poll_rate
2025-03-13 16:49:21 | razer.device0                  | DEBUG    | DBus call set_brightness
2025-03-13 16:49:21 | razer.device0                  | DEBUG    | Set persistence (backlight, brightness, 100)
2025-03-13 16:49:21 | razer.device0                  | DEBUG    | Sending observer message: ('effect', RazerBlackWidowV4Pro:IO2312F83100455, 'setBrightness', 255)
2025-03-13 16:49:21 | razer.device0                  | DEBUG    | DBus call set_static_effect
2025-03-13 16:49:21 | razer.device0                  | DEBUG    | Sending observer message: ('effect', RazerBlackWidowV4Pro:REDACTED, 'setStatic', 0, 255, 0)
2025-03-13 16:49:21 | razer.device0                  | DEBUG    | Set persistence (backlight, effect, static)
2025-03-13 16:49:21 | razer.device0                  | DEBUG    | Restoring persistence again (device quirk)
2025-03-13 16:49:21 | razer.device0                  | DEBUG    | DBus call set_static_effect
2025-03-13 16:49:21 | razer.device0                  | DEBUG    | Sending observer message: ('effect', RazerBlackWidowV4Pro:REDACTED, 'setStatic', 0, 255, 0)
2025-03-13 16:49:21 | razer.device0                  | DEBUG    | Set persistence (backlight, effect, static)
<class 'method'> 8000 [125, 250, 500, 1000, 2000, 4000, 8000]
2025-03-13 16:49:21 | razer.device0                  | DEBUG    | DBus call set_poll_rate
2025-03-13 16:49:22 | razer.device0                  | DEBUG    | DBus call set_brightness
2025-03-13 16:49:22 | razer.device0                  | DEBUG    | Set persistence (backlight, brightness, 100)
2025-03-13 16:49:22 | razer.device0                  | DEBUG    | Sending observer message: ('effect', RazerBlackWidowV4Pro:REDACTED, 'setBrightness', 255)
2025-03-13 16:49:22 | razer.device0.keymanager       | DEBUG    | Starting KeyWatcher
2025-03-13 16:49:22 | razer.device0                  | INFO     | Putting device into driver mode. Daemon will handle special functionality
2025-03-13 16:49:22 | razer                          | DEBUG    | Adding razer.devices.getSyncEffects method to DBus
2025-03-13 16:49:22 | razer                          | DEBUG    | Adding razer.devices.syncEffects method to DBus
2025-03-13 16:49:22 | razer                          | DEBUG    | Adding razer.devices.getOffOnScreensaver method to DBus
2025-03-13 16:49:22 | razer                          | DEBUG    | Adding razer.daemon.stop method to DBus
2025-03-13 16:49:22 | razer                          | DEBUG    | Adding razer.devices.enableTurnOffOnScreensaver method to DBus
2025-03-13 16:49:22 | razer                          | DEBUG    | Adding razer.devices.supportedDevices method to DBus
2025-03-13 16:49:22 | razer                          | DEBUG    | Adding razer.devices.getDevices method to DBus
2025-03-13 16:49:22 | razer                          | DEBUG    | Adding razer.daemon.version method to DBus
2025-03-13 16:49:22 | razer                          | INFO     | Serving DBus
2025-03-13 16:49:32 | razer                          | DEBUG    | State recently changed, writing to disk
2025-03-13 16:49:32 | razer                          | DEBUG    | Writing persistence config
```

Regarding the timeout question, there's nothing:

```
sudo dmesg | grep razer

[    5.552292] razerkbd 0003:1532:028D.0001: input,hidraw0: USB HID v1.11 Keyboard [Razer Razer BlackWidow V4 Pro] on usb-0000:0a:00.3-2/input0
[    5.770297] razerkbd 0003:1532:028D.0002: input,hidraw1: USB HID v1.00 Keyboard [Razer Razer BlackWidow V4 Pro] on usb-0000:0a:00.3-2/input1
[    5.848286] razerkbd 0003:1532:028D.0003: input,hidraw2: USB HID v1.00 Mouse [Razer Razer BlackWidow V4 Pro] on usb-0000:0a:00.3-2/input2
[    5.848367] razerkbd 0003:1532:028D.0004: hidraw3: USB HID v1.00 Device [Razer Razer BlackWidow V4 Pro] on usb-0000:0a:00.3-2/input3
[    5.848504] razerkbd 0003:1532:028D.0005: hiddev96,hidraw4: USB HID v1.00 Device [Razer Razer BlackWidow V4 Pro] on usb-0000:0a:00.3-2/input4
```

Edu4rdSHL, 2025-03-13:
Okay, after further debugging, I watched the `poll_rate` with `while true; do cat /sys/module/razerkbd/drivers/hid:razerkbd/0003:1532:028D.000*/poll_rate; done` and it's being set when the daemon starts, and then immediately reset to 1000 again.

```
...
1000
1000
1000
1000
1000
1000
1000
1000
8000
8000
8000
8000
8000
8000
8000
8000
8000
8000
1000
1000
1000
1000
1000
1000
1000
```
If I put a `time.sleep(30)` in https://github.com/openrazer/openrazer/blob/a09f59b5e0a2a51b5a11e369412da851942d3f75/daemon/openrazer_daemon/dbus_services/dbus_methods/mamba.py#L320 and the `poll_rate` is kept at 8000 in the 30 seconds, so it's something that happens before that reset it back.

lah7, 2025-03-13:
Looks like a firmware quirk. _Theoretically_, we don't know if the firmware is secretly at `8000`, but has a bug in reporting it back (`1000`), or if it really is dropping down to 1000.

Echoing `$(date)` might help in your debugging to match up with the daemon log to check whether certain requests (like `set_brightness`) caused the poll rate to change.

I'm not sure if online tools like [this one](https://joltfly.com/keyboard-scan-rate-test/) will be any help to verify the estimated poll rate. These online tools are reporting 25 Hz for my BlackWidow Ultimate 2016. _(I just discovered [my specs](https://mysupport.razer.com/app/answers/detail/a_id/3653/~/razer-blackwidow-ultimate-2016-%7C-rz03-0170x-support-%26-faqs) says 1000Hz Ultrapolling, I didn't know!?)_

Edu4rdSHL, 2025-03-13:
Thanks, it helped a lot. I went ahead and put a `time.sleep(30)` at https://github.com/lah7/openrazer/commit/b35fb1399ea52de1180f744a5f0ac8a1212fbeab#diff-03a1e124019cb030df54a275504f3cd4957252350122c2b35400cc5b702aadaaR339, and the rate is kept at 8000, so it wasn't `set_brightness`.

I suspected of the `putting device in driver mode` operation, and while having the sleep there, manually ran `printf "\x03\x00" > /sys/devices/pci0000:00/0000:00:08.1/0000:0a:00.3/usb3/3-2/3-2:1.2/0003:1532:028D.0003/device_mode`, which changed the rate again to `1000`.

So, it appears that the root problem is on this line: https://github.com/openrazer/openrazer/blob/a09f59b5e0a2a51b5a11e369412da851942d3f75/daemon/openrazer_daemon/hardware/keyboards.py#L31

Can't be this mode be initialized first and then the parameters applied? Or what workaround you suggest?

lah7, 2025-03-14:
Good find! You could try moving that line above  `super().__init__` since that's calling the parent classes, where this persistence restoration code runs. 

    _MacroKeyboard --> RazerDeviceBrightnessSuspend --> RazerDevice

Like so:

```diff
+       self.logger.info('Putting device into driver mode. Daemon will handle special functionality')
+       self.set_device_mode(0x03, 0x00)  # Driver mode
+
        super().__init__(*args, **kwargs)
        # Methods are loaded into DBus by this point

        self.key_manager = _KeyboardKeyManager(self._device_number, self.event_files, self, use_epoll=True, testing=self._testing)

-       self.logger.info('Putting device into driver mode. Daemon will handle special functionality')
-       self.set_device_mode(0x03, 0x00)  # Driver mode

```

If that works, we can figure out whether it would be safe to assume other Razer devices are put into driver mode first, or just make a change specific to the BlackWidow V4 Pro so only that device applies its poll rate a second time if persistence is enabled.  [My earlier change might not be necessary.](https://github.com/openrazer/openrazer/issues/2408#issuecomment-2721187820)

Edu4rdSHL, 2025-03-14:
Well, sadly that didn't work. Once I made the change, I got many errors about the device attributes not being recognized, which makes sense because the device hasn't been initialized.

```
2025-03-14 11:28:45 | razer                          | INFO     | Found device.0: 0003:1532:028D.000C
Traceback (most recent call last):
  File "/usr/lib/python3.13/site-packages/daemonize.py", line 248, in start
    self.action(*privileged_action_result)
    ~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/bin/openrazer-daemon", line 141, in run_daemon
    daemon = RazerDaemon(verbose=args.verbose,
                         log_dir=args.log_dir,
    ...<2 lines>...
                         persistence_file=args.persistence,
                         test_dir=args.test_dir)
  File "/usr/lib/python3.13/site-packages/openrazer_daemon/daemon.py", line 120, in __init__
    self._load_devices(first_run=True)
    ~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.13/site-packages/openrazer_daemon/daemon.py", line 495, in _load_devices
    razer_device = device_class(device_path=sys_path, device_number=device_number, config=self._config,
                                persistence=self._persistence, testing=self._test_dir is not None,
                                additional_interfaces=sorted(additional_interfaces),
                                additional_methods=[],
                                unknown_serial_counter=self._unknown_serial_counter)
  File "/usr/lib/python3.13/site-packages/openrazer_daemon/hardware/keyboards.py", line 74, in __init__
    super().__init__(*args, **kwargs)
    ~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.13/site-packages/openrazer_daemon/hardware/keyboards.py", line 26, in __init__
    self.logger.info('Putting device into driver mode. Daemon will handle special functionality')
    ^^^^^^^^^^^
AttributeError: 'RazerBlackWidowV4Pro' object has no attribute 'logger'

Exception ignored in: <function RazerDevice.__del__ at 0x732123db71a0>
Traceback (most recent call last):
  File "/usr/lib/python3.13/site-packages/openrazer_daemon/hardware/device_base.py", line 1298, in __del__
    self.close()
  File "/usr/lib/python3.13/site-packages/openrazer_daemon/hardware/device_base.py", line 1181, in close
    if not self._is_closed:
AttributeError: 'RazerBlackWidowV4Pro' object has no attribute '_is_closed'
Stopping daemon.
Exception ignored in atexit callback <bound method Daemonize.exit of <daemonize.Daemonize object at 0x7321257f7b60>>:
Traceback (most recent call last):
  File "/usr/lib/python3.13/site-packages/daemonize.py", line 72, in exit
    sys.exit(0)
SystemExit: 0
```
Removing the `logger` line takes me to:
```
AttributeError: 'RazerBlackWidowV4Pro' object has no attribute '_device_path'
```
which again, makes sense.

lah7, 2025-03-14:
Thanks for trying. Silly me, I should've checked `set_device_mode()` first! 🙃 

The second approach, to fix each device individually with device-specific code _(re-apply after everything else initialises)_ might be the way forward. Please try these changes: https://github.com/lah7/openrazer/commit/1922fdfb9d1d173f7ee727db99952b8e847f6c99 - To avoid duplicating the config key, I did a minor refactor in `device_base.py`

Edu4rdSHL, 2025-03-14:
Thank you for getting back, these changes resolved the issue. Just a quick note, wouldn't it make sense to change `self._disable_persistence` to `self._enable_persistence` globally? That will avoid the double negation on every condition that runs when persistence is enabled. i.e, instead of

```python
self._disable_persistence = config.getboolean('Startup', "restore_persistence") is not True
...

if not self._disable_persistence:
```
you'll have:
```python
self._enable_persistence = config.getboolean('Startup', "restore_persistence") is True
....

if self._enable_persistence:
```
That's just a styling suggestion.

lah7, 2025-03-14:
Yes, I'd agree. https://github.com/openrazer/openrazer/commit/1547925c80680ec5b4012a327b1c7c13cf86a190 first added the variable, it seems to be used internally for the suspend/resume calls. Tread carefully as there's also `disable_notify` which would make the other variables inconsistent.

Sure, feel free to take it from here, test and create a PR. A `Co-authored-by` attribution<sup>[[1]](https://docs.github.com/en/pull-requests/committing-changes-to-your-project/creating-and-editing-commits/creating-a-commit-with-multiple-authors)</sup> would be nice if you were to commit [the keyboard part](https://github.com/lah7/openrazer/commit/1922fdfb9d1d173f7ee727db99952b8e847f6c99) separately, but up to you!

Edu4rdSHL, 2025-03-15:
I started with the keyboard part, which is the priority now, thereafter, I will check the variable name change more meticulously.

PR #2416 (2025-07-06): daemon: Add DRIVER_MODE variable
> As reviewed in #2415

This fixes the BlackWidow V4 Pro being unable to apply a 8000 Hz poll rate when the daemon starts, as the device reset to 1000 Hz after entering "driver mode", defeating the point of persistence.

Adding a variable allows this device to enter driver mode first while retaining existing behaviour for all our current devices.

Fixes #2408

Release note suggestion:

> Fix BlackWidow V4 Pro unable to restore poll rates above 1000 Hz.
commit c35e8783 (2025-07-06): daemon: Introduce DRIVER_MODE variable

Introduce a variable to declaratively switch the device to driver mode.

An intended side effect is that we now switch to driver mode earlier,
*before* applying persistence.

This fixes a problem for the BlackWidow V4 Pro, where the keyboard would
reset to 1000 Hz after driver mode, defeating the 8000 Hz applied by
persistence.

Fixes #2408
closed 2025-07-06 by z3ntu