openrazer/openrazer #2409: Blackwidow v4 Pro as well as Bsilisk Ultimate + Charging station remain lit after PC is powered off.

joshuastal, 2025-03-08:
Title. I read another user with the same issue but those fixes didn't help. This is an open razer issue as I don't have the issue on windows, my PC is duel boot with separate physical drives. I'm using arch Linux. I need the usb devices powered so I can charge the mouse when the PC is not in use. I'm sure that if I were to turn off power to the USB devices it's either all are powered off or none are powered off. First time creating an issue on GitHub so forgive me for my inefficiency for creating this issue.

lah7, 2025-03-08:
I believe the problem is actually the BIOS/UEFI. It's responsible for handling the power when the system is shut down. Most likely, your BIOS/ACPI doesn't behave the same with Linux as it does in Windows. To rule it out further, observe what happens when you shut down from a bootloader _(systemd-boot, GRUB)_ and BIOS/UEFI _(DEL, F2)_ setup screen.

Which issues did you look at which didn't help? This one got into detail with some settings to look for:

* https://github.com/openrazer/openrazer/issues/2361

---

> This is an open razer issue as I don't have the issue on windows

To prove this, please uninstall all packages for OpenRazer:

    sudo pacman -R openrazer-driver-dkms openrazer-daemon python-openrazer

Shut down, completely power off at the PSU switch (to fully power cycle), then boot everything back up and shut down Linux again (without OpenRazer this time) and let us know if the issue persists.

joshuastal, 2025-03-09:
> I believe the problem is actually the BIOS/UEFI. It's responsible for handling the power when the system is shut down. Most likely, your BIOS/ACPI doesn't behave the same with Linux as it does in [...]

Yes I did look at that other issue and that is not the fix for me. This happened on Linux mint too so reinstalling wouldn't do anything to be honest. How do I shut down the computer from the boot loader menu?

lah7, 2025-03-09:
> How do I shut down the computer from the boot loader menu?

Since I don't know which bootloader you have installed with Arch Linux [(there's many)](https://wiki.archlinux.org/title/Category:Boot_loaders). It's probably easier to plug in the Arch Linux installer USB, then press <kbd>CTRL</kbd>+<kbd>O</kbd> to power off.

[image]

Do the LEDs stay on after shutting down from here?

joshuastal, 2025-03-21:
> > How do I shut down the computer from the boot loader menu?
> 
> Since I don't know which bootloader you have installed with Arch Linux [(there's many)](https://wiki.archlinux.org/title/Category:Boot_loaders). It's probably easier to plug in the Arch Linux installer USB, then press SHIFT+O to power off.
> 
> [image]
> 
> Do the LEDs stay on after shutting down from here?

I've since moved to Nobara and you were right it is a BIOS issue because it is happening there too. It was even happening without openrazer installed. However, I do charge my mouse through its USB dock when my PC is turned off. How can I edit my BIOS to keep charging my mouse while turning off my keyboard's lights?

lah7, 2025-03-21:
Check the manual for your motherboard - it should mention this "feature". Some boards may only keep specific USB ports active. It's to do with [S4 or S5 states](https://superuser.com/a/71843). It might called "USB power delivery", "Deep Sleep" or "ErP Ready", and could be under power management, ACPI or USB settings. You'll have to play around to see if any of these options do the trick.

The alternate solution (in #2361's case) might be to create service files to have the OS (Linux) suspend the USB devices at shutdown: https://unix.stackexchange.com/questions/789009/keyboard-and-mouse-rgb-not-turning-off-on-shutdown-when-connected-via-usb3-switc

Since it's not a problem with OpenRazer, we'll close this issue. If you get stuck, a Linux or distro forum might be better places for further assistance.

closed 2025-03-21 by lah7