openrazer/openrazer #2410: Device Is Detected, But Changes Cannot be Applied

cbmiller2610, 2025-03-11:
### Affected Device / PID

Razer Blackwidow X Chroma Tournament Edition

### Operating System / Version

openSUSE Tumbleweed / 20250307

### Kernel Version

6.13.5-1-default

### OpenRazer Version

3.10.0-1.2

### Support Checklist

- [x] My user was added to the plugdev group.
- [x] I have rebooted and the problem persists.
- [x] I looked at the [Troubleshooting](https://github.com/openrazer/openrazer/wiki/Troubleshooting) page and it did not solve the problem.

### Symptoms

- [ ] Uninstalling OpenRazer fixes the problem
- [ ] Re-plugging the device fixes the problem
- [ ] It worked before a system/kernel update.
- [ ] Only a problem if the daemon is running.
- [ ] Only a problem from power up or a reboot.
- [ ] I'm dual booting with Windows.
- [ ] Intermittent or erratic problem.
- [ ] My device is missing a feature.

### Please describe the issue

This is my first install of OpenRazer.  I have done all the troubleshooting instructions, linked below is my troubleshooting screen from Polychromatic: 
https://paste.opensuse.org/pastes/5fcbc2174f05

None of the changes I make in Polychromatic are being applied to my keyboard.

Looking at dmesg, I see the following error repeated:

```
[  967.855413] [   T7585] razer driver: Invalid USB response. USB Report length: 0
[  967.855415] [   T7585] razerkbd: Invalid Report Length. status: 00 transaction_id.id: 00 remaining_packets: 00 protocol_type: 00 data_size: 00, command_class: 00, command_id.id: 00 Params: 00000000000000000000000000000000 .
[  967.861413] [   T7585] razer driver: Invalid USB response. USB Report length: 0
[  967.861416] [   T7585] razerkbd: Invalid Report Length. status: 00 transaction_id.id: 00 remaining_packets: 00 protocol_type: 00 data_size: 00, command_class: 00, command_id.id: 00 Params: 00000000000000000000000000000000 .
[  967.867412] [   T7585] razer driver: Invalid USB response. USB Report length: 0
[  967.867415] [   T7585] razerkbd: Invalid Report Length. status: 00 transaction_id.id: 00 remaining_packets: 00 protocol_type: 00 data_size: 00, command_class: 00, command_id.id: 00 Params: 00000000000000000000000000000000 .
[  967.922414] [   T7585] razer driver: Invalid USB response. USB Report length: 0
[  967.922421] [   T7585] razerkbd: Invalid Report Length. status: 00 transaction_id.id: 00 remaining_packets: 00 protocol_type: 00 data_size: 00, command_class: 00, command_id.id: 00 Params: 00000000000000000000000000000000 .
[  967.928412] [   T7585] razer driver: Invalid USB response. USB Report length: 0
[  967.928415] [   T7585] razerkbd: Invalid Report Length. status: 00 transaction_id.id: 00 remaining_packets: 00 protocol_type: 00 data_size: 00, command_class: 00, command_id.id: 00 Params: 00000000000000000000000000000000 .
[  967.934412] [   T7585] razer driver: Invalid USB response. USB Report length: 0
[  967.934415] [   T7585] razerkbd: Invalid Report Length. status: 00 transaction_id.id: 00 remaining_packets: 00 protocol_type: 00 data_size: 00, command_class: 00, command_id.id: 00 Params: 00000000000000000000000000000000 .
[  967.940412] [   T7585] razer driver: Invalid USB response. USB Report length: 0
[  967.940415] [   T7585] razerkbd: Invalid Report Length. status: 00 transaction_id.id: 00 remaining_packets: 00 protocol_type: 00 data_size: 00, command_class: 00, command_id.id: 00 Params: 00000000000000000000000000000000 .
[  967.946413] [   T7585] razer driver: Invalid USB response. USB Report length: 0
[  967.946416] [   T7585] razerkbd: Invalid Report Length. status: 00 transaction_id.id: 00 remaining_packets: 00 protocol_type: 00 data_size: 00, command_class: 00, command_id.id: 00 Params: 00000000000000000000000000000000 .
[  967.952413] [   T7585] razer driver: Invalid USB response. USB Report length: 0
[  967.952416] [   T7585] razerkbd: Invalid Report Length. status: 00 transaction_id.id: 00 remaining_packets: 00 protocol_type: 00 data_size: 00, command_class: 00, command_id.id: 00 Params: 00000000000000000000000000000000 .
[  967.958413] [   T7585] razer driver: Invalid USB response. USB Report length: 0
[  967.958416] [   T7585] razerkbd: Invalid Report Length. status: 00 transaction_id.id: 00 remaining_packets: 00 protocol_type: 00 data_size: 00, command_class: 00, command_id.id: 00 Params: 00000000000000000000000000000000 .
[  968.013402] [   T7585] razer driver: Invalid USB response. USB Report length: 0
[  968.013407] [   T7585] razerkbd: Invalid Report Length. status: 00 transaction_id.id: 00 remaining_packets: 00 protocol_type: 00 data_size: 00, command_class: 00, command_id.id: 00 Params: 00000000000000000000000000000000 .
[  968.019402] [   T7585] razer driver: Invalid USB response. USB Report length: 0
[  968.019405] [   T7585] razerkbd: Invalid Report Length. status: 00 transaction_id.id: 00 remaining_packets: 00 protocol_type: 00 data_size: 00, command_class: 00, command_id.id: 00 Params: 00000000000000000000000000000000 .
[  968.025402] [   T7585] razer driver: Invalid USB response. USB Report length: 0
[  968.025405] [   T7585] razerkbd: Invalid Report Length. status: 00 transaction_id.id: 00 remaining_packets: 00 protocol_type: 00 data_size: 00, command_class: 00, command_id.id: 00 Params: 00000000000000000000000000000000 .
[  968.031388] [   T7585] razer driver: Invalid USB response. USB Report length: 0
[  968.031390] [   T7585] razerkbd: Invalid Report Length. status: 00 transaction_id.id: 00 remaining_packets: 00 protocol_type: 00 data_size: 00, command_class: 00, command_id.id: 00 Params: 00000000000000000000000000000000 .
[  968.037389] [   T7585] razer driver: Invalid USB response. USB Report length: 0
[  968.037391] [   T7585] razerkbd: Invalid Report Length. status: 00 transaction_id.id: 00 remaining_packets: 00 protocol_type: 00 data_size: 00, command_class: 00, command_id.id: 00 Params: 00000000000000000000000000000000 .
[  968.043387] [   T7585] razer driver: Invalid USB response. USB Report length: 0
[  968.043389] [   T7585] razerkbd: Invalid Report Length. status: 00 transaction_id.id: 00 remaining_packets: 00 protocol_type: 00 data_size: 00, command_class: 00, command_id.id: 00 Params: 00000000000000000000000000000000 .
[  968.049390] [   T7585] razer driver: Invalid USB response. USB Report length: 0
[  968.049392] [   T7585] razerkbd: Invalid Report Length. status: 00 transaction_id.id: 00 remaining_packets: 00 protocol_type: 00 data_size: 00, command_class: 00, command_id.id: 00 Params: 00000000000000000000000000000000 .
```

cbmiller2610, 2025-03-11:
Here is the output: 

```
2025-03-11 12:43:38 | razer                          | INFO     | Serving DBus
2025-03-11 12:43:43 | razer                          | DEBUG    | DBus called get_serial_list
2025-03-11 12:43:43 | razer.device0                  | DEBUG    | DBus call get_device_type
2025-03-11 12:43:43 | razer.device0                  | DEBUG    | DBus call get_device_name
2025-03-11 12:43:43 | razer.device0                  | DEBUG    | DBus call get_device_type
2025-03-11 12:43:43 | razer.device0                  | DEBUG    | DBus call get_firmware
2025-03-11 12:43:43 | razer.device0                  | DEBUG    | DBus call version
2025-03-11 12:43:43 | razer.device0                  | DEBUG    | DBus call has_matrix
2025-03-11 12:43:43 | razer.device0                  | DEBUG    | DBus call has_matrix
2025-03-11 12:43:43 | razer.device0                  | DEBUG    | DBus call get_keyboard_layout
2025-03-11 12:43:43 | razer                          | DEBUG    | DBus called get_serial_list
2025-03-11 12:43:43 | razer.device0                  | DEBUG    | DBus call get_device_type
2025-03-11 12:43:43 | razer.device0                  | DEBUG    | DBus call get_device_name
2025-03-11 12:43:43 | razer.device0                  | DEBUG    | DBus call get_device_type
2025-03-11 12:43:43 | razer.device0                  | DEBUG    | DBus call get_firmware
2025-03-11 12:43:43 | razer.device0                  | DEBUG    | DBus call version
2025-03-11 12:43:43 | razer.device0                  | DEBUG    | DBus call has_matrix
2025-03-11 12:43:43 | razer.device0                  | DEBUG    | DBus call has_matrix
2025-03-11 12:43:43 | razer.device0                  | DEBUG    | DBus call get_keyboard_layout
2025-03-11 12:43:43 | razer.device0                  | DEBUG    | DBus call get_current_effect
2025-03-11 12:43:43 | razer.device0                  | DEBUG    | DBus call get_current_wave_dir
2025-03-11 12:43:43 | razer.device0                  | DEBUG    | DBus call get_current_effect_speed
2025-03-11 12:43:43 | razer.device0                  | DEBUG    | DBus call get_current_effect_colors
2025-03-11 12:43:43 | razer.device0                  | DEBUG    | DBus call get_brightness
2025-03-11 12:43:43 | razer.device0                  | DEBUG    | DBus call get_game_mode
2025-03-11 12:43:45 | razer.device0                  | DEBUG    | DBus call set_breath_single_effect
2025-03-11 12:43:45 | razer.device0                  | DEBUG    | Sending observer message: ('effect', RazerBlackWidowXTournamentEditionChroma:UNKNOWN_1532021A_0000, 'setBreathSingle', dbus.Byte(0), dbus.Byte(255), dbus.Byte(0))
2025-03-11 12:43:45 | razer.device0                  | DEBUG    | Set persistence (backlight, effect, breathSingle)
2025-03-11 12:43:45 | razer.device0                  | DEBUG    | DBus call get_current_effect
2025-03-11 12:43:45 | razer.device0                  | DEBUG    | DBus call get_current_wave_dir
2025-03-11 12:43:45 | razer.device0                  | DEBUG    | DBus call get_current_effect_speed
2025-03-11 12:43:45 | razer.device0                  | DEBUG    | DBus call get_current_effect_colors
2025-03-11 12:43:45 | razer.device0                  | DEBUG    | DBus call get_brightness
2025-03-11 12:43:45 | razer.device0                  | DEBUG    | DBus call get_game_mode
2025-03-11 12:43:48 | razer                          | DEBUG    | State recently changed, writing to disk
2025-03-11 12:43:48 | razer                          | DEBUG    | Writing persistence config
2025-03-11 12:43:56 | razer.device0                  | DEBUG    | DBus call set_breath_single_effect
2025-03-11 12:43:56 | razer.device0                  | DEBUG    | Sending observer message: ('effect', RazerBlackWidowXTournamentEditionChroma:UNKNOWN_1532021A_0000, 'setBreathSingle', dbus.Byte(0), dbus.Byte(0), dbus.Byte(255))
2025-03-11 12:43:56 | razer.device0                  | DEBUG    | Set persistence (backlight, effect, breathSingle)
2025-03-11 12:43:58 | razer                          | DEBUG    | State recently changed, writing to disk
2025-03-11 12:43:58 | razer                          | DEBUG    | Writing persistence config
^[2025-03-11 12:44:23 | razer.device0                  | DEBUG    | DBus call set_static_effect
2025-03-11 12:44:23 | razer.device0                  | DEBUG    | Sending observer message: ('effect', RazerBlackWidowXTournamentEditionChroma:UNKNOWN_1532021A_0000, 'setStatic', dbus.Byte(0), dbus.Byte(0), dbus.Byte(255))
2025-03-11 12:44:23 | razer.device0                  | DEBUG    | Set persistence (backlight, effect, static)
2025-03-11 12:44:23 | razer.device0                  | DEBUG    | DBus call get_current_effect
2025-03-11 12:44:23 | razer.device0                  | DEBUG    | DBus call get_current_wave_dir
2025-03-11 12:44:23 | razer.device0                  | DEBUG    | DBus call get_current_effect_speed
2025-03-11 12:44:23 | razer.device0                  | DEBUG    | DBus call get_current_effect_colors
2025-03-11 12:44:23 | razer.device0                  | DEBUG    | DBus call get_brightness
2025-03-11 12:44:23 | razer.device0                  | DEBUG    | DBus call get_game_mode
2025-03-11 12:44:25 | razer.device0                  | DEBUG    | DBus call set_wave_effect
2025-03-11 12:44:25 | razer.device0                  | DEBUG    | Sending observer message: ('effect', RazerBlackWidowXTournamentEditionChroma:UNKNOWN_1532021A_0000, 'setWave', dbus.Int32(2))
[... cut ...]
2025-03-11 12:44:25 | razer.device0                  | DEBUG    | DBus call get_current_effect_speed
2025-03-11 12:44:25 | razer.device0                  | DEBUG    | DBus call get_current_effect_colors
2025-03-11 12:44:25 | razer.device0                  | DEBUG    | DBus call get_brightness
2025-03-11 12:44:25 | razer.device0                  | DEBUG    | DBus call get_game_mode
2025-03-11 12:44:28 | razer                          | DEBUG    | State recently changed, writing to disk
2025-03-11 12:44:28 | razer                          | DEBUG    | Writing persistence config
```

cbmiller2610, 2025-03-13:
I tried hard resetting by plugging into a Windows computer running Synapse 2. I was successfully able to change colors, effects, etc. I believe there was also a firmware update, which is weird but whatever. I think I used this keyboard with Synapse on windows as recently as 2023.

This resulted in some interesting behavior after plugging back into Linux:

1) I tried 2 USB ports, one USB 2.0 on the top of my PC, and one on my KVM switch. Neither worked. However, after plugging into a USB-C ->USB 3.0 hub, I was suddenly able to change colors and patterns. Things seemed to be working.
2) I then tested plugging back into the KVM, and USB2.0 port. Oddly enough, things stayed working... I was able to change colors and patterns.
3) I then rebooted my computer, with the keyboard plugged into the USB2.0 port. When things came back up, I found that it was not working again. Same errors as before.
4) I then plugged into the USB3.0 hub, things worked. I plugged back into the USB2.0 port, and got an error in polychromatic when changing settings: 
```
Traceback (most recent call last):
  File ’/usr/lib/python3.11/site-packages/polychromatic/controller/devices.py’, line 525, in _clicked_effect_button
    option.apply()
  File ’/usr/lib/python3.11/site-packages/polychromatic/backends/openrazer.py’, line 1046, in apply
    self._rzone.static(rgb[0], rgb[1], rgb[2])
  File ’/usr/lib/python3.11/site-packages/openrazer/client/fx.py’, line 211, in static
    self._lighting_dbus.setStatic(red, green, blue)
  File ’/usr/lib/python3.11/site-packages/dbus/proxies.py’, line 141, in __call__
    return self._connection.call_blocking(self._named_service,
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File ’/usr/lib/python3.11/site-packages/dbus/connection.py’, line 634, in call_blocking
    reply_message = self.send_message_with_reply_and_block(
                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
dbus.exceptions.DBusException: org.freedesktop.DBus.Error.UnknownMethod: Method ’setStatic’ with signature ’yyy’ on interface ’razer.device.lighting.chroma’ doesn’t exist
```

lah7, 2025-03-13:
I wish I asked for the firmware version before / after, just to observe which versions the keyboard had better support for.

Regarding the Polychromatic error, you might need to reload (F5) or hard reload (CTRL+F5) after re-plugging in.

Not 100% sure why it works in different ports at different times. Strange. You could see if unloading and rebinding the driver makes any difference:

```bash
# Close polychromatic (+ tray) if running
openrazer-daemon -s         # Stop daemon
sudo modprobe -r razerkbd   # Unload driver
# Wait a few seconds
sudo modprobe razerkbd     # Load our driver
# Start Polychromatic (this will autostart the daemon again)
```

By the way, is the keyboard in darkness when it's throwing `Invalid USB response` errors, or is it simply 'doing nothing'?

cbmiller2610, 2025-03-14:
I figured it out. It's my KVM switch. I have a TESmart Displayport KVM switch, and I just figured out why it's so inexpensive... the keyboard/mouse "passthrough" mode, which is supposed to do some magic to make the PCs think that the devices are connected all the time (to avoid constant USB connect/disconnect when swapping) is messed up. I think it does something weird that actually borks the OpenRazer driver on startup when connected to those USB slots, and it gets into a weird state where only some USB ports work (at least that's my theory). 

Luckily, the KVM does have a standard USB 2.0 port without that stuff, and the thing takes so long (like 1-2s) to swap monitors anyway I guess I don't care about the passthrough mode since by the time the image appears on the monitor the keyboard and mouse have already reconnected.

I think this issue can be closed, and maybe just add to the list of troubleshooting steps to ask whether or not people are using the keyboard with a KVM switch (probably rare).

closed 2025-03-14 by lah7