openrazer/openrazer #2411: Linux Support for Razer Huntsman v3

maveric7911, 2025-03-11:
### Device Name

huntsman v3 keyboard

### Product Number

 rz03-04970200-r3u1

### Model Number

rz03-0497

### VID:PID

Bus 005 Device 002: ID 1532:02a6 Razer USA, Ltd Razer Huntsman V3 Pro

### USB Details

```shell
I've tried everything including the generic drivers and I can't get the backlight to work. All I need is the backlight working. It work through the bios but as soon as the kernel starts to load the keyboard goes dark. I can type and all that but none of the keys are lit up which makes it kinda hard to type since my room is on the darker side. If you have a work around for now please let me know!!

[crice@sonicnix ~]$ lsusb | grep 1532
Bus 001 Device 003: ID 1532:0528 Razer USA, Ltd Razer BlackShark V2 Pro
Bus 001 Device 005: ID 1532:007a Razer USA, Ltd RC30-0305 Gaming Mouse [Viper Ultimate (Wired)]
Bus 001 Device 006: ID 1532:0e05 Razer USA, Ltd Razer Kiyo Pro
Bus 001 Device 007: ID 1532:48f0 Razer USA, Ltd Razer Base Station V2 Chroma
Bus 001 Device 009: ID 1532:0f20 Razer USA, Ltd Razer Base Station V2 Chroma
Bus 003 Device 003: ID 1532:0c04 Razer USA, Ltd Firefly V2
Bus 005 Device 002: ID 1532:02a6 Razer USA, Ltd Razer Huntsman V3 Pro
```

### Device Input Paths

```shell
[crice@sonicnix ~]$  ls -lah /dev/input/by-id/
total 0
drwxr-xr-x 2 root root 340 Mar 10 21:45 .
drwxr-xr-x 4 root root 760 Mar 10 21:45 ..
lrwxrwxrwx 1 root root  10 Mar 10 21:45 usb-Generic_Razer_Base_Station_V2_Chroma-event-if00 -> ../event17
lrwxrwxrwx 1 root root  10 Mar 10 21:45 usb-RAZER_Razer_Base_Station_V2_Chroma-event-mouse -> ../event16
lrwxrwxrwx 1 root root   9 Mar 10 21:45 usb-RAZER_Razer_Base_Station_V2_Chroma-mouse -> ../mouse2
lrwxrwxrwx 1 root root   9 Mar 10 21:45 usb-Razer_Razer_BlackShark_V2_Pro-event-if03 -> ../event5
lrwxrwxrwx 1 root root  10 Mar 10 21:45 usb-RAZER_Razer_Firefly_V2-event-mouse -> ../event12
lrwxrwxrwx 1 root root   9 Mar 10 21:45 usb-RAZER_Razer_Firefly_V2-mouse -> ../mouse1
lrwxrwxrwx 1 root root   9 Mar 10 21:45 usb-Razer_Razer_Huntsman_V3_Pro-event-if01 -> ../event6
lrwxrwxrwx 1 root root   9 Mar 10 21:45 usb-Razer_Razer_Huntsman_V3_Pro-event-kbd -> ../event3
lrwxrwxrwx 1 root root   9 Mar 10 21:45 usb-Razer_Razer_Huntsman_V3_Pro-if01-event-kbd -> ../event4
lrwxrwxrwx 1 root root   9 Mar 10 21:45 usb-Razer_Razer_Huntsman_V3_Pro-if02-event-mouse -> ../event9
lrwxrwxrwx 1 root root   9 Mar 10 21:45 usb-Razer_Razer_Huntsman_V3_Pro-if02-mouse -> ../mouse0
lrwxrwxrwx 1 root root  10 Mar 10 21:45 usb-Razer_Razer_Viper_Ultimate_000000000000-event-mouse -> ../event10
lrwxrwxrwx 1 root root  10 Mar 10 21:45 usb-Razer_Razer_Viper_Ultimate_000000000000-if01-event-kbd -> ../event13
lrwxrwxrwx 1 root root  10 Mar 10 21:45 usb-Razer_Razer_Viper_Ultimate_000000000000-if02-event-kbd -> ../event22
lrwxrwxrwx 1 root root   9 Mar 10 21:45 usb-Razer_Razer_Viper_Ultimate_000000000000-mouse -> ../mouse3
```

### USB HID Dump

```shell
[crice@sonicnix ~]$ sudo usbhid-dump -m 1532 -ed
[sudo] password for crice: 
005:002:005:DESCRIPTOR         1741658793.225124
 06 00 FF 09 00 A1 01 15 00 26 FF 00 19 00 2A FF
 00 75 08 95 40 09 00 81 02 09 00 91 02 C0

005:002:004:DESCRIPTOR         1741658793.233141
 05 59 09 01 A1 01 85 01 09 02 A1 02 09 03 15 00
 27 FF FF 00 00 75 10 95 01 B1 03 09 04 09 05 09
 06 09 07 09 08 15 00 27 FF FF FF 7F 75 20 95 05
 B1 03 C0 85 02 09 20 A1 02 09 21 15 00 27 FF FF
 00 00 75 10 95 01 B1 02 C0 85 03 09 22 A1 02 09
[... 134 lines cut ...]

001:009:000:DESCRIPTOR         1741658793.283853
 05 01 09 02 A1 01 09 01 A1 00 05 09 19 01 29 05
 15 00 25 01 75 01 95 05 81 02 75 01 95 03 81 03
 06 00 FF 09 40 75 08 95 02 15 81 25 7F 81 02 05
 01 09 38 15 81 25 7F 75 08 95 01 81 06 09 30 09
 31 16 00 80 26 FF 7F 75 10 95 02 81 06 C0 06 00
 FF 09 02 15 00 25 01 75 08 95 5A B1 01 C0

001:007:000:DESCRIPTOR         1741658793.286948
 05 0C 09 01 A1 01 85 03 15 00 25 01 75 01 95 02
 09 E9 09 EA 81 02 95 04 09 CD 09 CF 09 B6 09 B5
 81 02 95 02 81 01 85 04 09 00 75 08 95 26 91 02
 85 05 09 00 95 10 81 02 C0

[crice@sonicnix ~]$
```

### Packet Captures

let me know if you need this

### Synapse Screenshots

it just doesn't show the keyboard with poly or any other linux razer. I'm using Archlinux

maveric7911, 2025-03-11:
I've tried everything including the generic drivers and I can't get the backlight to work. All I need is the backlight working. It work through the bios but as soon as the kernel starts to load the keyboard goes dark. I can type and all that but none of the keys are lit up which makes it kinda hard to type since my room is on the darker side. If you have a work around for now please let me know!! Using Archlinux

lah7, 2025-03-11:
Hi. Your device (1532:02A6) was  recently added, but it's not in any release yet: 7b45c2ca217edcea6597f77b4e52e82ba5691233

For now, uninstall the regular packages:

    sudo pacman -R openrazer-driver-dkms openrazer-daemon python-openrazer

Then, install the `-git` versions from the AUR. For example, if you use `yay`:

    yay -S openrazer-driver-dkms-git openrazer-daemon-git python-openrazer-git

It should start working after a reboot! In the next release (v3.11), you can switch back to the regular packages.

maveric7911, 2025-03-11:
You are amazing! It works perfectly exactly what I needed some backlight!!

closed 2025-03-11 by lah7