openrazer/openrazer #2412: ModuleNotFoundError: No module named 'openrazer_daemon'

Asherathe, 2025-03-11:
### Affected Device / PID

Razer Viper Mini / 1532:008A

### Operating System / Version

openSUSE Tumbleweed

### Kernel Version

6.13.5-1-default

### OpenRazer Version

3.10.0

### Support Checklist

- [x] My user was added to the plugdev group.
- [x] I have rebooted and the problem persists.
- [x] I looked at the [Troubleshooting](https://github.com/openrazer/openrazer/wiki/Troubleshooting) page and it did not solve the problem.

### Symptoms

- [ ] Uninstalling OpenRazer fixes the problem
- [ ] Re-plugging the device fixes the problem
- [x] It worked before a system/kernel update.
- [ ] Only a problem if the daemon is running.
- [ ] Only a problem from power up or a reboot.
- [ ] I'm dual booting with Windows.
- [ ] Intermittent or erratic problem.
- [ ] My device is missing a feature.

### Please describe the issue

Since openSUSE Tumbleweed went to kernel 6.13.5-1 and python 3.13, openrazer-daemon v. 3.10.0, installed from https://download.opensuse.org/repositories/hardware:/razer/openSUSE_Tumbleweed/, no longer works. RazerGenie gives the following status output:

× openrazer-daemon.service - Daemon to manage razer devices in userspace
     Loaded: loaded (/usr/lib/systemd/user/openrazer-daemon.service; enabled; preset: disabled)
     Active: failed (Result: exit-code) since Tue 2025-03-11 00:40:49 EDT; 17ms ago
 Invocation: 017113a86c454c959c3d23993d4550f1
       Docs: man:openrazer-daemon(8)
    Process: 3651 ExecStart=/usr/bin/openrazer-daemon -F (code=exited, status=1/FAILURE)
   Main PID: 3651 (code=exited, status=1/FAILURE)
        CPU: 31ms

Mar 11 00:40:49 localhost systemd[2176]: Starting Daemon to manage razer devices in userspace...
Mar 11 00:40:49 localhost openrazer-daemon[3651]: Traceback (most recent call last):
Mar 11 00:40:49 localhost openrazer-daemon[3651]:   File "/usr/bin/openrazer-daemon", line 12, in <module>
Mar 11 00:40:49 localhost openrazer-daemon[3651]:     from openrazer_daemon.daemon import RazerDaemon, __version__
Mar 11 00:40:49 localhost openrazer-daemon[3651]: ModuleNotFoundError: No module named 'openrazer_daemon'
Mar 11 00:40:49 localhost systemd[2176]: openrazer-daemon.service: Main process exited, code=exited, status=1/FAILURE
Mar 11 00:40:49 localhost systemd[2176]: openrazer-daemon.service: Failed with result 'exit-code'.
Mar 11 00:40:49 localhost systemd[2176]: Failed to start Daemon to manage razer devices in userspace.

z3ntu, 2025-03-13:
Seems the package was automatically rebuilt two days ago, could you update your system and check again?

[image]

Asherathe, 2025-03-16:
Yes, everything seems to work correctly now.

z3ntu, 2025-03-16:
Good to know the automatic rebuilds generally do work, even if they're a bit slow sometimes it seems.

closed 2025-03-16 by z3ntu