openrazer/openrazer #2413: openrazer-daemon.service fails to start on Arch Linux (SystemExit: 0)

Martijnhofstra, 2025-03-12:
# OpenRazer Daemon Fails to Start on Arch Linux (SystemExit: 0)

## System Information

- **Operating System:** Arch Linux
- **Kernel Version:** `6.13.6-arch1-1`
- **Razer Device:** Nommo Chroma (USB-ID `1532:0517`)

## Steps Taken and Observations

### 1. Attempted to Start OpenRazer Daemon

**Command:**
```bash
systemctl --user start openrazer-daemon.service
```

**Outcome:**  
The job failed because the control process exited with an error code.

---

### 2. Checked Service Status and Logs

**Commands:**
```bash
systemctl --user status openrazer-daemon.service
journalctl --user -xeu openrazer-daemon.service
```

**Outcome:**  
The logs indicated a `SystemExit: 0` exception, causing the daemon to exit unexpectedly.

---

### 3. Attempted to Start Daemon with Elevated Privileges

**Command:**
```bash
sudo systemctl --user start openrazer-daemon.service
```

**Outcome:**  
Received an error stating failure to connect to the user scope bus due to undefined `$DBUS_SESSION_BUS_ADDRESS` and `$XDG_RUNTIME_DIR`.

---

### 4. Checked Kernel Headers Installation

**Command:**
```bash
sudo pacman -Syu linux-headers
```

**Outcome:**  
Kernel headers were up to date.

---

### 5. Added User to `plugdev` Group

**Commands:**
```bash
sudo groupadd plugdev
sudo gpasswd -a $USER plugdev
```

**Outcome:**  
The `plugdev` group was created, and the user was added successfully.

---

### 6. Rebooted System

Restarted the system to apply group membership changes.

---

### 7. Attempted to Start OpenRazer Daemon Again

**Command:**
```bash
systemctl --user start openrazer-daemon.service
```

**Outcome:**  
The daemon failed to start, with logs showing the same `SystemExit: 0` exception.

Martijnhofstra, 2025-03-12:
[Morra@archlinux ~]$ systemctl --user status openrazer-daemon.service
× openrazer-daemon.service - Daemon to manage razer devices in userspace
     Loaded: loaded (/usr/lib/systemd/user/openrazer-daemon.service; enabled; preset: enabled)
     Active: failed (Result: exit-code) since Thu 2025-03-13 00:26:17 CET; 2min 50s ago
 Invocation: 744e2123b33d4f4fa487d127399c07a1
       Docs: man:openrazer-daemon(8)
    Process: 2694 ExecStart=/usr/bin/openrazer-daemon -F (code=exited, status=1/FAILURE)
   Main PID: 2694 (code=exited, status=1/FAILURE)
   Mem peak: 7.2M
        CPU: 64ms

Mar 13 00:26:17 archlinux systemd[734]: Starting Daemon to manage razer devices in userspace...
Mar 13 00:26:17 archlinux openrazer-daemon[2694]: Traceback (most recent call last):
Mar 13 00:26:17 archlinux openrazer-daemon[2694]:   File "/usr/bin/openrazer-daemon", line 12, in <module>
Mar 13 00:26:17 archlinux openrazer-daemon[2694]:     from openrazer_daemon.daemon import RazerDaemon, __vers>
Mar 13 00:26:17 archlinux openrazer-daemon[2694]: ModuleNotFoundError: No module named 'openrazer_daemon'
Mar 13 00:26:17 archlinux systemd[734]: openrazer-daemon.service: Main process exited, code=exited, status=1/>
Mar 13 00:26:17 archlinux systemd[734]: openrazer-daemon.service: Failed with result 'exit-code'.
Mar 13 00:26:17 archlinux systemd[734]: Failed to start Daemon to manage razer devices in userspace.

regalis1, 2025-05-11:
i have the same problem, and i didnt messed up anything as far as i know.
It used to work..now it doesnt anymore.

Martijnhofstra, 2025-05-11:
> i have the same problem, and i didnt messed up anything as far as i know.
> It used to work..now it doesnt anymore.

I have changed to steelseries on linux so this has become irrelevant for me. Best thing is probably to downgrade for now.

z3ntu, 2025-07-06:
Seems this is resolved, closing.