openrazer/openrazer #2417: BlackWidow V4 X, media buttons not working with openrazer-daemon

ShadowBelmolve, 2025-03-17:
### Affected Device / PID

Razer USA, Ltd Razer BlackWidow V4 X, 1532:0293

### Operating System / Version

Arch Linux

### Kernel Version

6.13.5-arch1-1

### OpenRazer Version

3.10.1

### Support Checklist

- [x] My user was added to the plugdev group.
- [x] I have rebooted and the problem persists.
- [x] I looked at the [Troubleshooting](https://github.com/openrazer/openrazer/wiki/Troubleshooting) page and it did not solve the problem.

### Symptoms

- [ ] Uninstalling OpenRazer fixes the problem
- [ ] Re-plugging the device fixes the problem
- [ ] It worked before a system/kernel update.
- [x] Only a problem if the daemon is running.
- [ ] Only a problem from power up or a reboot.
- [ ] I'm dual booting with Windows.
- [ ] Intermittent or erratic problem.
- [x] My device is missing a feature.

### Please describe the issue

BlackWidow V4 X media buttons are not working with openrazer-daemon.

The problematic keys are play/pause, next, previous, and mute.
The volume control "wheel" works fine.

When openrazer-daemon is running, the "fn" key seems to be ignored, and the media keys are reported as the default keys (play/pause = home, next = pg up).

When I stop the daemon with `openrazer-daemon -s`, the media keys start working.

I installed the git version of the driver and daemon from AUR, but the problem persists.

z3ntu, 2025-05-10:
You could try some different things in the driver, e.g. remove this line https://github.com/openrazer/openrazer/blob/master/driver/razerkbd_driver.c#L3797 so that `razer_raw_event_standard` gets used. 

Another place is https://github.com/openrazer/openrazer/blob/master/driver/razerkbd_driver.c#L3467 but I don't think it matches the behavior you see, especially regarding Fn button. For the rest it could work.

ShadowBelmolve, 2025-06-14:
Sorry for the delay

I tried both places and removed all `USB_DEVICE_ID_RAZER_BLACKWIDOW_V4*` just to be sure but it did not work.

By the way, is there a good way to reload the driver correctly? When I tried with modprobe I needed to use sudo and openrazer-daemon failed because of permisson issues.
I tried following https://github.com/openrazer/openrazer/wiki/Setting-up-the-keyboard-driver but it seems outdated.
I am rebooting every time I rebuild the driver.