openrazer/openrazer #2418: `plugdev` is discouraged

es20490446e, 2025-03-17:
The current code relies on the user being part of the `plugdev` group. But having such group is strongly discouraged by most distros. As seen [here](https://wiki.archlinux.org/title/Udev#Allowing_regular_users_to_use_devices).

z3ntu, 2025-03-18:
Hi, if you can suggest an alternative we'll happily look into it. But I don't think anything has changed since we did that last, I quickly found these issues/comments about the topic: #456 , https://github.com/openrazer/openrazer/pull/676#issuecomment-420664570 and https://github.com/openrazer/openrazer/issues/1747#issuecomment-1607661725

es20490446e, 2025-03-18:
The question is: if this doesn't get fixed, the software may be easily removed from `systemd` based distros.

Maybe the `plugdev` logic could just be omitted if `systemd` is present:

- At `install_files/udev/99-razer.rules` use `uaccess` ([docs](https://wiki.archlinux.org/title/Udev#Allowing_regular_users_to_use_devices)).
- At `daemon/openrazer_daemon/daemon.py` assume `true` to all `plugdev` logic if `/usr/bin/systemctl` is present.
- Same with `scripts/troubleshoot.sh` and `install_files/udev/razer_mount`.

Do you think this may work?

@daissi @deisi @ivazqueznet @rscm

z3ntu, 2025-03-18:
> The question is: if this doesn't get fixed, the software may be easily removed from systemd based distros.

Not sure what this means. OpenRazer has been packaged for plenty of distros using systemd, what's the problem there?

> At install_files/udev/99-razer.rules use uaccess ([docs](https://wiki.archlinux.org/title/Udev#Allowing_regular_users_to_use_devices)).

Please read the linked issues, at least last time I looked at this uaccess didn't work for the sysfs attributes the driver is using.

es20490446e, 2025-03-19:
After reading these:

- [Arch Wiki](https://wiki.archlinux.org/title/Udev#Allowing_regular_users_to_use_devices): Ubuntu's approach is to create a `plugdev` group that devices are added to, but this practice is not only discouraged by the `systemd` developers, [[2]](https://bugzilla.redhat.com/show_bug.cgi?id=815093) but considered a bug when shipped in `udev rules` on Arch ([FS#35602](https://bugs.archlinux.org/task/35602)).

- [Kay Sievers](https://bugzilla.redhat.com/show_bug.cgi?id=815093#c2): The group `plugdev` is a pretty misguided and inflexible Ubuntu concept to grant device access to ordinary users. We do not put users in system groups to grant access to devices, especially do not define non-sensical groups like: "the group of users that can plug devices into the system". Fedora supports dynamic user ACLs, and does not want to support any static groups for users. The rule needs to be fixed, the group `plugdev` should not exist on Fedora.

- [Toolybird](https://gitlab.archlinux.org/toolybird): We have numerous other pkgs where the reference to `plugdev` is removed via `sed`.

I anticipate that Arch Linux will either decide to patch `openrazer`, on this [report](https://gitlab.archlinux.org/archlinux/packaging/packages/openrazer/-/issues/2), or if not possible remove it.

z3ntu, 2025-03-19:
Again, can you name an alternative to the current situation?

We could create a custom "openrazer" group - not sure this improves anything as it's just a different name for the same thing.

We could set the sysfs attributes from the driver to be read/writeable by any user (e.g. mode 0666)

But apart from that I can't think of any alternative that would work for the current architecture of OpenRazer. And neither really improves much upon the current situation.

es20490446e, 2025-03-20:
Naming the group `openrazer` could have the advantage of automating its removal when uninstalling the package. If it is `plugdev` you can't do that, as other package may be using that group.

Does setting the driver to be read/writable by any user have any disadvantage?

---

If you want to know what I have in my mind, here it goes:

For example, I had this quite popular Kwin add-on called [Autocomposer](https://www.pling.com/p/1502826). Written in JavaScript.

Eventually, when making Wayland the default, Kwin developers decided to remove the API calls that made Autocomposer being able to disable composition.

What do you think I did? Saying that the architecture will no longer allow it to work?

No. I decided to completely rewrite the program in C to do everything by itself. Calling the X11 API directly, which I had no idea how it worked like.

For instance Autocomposer was original written in Bash, calling terminal commands. So I eventually decided to port it to JavaScript, and later to C. This was the third time I completely rewrote the program.

If something is solvable, whatever it takes, I will solve it.

Now I understand that the rest of people may be not as willing or dedicated with their own programs 😁

z3ntu, 2025-04-19:
> Does setting the driver to be read/writable by any user have any disadvantage?

I'm not too sure, it's been like that since before I took over maintenance of the project many years ago. We've had a few issues in the past where "untrusted" access to the interface could've been abused but in reality any user actually using OpenRazer will have added themselves to the group anyways.

The long term goal is moving most of the RGB handling to user space which mitigates the problem as well. Unfortunately with the hundreds of supported devices, doing such a migration is very time intensive and difficult to do without physical access to those devices. I haven't looked into your autocomposer program, but I dare to say OpenRazer is a bit bigger in scope.

es20490446e, 2025-04-19:
What I have done in my distro, [Zenned](https://zenned.gitlab.io/), is to made the [`openrazer` package](https://gitlab.com/es20490446e/express/-/tree/master/recipes/openrazer?ref_type=heads) to use its own group.

z3ntu, 2025-05-10:
Making the sysfs files read/writeable by any user would also allow other system users, e.g. ones used for system daemons to interact with the kernel driver which is not really desirable and does open up the surface more than it's currently.

If distros desire to change the name, they are free to do so, but I won't do it upstream since it will break a lot of user setups without any improvement for the user.

Also user groups that are added in the system should normally not be removed since otherwise the group ID could accidentally be reused for a different user in the future so e.g. systemd-sysusers does not remove users or groups again after they were added.

closed 2025-05-10 by z3ntu

es20490446e, 2025-05-10:
- The `systemd` policy isn't context aware.
- Removing unused stuff is basic quality control, as seen [here](https://www.google.com/search?q=inventory+hides+problems&udm=2).