openrazer/openrazer #2419: openrazer-daemon-0:3.10.1-1.1.noarch not found on DNF mirror - what to do?

Matheos96, 2025-03-19:
### Affected Device / PID

no device

### Operating System / Version

Fedora WS 41

### Kernel Version

6.13.5-200.fc41.x86_64

### OpenRazer Version

3.10.0

### Support Checklist

- [x] My user was added to the plugdev group.
- [x] I have rebooted and the problem persists.
- [x] I looked at the [Troubleshooting](https://github.com/openrazer/openrazer/wiki/Troubleshooting) page and it did not solve the problem.

### Symptoms

- [ ] Uninstalling OpenRazer fixes the problem
- [ ] Re-plugging the device fixes the problem
- [ ] It worked before a system/kernel update.
- [ ] Only a problem if the daemon is running.
- [ ] Only a problem from power up or a reboot.
- [ ] I'm dual booting with Windows.
- [ ] Intermittent or erratic problem.
- [ ] My device is missing a feature.

### Please describe the issue

I am not quite sure if this is the place for this... Hopefully someone can tell me where and how to report this, and potentially how to work around it.

I am trying to update my packages on fedora but it fails as the DNF mirror for openrazer-daemon does not contain the package...

```
158/270] openrazer-daemon-0:3.10.1-1.1.noarch                                                                                                                             100% | 179.0   B/s | 153.0   B |  00m01s
>>> Status code: 404 for https://mirror.aardsoft.fi/opensuse/repositories/hardware%3A/razer/Fedora_41/noarch/openrazer-daemon-3.10.1-1.1.noarch.rpm (IP: 2a01:4f9:1a:7200::135) - https://mirror.aardsoft.fi/opensu
>>> Status code: 404 for https://mirror.aardsoft.fi/opensuse/repositories/hardware%3A/razer/Fedora_41/noarch/openrazer-daemon-3.10.1-1.1.noarch.rpm (IP: 2a01:4f9:1a:7200::135) - https://mirror.aardsoft.fi/opensu
>>> Status code: 404 for https://mirror.aardsoft.fi/opensuse/repositories/hardware%3A/razer/Fedora_41/noarch/openrazer-daemon-3.10.1-1.1.noarch.rpm (IP: 2a01:4f9:1a:7200::135) - https://mirror.aardsoft.fi/opensu
>>> Status code: 404 for https://mirror.aardsoft.fi/opensuse/repositories/hardware%3A/razer/Fedora_41/noarch/openrazer-daemon-3.10.1-1.1.noarch.rpm (IP: 2a01:4f9:1a:7200::135) - https://mirror.aardsoft.fi/opensu
>>> No more mirrors to try - All mirrors were already tried without success 
```

I checked manually at the mirror and yes indeed, the Fedora_41 folder is not present. How could I get it to try a different mirror? https://download.opensuse.org/repositories/hardware:/razer/Fedora_41/ has the package but something indeed forces my requests to that mirror instead of the original repo url... I even tried to modify the hardware:razer.repo in `etc/yum.repos.d/` to be staticly https://download.opensuse.org/repositories/hardware:/razer/Fedora_41/ but it did not change anything..

Thanks for the help, and sorry for this probably not being in the correct place.

z3ntu, 2025-03-19:
Hm that's annoying. So https://mirrors.opensuse.org/ is showing green in the "Build Service" column for mirror.aardsoft.fi so not sure what to do.

But sounds like https://github.com/openSUSE/MirrorCache/issues/558 is about the same and that essentially says it'll be resolved automatically at some point, not sure when though.. Perhaps for now you can force it to use a different mirror (listed in the first link) which contains the Fedora 41 files?

Matheos96, 2025-03-20:
Hi.

Yeah I tried to force it yesterday but probably did something wrong.. Anyway, today it already works. I guess complaining about stuff makes stuff fix themselves :)

closed 2025-03-20 by Matheos96