openrazer/openrazer #2420: Separate ear colors on Razer Kraken Kitty Edition

GoldsteinE, 2025-03-21:
### Affected Device / PID

Razer Kraken Kitty Edition / 1532:0521

### Operating System / Version

NixOS unstable

### Kernel Version

6.13.5

### OpenRazer Version

3.9.0

### Support Checklist

- [x] My user was added to the plugdev group.
- [x] I have rebooted and the problem persists.
- [x] I looked at the [Troubleshooting](https://github.com/openrazer/openrazer/wiki/Troubleshooting) page and it did not solve the problem.

### Symptoms

- [ ] Uninstalling OpenRazer fixes the problem
- [ ] Re-plugging the device fixes the problem
- [ ] It worked before a system/kernel update.
- [ ] Only a problem if the daemon is running.
- [ ] Only a problem from power up or a reboot.
- [ ] I'm dual booting with Windows.
- [ ] Intermittent or erratic problem.
- [x] My device is missing a feature.

### Please describe the issue

Kraken Kitty supports setting colors for ears separately (it works on my Windows machine). When I try to set them with OpenRazer, I only have one zone, so I can only set a single color for both ears and the Razer logos on the sides.

z3ntu, 2025-04-19:
You need to use the "custom effects" that we have in OpenRazer, so with polychromatic this Effect Editor

[image]

Or with RazerGenie you can also open this custom editor functionality and assign static colors to the 4 fields you can control

[image]

Unfortunately we don't have this better integrated yet, but I think the polychromatic Effect Editor comes close to what you're looking for.

Does this answer your question?

GoldsteinE, 2025-04-19:
Yes, 0:2 and 0:3 custom colors work, thank you!

closed 2025-04-19 by GoldsteinE