openrazer/openrazer #2422: Support for Razer Stream Controller

jcgit1, 2025-03-22:
### Device Name

Razer Stream Controller

### Product Number

RZ20-0435

### Model Number

RZ20-0435

### VID:PID

1532:0d06

### USB Details

```shell
Bus 001 Device 023: ID 1532:0d06 Razer USA, Ltd Razer Stream Controller
Device Descriptor:
bLength                18
bDescriptorType         1
bcdUSB               2.00
bDeviceClass          239 Miscellaneous Device
bDeviceSubClass         2 [unknown]
bDeviceProtocol         1 Interface Association
bMaxPacketSize0        64
idVendor           0x1532 Razer USA, Ltd
idProduct          0x0d06 Razer Stream Controller
bcdDevice            1.00
iManufacturer           1 Razer Inc.
iProduct                2 Razer Stream Controller
iSerial                 3 RZ200435014023360400879B1003
bNumConfigurations      1
Configuration Descriptor:
bLength                 9
bDescriptorType         2
wTotalLength       0x009a
bNumInterfaces          4
bConfigurationValue     1
iConfiguration          4 Configuration
bmAttributes         0xa0
(Bus Powered)
Remote Wakeup
MaxPower              100mA
Interface Association:
bLength                 8
bDescriptorType        11
bFirstInterface         0
bInterfaceCount         2
bFunctionClass          2 Communications
bFunctionSubClass       2 Abstract (modem)
bFunctionProtocol       0
iFunction               5 Razer Stream Controller
Interface Descriptor:
bLength                 9
bDescriptorType         4
bInterfaceNumber        0
bAlternateSetting       0
bNumEndpoints           1
bInterfaceClass         2 Communications
bInterfaceSubClass      2 Abstract (modem)
bInterfaceProtocol      1 AT-commands (v.25ter)
iInterface              5 Razer Stream Controller
CDC Header:
bcdCDC               1.10
CDC Call Management:
bmCapabilities       0x03
call management
use DataInterface
bDataInterface          1
CDC ACM:
bmCapabilities       0x06
sends break
line coding and serial state
CDC Union:
bMasterInterface        0
bSlaveInterface         1
Endpoint Descriptor:
bLength                 7
bDescriptorType         5
bEndpointAddress     0x82  EP 2 IN
bmAttributes            3
Transfer Type            Interrupt
Synch Type               None
Usage Type               Data
wMaxPacketSize     0x0040  1x 64 bytes
bInterval               7
Interface Descriptor:
bLength                 9
bDescriptorType         4
bInterfaceNumber        1
bAlternateSetting       0
bNumEndpoints           2
bInterfaceClass        10 CDC Data
bInterfaceSubClass      0 [unknown]
bInterfaceProtocol      0
iInterface              6 CDC DATA interface
Endpoint Descriptor:
bLength                 7
bDescriptorType         5
bEndpointAddress     0x01  EP 1 OUT
bmAttributes            2
Transfer Type            Bulk
Synch Type               None
Usage Type               Data
wMaxPacketSize     0x0200  1x 512 bytes
bInterval               1
Endpoint Descriptor:
bLength                 7
bDescriptorType         5
bEndpointAddress     0x81  EP 1 IN
bmAttributes            2
Transfer Type            Bulk
Synch Type               None
Usage Type               Data
wMaxPacketSize     0x0200  1x 512 bytes
bInterval               1
Interface Descriptor:
bLength                 9
bDescriptorType         4
bInterfaceNumber        2
bAlternateSetting       0
bNumEndpoints           0
bInterfaceClass         1 Audio
bInterfaceSubClass      1 Control Device
bInterfaceProtocol      0
iInterface              7 Razer Stream Controller
AudioControl Interface Descriptor:
bLength                 9
bDescriptorType        36
bDescriptorSubtype      1 (HEADER)
bcdADC               1.00
wTotalLength       0x0009
bInCollection           1
baInterfaceNr(0)        3
Interface Descriptor:
bLength                 9
bDescriptorType         4
bInterfaceNumber        3
bAlternateSetting       0
bNumEndpoints           2
bInterfaceClass         1 Audio
bInterfaceSubClass      3 MIDI Streaming
bInterfaceProtocol      0
iInterface              8 MIDI Streaming
MIDIStreaming Interface Descriptor:
bLength                 7
bDescriptorType        36
bDescriptorSubtype      1 (HEADER)
bcdADC               1.00
wTotalLength       0x001c
MIDIStreaming Interface Descriptor:
bLength                 6
bDescriptorType        36
bDescriptorSubtype      2 (MIDI_IN_JACK)
bJackType               1 Embedded
bJackID                 1
iJack                   4 Configuration
MIDIStreaming Interface Descriptor:
bLength                 6
bDescriptorType        36
bDescriptorSubtype      2 (MIDI_IN_JACK)
bJackType               2 External
bJackID                 2
iJack                   5 Razer Stream Controller
MIDIStreaming Interface Descriptor:
bLength                 9
bDescriptorType        36
bDescriptorSubtype      3 (MIDI_OUT_JACK)
bJackType               1 Embedded
bJackID                 3
bNrInputPins            1
baSourceID( 0)          2
BaSourcePin( 0)         1
iJack                   6 CDC DATA interface
Endpoint Descriptor:
bLength                 7
bDescriptorType         5
bEndpointAddress     0x83  EP 3 IN
bmAttributes            2
Transfer Type            Bulk
Synch Type               None
Usage Type               Data
wMaxPacketSize     0x0200  1x 512 bytes
bInterval               1
MIDIStreaming Endpoint Descriptor:
bLength                 5
bDescriptorType        37
bDescriptorSubtype      1 (Invalid)
bNumEmbMIDIJack         1
baAssocJackID( 0)       3
Endpoint Descriptor:
bLength                 7
bDescriptorType         5
bEndpointAddress     0x03  EP 3 OUT
bmAttributes            2
Transfer Type            Bulk
Synch Type               None
Usage Type               Data
wMaxPacketSize     0x0200  1x 512 bytes
bInterval               1
MIDIStreaming Endpoint Descriptor:
bLength                 5
bDescriptorType        37
bDescriptorSubtype      1 (Invalid)
bNumEmbMIDIJack         1
baAssocJackID( 0)       1
Device Qualifier (for other device speed):
bLength                10
bDescriptorType         6
bcdUSB               2.00
bDeviceClass          239 Miscellaneous Device
bDeviceSubClass         2 [unknown]
bDeviceProtocol         1 Interface Association
bMaxPacketSize0        64
bNumConfigurations      1
Device Status:     0x0000
(Bus Powered)
```

### Device Input Paths

```shell
Doesn't show up.
```

### USB HID Dump

```shell
$ sudo usbhid-dump -m 1532 -ed
001:021:001:DESCRIPTOR         1742661928.327829
05 0D 09 05 A1 01 85 01 09 22 A1 02 15 00 25 01
09 47 09 42 95 02 75 01 81 02 95 01 75 03 25 05
09 51 81 02 75 01 95 03 81 03 05 01 15 00 26 44
0A 75 10 55 0E 65 11 09 30 35 00 46 F9 01 95 01
81 02 46 F6 00 26 34 05 09 31 81 02 C0 05 0D 09
22 A1 02 15 00 25 01 09 47 09 42 95 02 75 01 81
02 95 01 75 03 25 05 09 51 81 02 75 01 95 03 81
03 05 01 15 00 26 44 0A 75 10 55 0E 65 11 09 30
35 00 46 F9 01 95 01 81 02 46 F6 00 26 34 05 09
31 81 02 C0 05 0D 09 22 A1 02 15 00 25 01 09 47
09 42 95 02 75 01 81 02 95 01 75 03 25 05 09 51
81 02 75 01 95 03 81 03 05 01 15 00 26 44 0A 75
10 55 0E 65 11 09 30 35 00 46 F9 01 95 01 81 02
46 F6 00 26 34 05 09 31 81 02 C0 55 0C 66 01 10
47 FF FF 00 00 27 FF FF 00 00 75 10 95 01 05 0D
09 56 81 02 09 54 25 7F 95 01 75 08 81 02 05 09
09 01 25 01 75 01 95 01 81 02 95 07 81 03 05 0D
85 08 09 55 09 59 75 04 95 02 25 0F B1 02 06 00
FF 85 07 09 C5 15 00 26 FF 00 75 08 95 10 B1 02
C0 05 0D 09 0E A1 01 85 F0 09 22 A1 02 09 52 15
00 25 0A 75 08 95 01 B1 02 C0 09 22 A1 00 85 F1
09 57 09 58 75 01 95 02 25 01 B1 02 95 06 B1 03
C0 C0
```

### Packet Captures

### Synapse Screenshots

[image]
[image]
[image]
[image]
[image]
[image]

jcgit1, 2025-03-22:
It wouldn't let me upload the .pcapng file or a .tgz file with the pcapng in it. Could have been a size limitation.
Try this link: https://owncloud.lm9.org/index.php/s/BAjcKr0n7wGgdt2

z3ntu, 2025-04-19:
See this comment: https://github.com/openrazer/openrazer/issues/2014#issuecomment-1426801797

closed 2025-04-19 by z3ntu