openrazer/openrazer #2424: Compiling OpenRazer kernel modules for kernel 6.14.0. DKMS make.log for openrazer-driver-3.6.1 for kernel 6.14.0 (x86_64)

sleeplessaa, 2025-03-27:
Compiling OpenRazer kernel modules for kernel 6.14.0. 

ii  openrazer-driver-dkms                 3.6.1-0
iF  linux-image-6.14.0                    6.14.0-1

Installing deb package for new kernel build from source : 

KERNELDIR=/lib/modules/6.14.0/build make driver....(bad exit status: 2)
Error! Bad return status for module build on kernel: 6.14.0 (x86_64)
Consult /var/lib/dkms/openrazer-driver/3.6.1/build/make.log for more information.
Error! One or more modules failed to install during autoinstall.
Refer to previous errors for more information.
dkms: autoinstall for kernel: 6.14.0 failed!

from :   /var/lib/dkms/openrazer-driver/3.6.1/build/make.log

DKMS make.log for openrazer-driver-3.6.1 for kernel 6.14.0 (x86_64)
Thu Mar 27 22:09:42 EET 2025
-e 
::ESC[32m Compiling OpenRazer kernel modulesESC[0m
========================================
make -C /lib/modules/6.14.0/build M=/var/lib/dkms/openrazer-driver/3.6.1/build/driver modules
make[1]: Entering directory '/usr/src/linux-headers-6.14.0'
make[2]: Entering directory '/var/lib/dkms/openrazer-driver/3.6.1/build/driver'
  CC [M]  razerkbd_driver.o
  CC [M]  razercommon.o
  CC [M]  razerchromacommon.o
razerchromacommon.c:507:21: warning: no previous prototype for 'razer_chroma_extended_matrix_effect_base' [-Wmissing-prototypes]
  507 | struct razer_report razer_chroma_extended_matrix_effect_base(unsigned char arg_size, unsigned char variable_storage, unsigned char led_id, unsigned char effect_id)
      |                     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
razerchromacommon.c:795:21: warning: no previous prototype for 'razer_chroma_mouse_extended_matrix_effect_base' [-Wmissing-prototypes]
  795 | struct razer_report razer_chroma_mouse_extended_matrix_effect_base(unsigned char arg_size, unsigned char variable_storage, unsigned char led_id, unsigned char effect_id)
      |                     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
razerchromacommon.c:1572:21: warning: no previous prototype for 'razer_chroma_misc_get_hyperpolling_wireless_dongle_indicator_led_mode' [-Wmissing-prototypes]
 1572 | struct razer_report razer_chroma_misc_get_hyperpolling_wireless_dongle_indicator_led_mode(void)
      |                     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
  LD [M]  razerkbd.o
  CC [M]  razermouse_driver.o
razermouse_driver.c:4777:5: warning: no previous prototype for 'mouse_button_to_usage' [-Wmissing-prototypes]
 4777 | u32 mouse_button_to_usage(__u16 code)
      |     ^~~~~~~~~~~~~~~~~~~~~
razermouse_driver.c:4786:6: warning: no previous prototype for 'input_button_msc_scan' [-Wmissing-prototypes]
 4786 | void input_button_msc_scan(struct input_dev *input, __u16 button)
      |      ^~~~~~~~~~~~~~~~~~~~~
razermouse_driver.c:4795:6: warning: no previous prototype for 'input_rep4_code' [-Wmissing-prototypes]
 4795 | void input_rep4_code(struct input_dev *input, u8 code, __s32 value)
      |      ^~~~~~~~~~~~~~~
razermouse_driver.c:4858:23: warning: no previous prototype for 'find_intf_with_proto' [-Wmissing-prototypes]
 4858 | struct usb_interface *find_intf_with_proto(struct usb_device *usbdev, u8 proto)
      |                       ^~~~~~~~~~~~~~~~~~~~
razermouse_driver.c: In function 'find_mouse':
razermouse_driver.c:4889:65: error: passing argument 3 of 'device_find_child' from incompatible pointer type [-Werror=incompatible-pointer-types]
 4889 |     dev = device_find_child(&m_intf->dev, (void *)hid_bus_type, dev_is_on_bus);
      |                                                                 ^~~~~~~~~~~~~
      |                                                                 |
      |                                                                 int (*)(struct device *, void *)
In file included from /usr/src/linux-headers-6.14.0/include/linux/usb.h:19,
                 from /usr/src/linux-headers-6.14.0/include/linux/usb/input.h:9,
                 from razermouse_driver.c:10:
/usr/src/linux-headers-6.14.0/include/linux/device.h:1101:49: note: expected 'device_match_t' {aka 'int (*)(struct device *, const void *)'} but argument is of type 'int (*)(struct device *, void *)'
 1101 |                                  device_match_t match);
      |                                  ~~~~~~~~~~~~~~~^~~~~
cc1: some warnings being treated as errors
make[4]: *** [/usr/src/linux-headers-6.14.0/scripts/Makefile.build:207: razermouse_driver.o] Error 1
make[3]: *** [/usr/src/linux-headers-6.14.0/Makefile:1994: .] Error 2
make[2]: *** [/usr/src/linux-headers-6.14.0/Makefile:251: __sub-make] Error 2
make[2]: Leaving directory '/var/lib/dkms/openrazer-driver/3.6.1/build/driver'
make[1]: *** [Makefile:251: __sub-make] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-6.14.0'
make: *** [Makefile:38: driver] Error 2

Any idea how to resolve this?

Thank you.

sleeplessaa, 2025-03-27:
Upgraded to :  

ii openrazer-driver-dkms                 3.10.1-0 

was fixed the problem.

z3ntu, 2025-03-28:
Correct, as per changelog you need to upgrade to v3.10.0 or newer: https://github.com/openrazer/openrazer/releases/tag/v3.10.0

closed 2025-03-28 by z3ntu