openrazer/openrazer #2425: Problem

evgenyez, 2025-03-30:
### Affected Device / PID

Razer Ornata V3 X

### Operating System / Version

Arch Linux 

### Kernel Version

6.13.8-zen1-1-zen

### OpenRazer Version

3.7.0

### Support Checklist

- [x] My user was added to the plugdev group.
- [x] I have rebooted and the problem persists.
- [x] I looked at the [Troubleshooting](https://github.com/openrazer/openrazer/wiki/Troubleshooting) page and it did not solve the problem.

### Symptoms

- [ ] Uninstalling OpenRazer fixes the problem
- [ ] Re-plugging the device fixes the problem
- [ ] It worked before a system/kernel update.
- [ ] Only a problem if the daemon is running.
- [ ] Only a problem from power up or a reboot.
- [x] I'm dual booting with Windows.
- [ ] Intermittent or erratic problem.
- [ ] My device is missing a feature.

### Please describe the issue

So, im using arch and like a month ago i used razergenie and it worked but now it didnt, daemon didnt showing my keyboard but in polychromatic it works well, but still no effects and other, so, i think problem in Dbus but if im trying to fix it my pc just turn off. imma add some screenshots so you can see.

[image]
[image]

z3ntu, 2025-03-31:
What's the output of `sudo dkms status | grep openrazer` ?

Also you might wanna upgrade your openrazer version regardless, 3.7.0 is quite old.

lah7, 2025-03-31:
There might be a mismatch of versions installed. The post says 3.7.0 but Polychromatic says 3.10.1.

Just to be sure, are all 3 packages being used from Arch's repository?

    pacman -Q | grep razer

evgenyez, 2025-03-31:
> There might be a mismatch of versions installed. The post says 3.7.0 but Polychromatic says 3.10.1.
> 
> Just to be sure, are all 3 packages being used from Arch's repository?
> 
> ```
> pacman -Q | grep razer
> ```

┌───(jeniz@torring)-[~]
└─$ pacman -Q | grep razer
openrazer-daemon 3.10.1-1
openrazer-driver-dkms 3.10.1-1
python-openrazer 3.10.1-1

idk? installed from same repository

lah7, 2025-03-31:
The driver might not be built for the `linux-zen` kernel you have installed, assuming `6.13.8-arch1-1` is for the regular `linux` kernel (and `linux-headers`).

Do you have `linux-zen-headers` package installed? DKMS should automatically build the driver after installing that package.

closed 2025-08-30 by github-actions[bot]