openrazer/openrazer #2431: Fn keys don't work when the daemon is running

dybdeskarphet, 2025-04-07:
### Affected Device / PID

Razer Ornata V3 Tenkeyless, 1532:02A3

### Operating System / Version

Arch Linux

### Kernel Version

6.13.8-zen1-1-zen

### OpenRazer Version

3.10.1

### Support Checklist

- [x] My user was added to the plugdev group.
- [x] I have rebooted and the problem persists.
- [x] I looked at the [Troubleshooting](https://github.com/openrazer/openrazer/wiki/Troubleshooting) page and it did not solve the problem.

### Symptoms

- [ ] Uninstalling OpenRazer fixes the problem
- [ ] Re-plugging the device fixes the problem
- [ ] It worked before a system/kernel update.
- [x] Only a problem if the daemon is running.
- [ ] Only a problem from power up or a reboot.
- [ ] I'm dual booting with Windows.
- [ ] Intermittent or erratic problem.
- [ ] My device is missing a feature.

### Please describe the issue

I'm having the same problem as #1174. But more specfically, only the input related Fn keys doesn't work, keyboard related (brightness control, macro recording) Fn keys are working as expected but they are not related to `openrazer` anyway.

[image]

The keys that don't work are the 6 keys on top of the directional keys.

z3ntu, 2025-04-19:
Currently the driver has handling for these keys for Ornata V3 TKL:
```
    { KEY_F9, RAZER_MACRO_KEY },
    { KEY_F10, RAZER_GAME_KEY },
    { KEY_F11, RAZER_BRIGHTNESS_DOWN },
    { KEY_F12, RAZER_BRIGHTNESS_UP },
    { KEY_PAUSE, KEY_SLEEP },
```

Which other ones are needed? E.g. DeathStalker V2 Pro TKL has this key handling:
```
    { KEY_F9, RAZER_MACRO_KEY },
    { KEY_F10, RAZER_GAME_KEY },
    { KEY_F11, RAZER_BRIGHTNESS_DOWN },
    { KEY_F12, RAZER_BRIGHTNESS_UP },
    { KEY_INSERT, KEY_SYSRQ },
    { KEY_HOME, KEY_SCROLLLOCK },
    { KEY_PAGEUP, KEY_PAUSE },
    { KEY_PAGEDOWN, KEY_SLEEP },
```
Does this fit better with your keyboard?

dybdeskarphet, 2025-04-20:
Actually, DeathStalker V2 Pro TKL Fn keys match up with my keyboard perfectly. There are no other keys left. Thank you for your interest in this issue.

dybdeskarphet, 2025-04-20:
Sorry, I checked my keyboard again and there is also the Mute key which is the Fn key of the Delete key.

commit 24519e16 (2025-07-06): razerkbd: Add correct key translation for Ornata V3 Tenkeyless

Fixes #2431
PR #2470 (2025-07-06): razerkbd: Add correct key translation for Ornata V3 Tenkeyless
Fixes #2431
closed 2025-07-06 by z3ntu