openrazer/openrazer #2433: Potential read out of bound in razer_attr_write_matrix_custom_frame functions

DiRaltvein, 2025-04-10:
Hello,

I am fairly certain that I have discovered a read out of bound in the latest version (67cf917) in function razer_attr_write_matrix_custom_frame in potentially all razeraccessory_driver.c, razerkbd_driver.c and razermouse_driver.c files. On an example from razeraccessory_driver.c file. If razer_attr_write_matrix_custom_frame receives an input of buf = {0x00, 0x00, 0x55, 0x00, 0x00} and count of 5 then duringf buffer parsing row_length will overflow
- ((stop_col+1) - start_col) * 3 = ((85 + 1) - 0) * 3 = 86 * 3 = 258 (does not fit into unsigned char)

This overflow will later lead to access of buf out of bound. For example in razer_chroma_standard_matrix_set_custom_frame where row_length will not overflow and will be equal to 258. Later it will be set to 76 on line 458 but even then rgb_data buffer will overflow during memcpy as actual size of buf and of rgb_data buffer is only 5 bytes but code will try to read 76 bytes from it.

[image]

[image]

z3ntu, 2025-04-12:
This issue is fixed with d869abd and 5761051 and released as [v3.10.2](https://github.com/openrazer/openrazer/releases/tag/v3.10.2). The distribution packages we maintain directly are updated, other distros should update soon.

Thanks for reporting!

closed 2025-04-12 by z3ntu

z3ntu, 2025-04-12:
Security advisory is published: https://github.com/openrazer/openrazer/security/advisories/GHSA-835j-6976-46jx Should get a CVE ID in the next days.