openrazer/openrazer #2434: Support for Razer Kraken V4 X

NoDuz2, 2025-04-11:
### Device Name

Razer Kraken V4 X

### Product Number

RZ04-05180100-R3M1

### Model Number

RZ04-0518

### VID:PID

1532:056d

### USB Details

```shell
USB Details
Bus 003 Device 002: ID 1532:056d Razer USA, Ltd Razer Kraken V4 X
Couldn't open device, some information will be missing
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1532 Razer USA, Ltd
  idProduct          0x056d 
  bcdDevice            0.00
  iManufacturer           1 Razer
  iProduct                2 Razer Kraken V4 X
  iSerial                 3 000000000001
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0119
    bNumInterfaces          4
    bConfigurationValue     1
    iConfiguration          4 
    bmAttributes         0x80
      (Bus Powered)
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           0
      bInterfaceClass         1 Audio
      bInterfaceSubClass      1 Control Device
      bInterfaceProtocol      0 
      iInterface              0 
      AudioControl Interface Descriptor:
        bLength                10
        bDescriptorType        36
        bDescriptorSubtype      1 (HEADER)
        bcdADC               1.00
        wTotalLength       0x0071
        bInCollection           2
        baInterfaceNr(0)        1
        baInterfaceNr(1)        2
      AudioControl Interface Descriptor:
        bLength                12
        bDescriptorType        36
        bDescriptorSubtype      2 (INPUT_TERMINAL)
[... 286 lines cut ...]
 00 25 01 09 E9 09 EA 75 01 95 02 81 06 09 00 95
 06 81 01 85 70 15 00 26 FF 00 09 00 75 08 95 04
 91 02 85 71 09 00 75 08 95 04 81 02 C0 05 59 09
 01 A1 01 85 81 09 02 A1 02 09 03 15 00 27 FF FF
 00 00 75 10 95 01 B1 03 09 04 09 05 09 06 09 07
 09 08 15 00 27 FF FF FF 7F 75 20 95 05 B1 03 C0
 85 82 09 20 A1 02 09 21 15 00 27 FF FF 00 00 75
 10 95 01 B1 02 C0 85 83 09 22 A1 02 09 21 15 00
 27 FF FF 00 00 75 10 95 01 B1 02 09 23 09 24 09
 25 09 27 09 26 15 00 27 FF FF FF 7F 75 20 95 05
 B1 02 09 28 09 29 09 2A 09 2B 09 2C 09 2D 15 00
 26 FF 00 75 08 95 06 B1 02 C0 85 84 09 50 A1 02
 09 03 09 55 15 00 25 08 75 08 95 02 B1 02 09 21
 15 00 27 FF FF 00 00 75 10 95 08 B1 02 09 51 09
 52 09 53 09 54 09 51 09 52 09 53 09 54 09 51 09
 52 09 53 09 54 09 51 09 52 09 53 09 54 09 51 09
 52 09 53 09 54 09 51 09 52 09 53 09 54 09 51 09
 52 09 53 09 54 09 51 09 52 09 53 09 54 15 00 26
 FF 00 75 08 95 20 B1 02 C0 85 85 09 60 A1 02 09
 55 15 00 25 08 75 08 95 01 B1 02 09 61 09 62 15
 00 27 FF FF 00 00 75 10 95 02 B1 02 09 51 09 52
 09 53 09 54 15 00 26 FF 00 75 08 95 04 B1 02 C0
 85 86 09 70 A1 02 09 71 15 00 25 01 75 08 95 01
 B1 02 C0 C0
```

z3ntu, 2025-04-19:
Thanks for opening the issue! You can also take a look at https://github.com/openrazer/openrazer/blob/master/DEVELOPMENT.md how you could add support for this device yourself.

deimios, 2026-04-18:
### Device Name
Razer Kraken V4

### Product Number
RZ04-05180100-R3M1

### Model Number
RZ04-0518

### VID:PID
1532:056d

### USB Details
Please note that the Device ID incremented whenever reconnecting, so the packet capture was started with Device 003.

```
Bus 007 Device 006: ID 1532:056d Razer USA, Ltd Razer Kraken V4 X
Negotiated speed: Full Speed (12Mbps)
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 [unknown]
  bDeviceSubClass         0 [unknown]
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1532 Razer USA, Ltd
  idProduct          0x056d Razer Kraken V4 X
  bcdDevice            0.00
  iManufacturer           1 Razer
  iProduct                2 Razer Kraken V4 X
  iSerial                 3 000000000001
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0119
    bNumInterfaces          4
    bConfigurationValue     1
    iConfiguration          4 USB Audio Device
    bmAttributes         0x80
      (Bus Powered)
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           0
      bInterfaceClass         1 Audio
      bInterfaceSubClass      1 Control Device
      bInterfaceProtocol      0 
      iInterface              0 
      AudioControl Interface Descriptor:
        bLength                10
        bDescriptorType        36
        bDescriptorSubtype      1 (HEADER)
        bcdADC               1.00
        wTotalLength       0x0071
        bInCollection           2
        baInterfaceNr(0)        1
        baInterfaceNr(1)        2
      AudioControl Interface Descriptor:
        bLength                12
        bDescriptorType        36
        bDescriptorSubtype      2 (INPUT_TERMINAL)
        bTerminalID             1
        wTerminalType      0x0101 USB Streaming
        bAssocTerminal          0
        bNrChannels             2
[... 305 lines cut ...]
 85 86 09 70 A1 02 09 71 15 00 25 01 75 08 95 01
 B1 02 C0 C0

```

In the attached packet capture I performed the following operations:
Lower the brightness from 100% to 0% in 10% decrements
Increase the brightness from 0% to 100% in 10% increments
Set the Quick Effects from Spectru Cycling to Static
Selected color Red (col 2 row 3 in palette)
Selected color Green (col 5 row 3 in palette)
Selected color Blue (col 7 row 3 in palette)
Selected all colors one by one starting with top left, going down then right. So col 1 row 1, col 1 row 2, ..., col 2 row 1, col 2 row 2 ...
Physically disconnected the headphones
Physically connected the headphones
Physically disconnected the headphones
Physically connected the headphones
Physically disconnected the headphones
Physically connected the headphones

[image]

Couldn't attach the capture, so here's a link to it:

https://mega.nz/file/rdpTiL7T#rkDcprQKe_hy5jHS1-p6qvrYqU46Ixc8YiT2guf-B_U