openrazer/openrazer #2438: Cant use the Primary Profile on my Razer DeathAdder v2

NikolasSaib, 2025-04-13:
### Affected Device / PID

Razer DeathAdder v2 /1532:0084

### Operating System / Version

Fedora 41

### Kernel Version

6.13.9-103.bazzite.fc41.x86_64

### OpenRazer Version

3.10.2

### Support Checklist

- [x] My user was added to the plugdev group.
- [x] I have rebooted and the problem persists.
- [x] I looked at the [Troubleshooting](https://github.com/openrazer/openrazer/wiki/Troubleshooting) page and it did not solve the problem.

### Symptoms

- [ ] Uninstalling OpenRazer fixes the problem
- [ ] Re-plugging the device fixes the problem
- [x] It worked before a system/kernel update.
- [ ] Only a problem if the daemon is running.
- [ ] Only a problem from power up or a reboot.
- [x] I'm dual booting with Windows.
- [ ] Intermittent or erratic problem.
- [x] My device is missing a feature.

### Please describe the issue

Up until  some time ago i could use my primary profile of my mouse(DeathAdder v2) with changed buttons, Hypershift and macros from my Dualboot with Windows and Razer Synapse on my Fedora system just fine. But now i can't use these functions anymore. Then i put macros on my Keyboard (Ornata v3) an there is the same problem with me not having Profiles anymore after switching from Windows back to Fedora. I am sure i was able to use the primary profile of my mouse at least until 2 months ago. Sadly, i cant really say since when this happend for me. 
I tried everything i could, to make the profile run again, but sadly i couldn't find a way. So i'm seeking help here :)

lah7, 2025-04-14:
If it's one of those mice with a profile button on the mouse itself, OpenRazer doesn't support that functionality [<sup>[1]</sup>](https://github.com/openrazer/openrazer/issues/1630#issuecomment-939265674) [<sup>[2]</sup>](https://github.com/openrazer/openrazer/issues/2319#issuecomment-2400876755) unfortunately. Only the primary profile (⚪  white?) will work as expected under OpenRazer/Linux. Things might be buggy if the mouse starts/runs in a different profile.

When it's not working, make sure it's set to the primary profile (on the hardware) and try re-plugging the mouse.

Worst case scenario, you could try resetting the mouse. Possibly via Razer Synapse (Windows) or by holding some buttons (check manual/online).

NikolasSaib, 2025-04-14:
i allready tried resetting the mouse multiple times, and i was using only the primary profile (white led at the profile button) of the mouse. 
but i now found the issue. for some reason my mouse only gets the profile taken over from windows, if i use razer synapse 3 instead of 4. so with razer synapse 3 on my windows suddenly everything works again :)

closed 2025-04-14 by NikolasSaib