openrazer/openrazer #2440: Support for Razer Kraken Kitty Edition 1532:0521

adammilner, 2025-04-16:
### Device Name

Razer Kraken Kitty

### Product Number

RZ04-0298

### Model Number

RZ04-0298

### VID:PID

1532:0521

### USB Details

```shell
Bus 005 Device 013: ID 1532:0521 Razer USA, Ltd Kraken Kitty Edition
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 [unknown]
  bDeviceSubClass         0 [unknown]
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1532 Razer USA, Ltd
  idProduct          0x0521 Kraken Kitty Edition
  bcdDevice            0.70
  iManufacturer           1 Razer
  iProduct                2 Razer Kraken Kitty Edition
  iSerial                 3 00000000
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0123
    bNumInterfaces          4
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0x80
      (Bus Powered)
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           0
      bInterfaceClass         1 Audio
      bInterfaceSubClass      1 Control Device
      bInterfaceProtocol      0 
      iInterface              0 
      AudioControl Interface Descriptor:
        bLength                10
        bDescriptorType        36
        bDescriptorSubtype      1 (HEADER)
        bcdADC               1.00
        wTotalLength       0x0072
        bInCollection           2
        baInterfaceNr(0)        1
        baInterfaceNr(1)        2
      AudioControl Interface Descriptor:
        bLength                12
        bDescriptorType        36
        bDescriptorSubtype      2 (INPUT_TERMINAL)
        bTerminalID            10
        wTerminalType      0x0201 Microphone
[... 269 lines cut ...]
```

### Device Input Paths

```shell
total 0
drwxr-xr-x 2 root root 240 Apr 16 16:29 .
drwxr-xr-x 4 root root 860 Apr 16 16:29 ..
lrwxrwxrwx 1 root root  10 Apr 16 16:28 usb-Cooler_Master_Technology_Inc._MasterKeys_Pro_S_white-event-if02 -> ../event21
lrwxrwxrwx 1 root root  10 Apr 16 16:28 usb-Cooler_Master_Technology_Inc._MasterKeys_Pro_S_white-event-kbd -> ../event20
lrwxrwxrwx 1 root root  10 Apr 16 16:28 usb-Cooler_Master_Technology_Inc._MasterKeys_Pro_S_white-if02-event-kbd -> ../event25
lrwxrwxrwx 1 root root  10 Apr 16 16:28 usb-Generic_USB_Audio-event-if00 -> ../event24
lrwxrwxrwx 1 root root  10 Apr 16 16:28 usb-Logitech_USB_Receiver-if02-event-kbd -> ../event26
lrwxrwxrwx 1 root root  10 Apr 16 16:28 usb-Logitech_USB_Receiver-if02-event-mouse -> ../event22
lrwxrwxrwx 1 root root   9 Apr 16 16:28 usb-Logitech_USB_Receiver-if02-mouse -> ../mouse4
lrwxrwxrwx 1 root root  10 Apr 16 16:29 usb-Razer_Razer_Kraken_Kitty_Chroma_Control-event-if00 -> ../event28
lrwxrwxrwx 1 root root  10 Apr 16 16:29 usb-Razer_Razer_Kraken_Kitty_Edition_00000000-event-if03 -> ../event30
lrwxrwxrwx 1 root root  10 Apr 16 16:28 usb-Yubico_YubiKey_OTP+FIDO+CCID-event-kbd -> ../event10
```

### USB HID Dump

```shell

```

adammilner, 2025-04-16:
This device was previously supported and the pink version still seems to be. Was it removed for some reason?

z3ntu, 2025-04-19:
The device is supported with the USB VID:PID 1532:0f19, is your not showing up with that ID as well? The original support issue was #949

adammilner, 2025-04-22:
After multiple reboots for other reasons it started showing up, not sure what happened.

closed 2025-04-22 by adammilner