openrazer/openrazer #2441: Backlight settings not restored after sleep

andy9a9, 2025-04-17:
### Affected Device / PID

Razer Basilisk V3, 1532:0099

### Operating System / Version

Ubuntu 22.04

### Kernel Version

6.8.0-57-generic

### OpenRazer Version

3.10.2

### Support Checklist

- [x] My user was added to the plugdev group.
- [x] I have rebooted and the problem persists.
- [x] I looked at the [Troubleshooting](https://github.com/openrazer/openrazer/wiki/Troubleshooting) page and it did not solve the problem.

### Symptoms

- [ ] Uninstalling OpenRazer fixes the problem
- [ ] Re-plugging the device fixes the problem
- [ ] It worked before a system/kernel update.
- [ ] Only a problem if the daemon is running.
- [x] Only a problem from power up or a reboot.
- [ ] I'm dual booting with Windows.
- [ ] Intermittent or erratic problem.
- [ ] My device is missing a feature.

### Please describe the issue

Dear community!

I'm using Polychromatic SW to configure my mouse. I enabled the smart wheel and disable backlight by settings the `Brightness` to `0%`. Everything is working until my PC is going to sleep. After wake-up & login back, backlight is enabled, which shouldn't. Once I start the Polychromatic, looks like, that value stayed on `0%`. When I move the slider to some value and back to `0%` backlight is turned off again.

I saw the message in journal after wake-up from sleep.
```
Apr 17 12:04:05 lt-aval openrazer-daemon[5115]: 2025-04-17 12:04:05 | razer.device0                  | INFO     | Resuming RazerBasiliskV3
```

How can I turn-off the backlight permanently, of keep it of after wake-up?

Thanks

andy9a9, 2025-04-19:
No, it's set to `False`. Does it mean that it will load setting from persistent conf?

z3ntu, 2025-04-19:
The help text is
```
# Apply effects saved to disk when daemon starts
```
Then it would apply the effect you have set last to the device whenever the daemon starts e.g. bootup.

Not sure if this changes anything with regular suspend or not.

andy9a9, 2025-04-19:
Will try it later with enabling more logs and via `systemctl suspend` command.

andy9a9, 2025-04-21:
Ok, it didn't help.

- razer.conf
  - ```
    [General]
    # Verbose logging (logs debug messages - lotsa spam)
    verbose_logging = True
    
    
    [Startup]
    # Set the sync effects flag to true so any assignment of effects will work across devices
    sync_effects_enabled = True
    
    # Turn off the devices when the systems screensaver kicks in
    devices_off_on_screensaver = True
    
    # Battery notifier
    battery_notifier = True
    
    # Battery notification frequency [s] (0 to disable)
    battery_notifier_freq = 600
    
    # Apply effects saved to disk when daemon starts
    restore_persistence = True
    ```
- persistence.conf 
  - ```
    [PMxxx]
    dpi_x = 1800
    dpi_y = 1800
    poll_rate = 1000
    backlight_active = True
    backlight_brightness = 0
    backlight_effect = spectrum
    backlight_colors = 0 255 0 0 255 255 0 0 255
    backlight_speed = 1
    backlight_wave_dir = 2
    logo_active = True
    logo_brightness = 75
    logo_effect = spectrum
    logo_colors = 0 255 0 0 255 255 0 0 255
    logo_speed = 1
    logo_wave_dir = 1
    scroll_active = True
    scroll_brightness = 75
    scroll_effect = spectrum
    scroll_colors = 0 255 0 0 255 255 0 0 255
    scroll_speed = 1
    scroll_wave_dir = 2
    ```
Even if the ` backlight_brightness = 0` once the PC was awake from suspend, backlight was restored to some default value.
```
Apr 21 12:18:39 lt-aval openrazer-daemon[12492]: 2025-04-21 12:18:39 | razer                          | INFO     | Initialising Daemon (v3.10.2). Pid: 12492
Apr 21 12:18:39 lt-aval openrazer-daemon[12492]: 2025-04-21 12:18:39 | razer.screensaver              | INFO     | Initialising DBus Screensaver Monitor
...
Apr 21 12:18:39 lt-aval openrazer-daemon[12492]: 2025-04-21 12:18:39 | razer                          | DEBUG    | Loaded device specification: RazerViperV3ProWireless ----------------- (1532:00C1)
Apr 21 12:18:39 lt-aval openrazer-daemon[12492]: 2025-04-21 12:18:39 | razer                          | INFO     | Found device.0: 0003:1532:0099.0008
Apr 21 12:18:39 lt-aval openrazer-daemon[12492]: 2025-04-21 12:18:39 | razer.device0                  | INFO     | Initialising device.0 RazerBasiliskV3
Apr 21 12:18:39 lt-aval openrazer-daemon[12492]: 2025-04-21 12:18:39 | razer.device0                  | DEBUG    | Adding razer.device.misc.getDriverVersion method to DBus
Apr 21 12:18:39 lt-aval openrazer-daemon[12492]: 2025-04-21 12:18:39 | razer.device0                  | DEBUG    | Adding razer.device.misc.suspendDevice method to DBus
Apr 21 12:18:39 lt-aval openrazer-daemon[12492]: 2025-04-21 12:18:39 | razer.device0                  | DEBUG    | Adding razer.device.misc.getDeviceImage method to DBus
Apr 21 12:18:39 lt-aval openrazer-daemon[12492]: 2025-04-21 12:18:39 | razer.device0                  | DEBUG    | Adding razer.device.misc.getDeviceMode method to DBus
Apr 21 12:18:39 lt-aval openrazer-daemon[12492]: 2025-04-21 12:18:39 | razer.device0                  | DEBUG    | Adding razer.device.misc.getRazerUrls method to DBus

Apr 21 12:18:39 lt-aval openrazer-daemon[12492]: 2025-04-21 12:18:39 | razer.device0                  | DEBUG    | Adding razer.device.misc.resumeDevice method to DBus
Apr 21 12:18:39 lt-aval openrazer-daemon[12492]: 2025-04-21 12:18:39 | razer.device0                  | DEBUG    | Adding razer.device.misc.setDeviceMode method to DBus
Apr 21 12:18:39 lt-aval openrazer-daemon[12492]: 2025-04-21 12:18:39 | razer.device0                  | DEBUG    | Adding razer.device.lighting.chroma.restoreLastEffect method to DBus
Apr 21 12:18:39 lt-aval openrazer-daemon[12492]: 2025-04-21 12:18:39 | razer.device0                  | DEBUG    | Adding razer.device.misc.hasDedicatedMacroKeys method to DBus
Apr 21 12:18:39 lt-aval openrazer-daemon[12492]: 2025-04-21 12:18:39 | razer.device0                  | DEBUG    | Adding razer.device.misc.getSerial method to DBus
Apr 21 12:18:39 lt-aval openrazer-daemon[12492]: 2025-04-21 12:18:39 | razer.device0                  | DEBUG    | Adding razer.device.misc.getVidPid method to DBus
[... 245 lines cut ...]
Apr 21 12:21:05 lt-aval openrazer-daemon[12492]: 2025-04-21 12:21:05 | razer.device0                  | DEBUG    | DBus call get_scroll_acceleration
Apr 21 12:21:05 lt-aval openrazer-daemon[12492]: 2025-04-21 12:21:05 | razer.device0                  | DEBUG    | DBus call get_scroll_smart_reel
Apr 21 12:21:05 lt-aval openrazer-daemon[12492]: 2025-04-21 12:21:05 | razer.device0                  | DEBUG    | DBus call get_current_effect
Apr 21 12:21:05 lt-aval openrazer-daemon[12492]: 2025-04-21 12:21:05 | razer.device0                  | DEBUG    | DBus call get_current_wave_dir
Apr 21 12:21:05 lt-aval openrazer-daemon[12492]: 2025-04-21 12:21:05 | razer.device0                  | DEBUG    | DBus call get_current_effect_speed
Apr 21 12:21:05 lt-aval openrazer-daemon[12492]: 2025-04-21 12:21:05 | razer.device0                  | DEBUG    | DBus call get_current_effect_colors
Apr 21 12:21:05 lt-aval openrazer-daemon[12492]: 2025-04-21 12:21:05 | razer.device0                  | DEBUG    | DBus call get_brightness
Apr 21 12:21:05 lt-aval openrazer-daemon[12492]: 2025-04-21 12:21:05 | razer.device0                  | DEBUG    | DBus call get_poll_rate
Apr 21 12:21:05 lt-aval openrazer-daemon[12492]: 2025-04-21 12:21:05 | razer.device0                  | DEBUG    | DBus call get_scroll_mode
[... cut ...]
Apr 21 12:21:05 lt-aval openrazer-daemon[12492]: 2025-04-21 12:21:05 | razer.device0                  | DEBUG    | DBus call get_current_logo_effect_colors
Apr 21 12:21:05 lt-aval openrazer-daemon[12492]: 2025-04-21 12:21:05 | razer.device0                  | DEBUG    | DBus call get_logo_brightness
Apr 21 12:21:05 lt-aval openrazer-daemon[12492]: 2025-04-21 12:21:05 | razer.device0                  | DEBUG    | DBus call get_dpi_xy
Apr 21 12:21:05 lt-aval openrazer-daemon[12492]: 2025-04-21 12:21:05 | razer.device0                  | DEBUG    | DBus call get_dpi_xy
```

lah7, 2025-04-21:
Try turning off this setting:

    devices_off_on_screensaver = False

Then restart the daemon. Sleeping usually triggers the lock screen. Turning that feature off will leave the device's brightness alone.