openrazer/openrazer #2443: Razer Blackwidow V3 goes to a default mode whenever daemon restarts

cgdiaz, 2025-04-18:
### Affected Device / PID

Razer Blackwidow V3 1532:024E

### Operating System / Version

POP OS 22.04

### Kernel Version

6.12.10-76061203-generic

### OpenRazer Version

3.10.2-ubuntu22.04.1

### Support Checklist

- [x] My user was added to the plugdev group.
- [x] I have rebooted and the problem persists.
- [x] I looked at the [Troubleshooting](https://github.com/openrazer/openrazer/wiki/Troubleshooting) page and it did not solve the problem.

### Symptoms

- [ ] Uninstalling OpenRazer fixes the problem
- [ ] Re-plugging the device fixes the problem
- [ ] It worked before a system/kernel update.
- [ ] Only a problem if the daemon is running.
- [ ] Only a problem from power up or a reboot.
- [ ] I'm dual booting with Windows.
- [ ] Intermittent or erratic problem.
- [ ] My device is missing a feature.

### Please describe the issue

Whenever I restart my OS, the settings for my keyboard go back to a default static backlight.  I tried enabling persistence and the dual boot persistence even though I'm not dual booting.

Without a reboot, I enabled logging and restarted the daemon and realized that the problem happens every time the daemon starts.  Even though the logs state the correct setting, the keyboard stays in the default state until I manually select a new effect.

Logs attached.

Thanks!

2025-04-17 17:13:49 | razer                          | INFO     | Initialising Daemon (v3.10.2). Pid: 5356
2025-04-17 17:13:49 | razer.screensaver              | INFO     | Initialising DBus Screensaver Monitor
2025-04-17 17:13:49 | razer                          | DEBUG    | Loaded device specification: RazerAbyssus ---------------------------- (1532:0042)
2025-04-17 17:13:49 | razer                          | DEBUG    | Loaded device specification: RazerAbyssus1800 ------------------------ (1532:0020)
2025-04-17 17:13:49 | razer                          | DEBUG    | Loaded device specification: RazerAbyssus2000 ------------------------ (1532:005E)
2025-04-17 17:13:49 | razer                          | DEBUG    | Loaded device specification: RazerAbyssusEliteDVaEdition ------------- (1532:006A)
2025-04-17 17:13:49 | razer                          | DEBUG    | Loaded device specification: RazerAbyssusEssential ------------------- (1532:006B)
2025-04-17 17:13:49 | razer                          | DEBUG    | Loaded device specification: RazerAbyssusV2 -------------------------- (1532:005B)
2025-04-17 17:13:49 | razer                          | DEBUG    | Loaded device specification: RazerAnansi ----------------------------- (1532:010F)
2025-04-17 17:13:49 | razer                          | DEBUG    | Loaded device specification: RazerAtherisReceiver -------------------- (1532:0062)
2025-04-17 17:13:49 | razer                          | DEBUG    | Loaded device specification: RazerBaseStationChroma ------------------ (1532:0F08)
2025-04-17 17:13:49 | razer                          | DEBUG    | Loaded device specification: RazerBaseStationV2Chroma ---------------- (1532:0F20)
2025-04-17 17:13:49 | razer                          | DEBUG    | Loaded device specification: RazerBasilisk --------------------------- (1532:0064)
2025-04-17 17:13:49 | razer                          | DEBUG    | Loaded device specification: RazerBasiliskEssential ------------------ (1532:0065)
2025-04-17 17:13:49 | razer                          | DEBUG    | Loaded device specification: RazerBasiliskUltimateReceiver ----------- (1532:0088)
2025-04-17 17:13:49 | razer                          | DEBUG    | Loaded device specification: RazerBasiliskUltimateWired -------------- (1532:0086)
2025-04-17 17:13:49 | razer                          | DEBUG    | Loaded device specification: RazerBasiliskV2 ------------------------- (1532:0085)
2025-04-17 17:13:49 | razer                          | DEBUG    | Loaded device specification: RazerBasiliskV3 ------------------------- (1532:0099)
2025-04-17 17:13:49 | razer                          | DEBUG    | Loaded device specification: RazerBasiliskV3Pro35KWired -------------- (1532:00CC)
2025-04-17 17:13:49 | razer                          | DEBUG    | Loaded device specification: RazerBasiliskV3Pro35KWireless ----------- (1532:00CD)
2025-04-17 17:13:49 | razer                          | DEBUG    | Loaded device specification: RazerBasiliskV3ProWired ----------------- (1532:00AA)
2025-04-17 17:13:49 | razer                          | DEBUG    | Loaded device specification: RazerBasiliskV3ProWireless -------------- (1532:00AB)
2025-04-17 17:13:49 | razer                          | DEBUG    | Loaded device specification: RazerBasiliskV3XHyperSpeed -------------- (1532:00B9)
2025-04-17 17:13:49 | razer                          | DEBUG    | Loaded device specification: RazerBasiliskV3_35K --------------------- (1532:00CB)
2025-04-17 17:13:49 | razer                          | DEBUG    | Loaded device specification: RazerBasiliskXHyperSpeed ---------------- (1532:0083)
2025-04-17 17:13:49 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidow2019 --------------------- (1532:0241)
2025-04-17 17:13:49 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowChroma ------------------- (1532:0203)
[... 314 lines cut ...]
2025-04-17 17:13:52 | razer.device0                  | DEBUG    | DBus call get_device_name
2025-04-17 17:13:52 | razer.device0                  | DEBUG    | DBus call get_device_type
2025-04-17 17:13:52 | razer.device0                  | DEBUG    | DBus call get_firmware
2025-04-17 17:13:52 | razer.device0                  | DEBUG    | DBus call version
2025-04-17 17:13:52 | razer.device0                  | DEBUG    | DBus call has_matrix
2025-04-17 17:13:52 | razer.device0                  | DEBUG    | DBus call has_matrix
2025-04-17 17:13:52 | razer.device0                  | DEBUG    | DBus call get_keyboard_layout
2025-04-17 17:13:52 | razer.device0                  | DEBUG    | DBus call get_current_effect
2025-04-17 17:13:52 | razer.device0                  | DEBUG    | DBus call get_current_wave_dir
2025-04-17 17:13:52 | razer.device0                  | DEBUG    | DBus call get_current_effect_speed
2025-04-17 17:13:52 | razer.device0                  | DEBUG    | DBus call get_current_effect_colors
2025-04-17 17:13:52 | razer.device0                  | DEBUG    | DBus call get_brightness
2025-04-17 17:13:52 | razer.device0                  | DEBUG    | DBus call get_game_mode
2025-04-17 17:13:59 | razer                          | DEBUG    | State recently changed, writing to disk
2025-04-17 17:13:59 | razer                          | DEBUG    | Writing persistence config
2025-04-17 17:14:00 | razer.device0                  | DEBUG    | DBus call set_starlight_random
2025-04-17 17:14:00 | razer.device0                  | DEBUG    | Sending observer message: ('effect', RazerBlackWidowV3:IO2209F43301145, 'setStarlightRandom')
2025-04-17 17:14:00 | razer.device0                  | DEBUG    | Set persistence (backlight, effect, starlightRandom)
2025-04-17 17:14:00 | razer.device0                  | DEBUG    | Set persistence (backlight, speed, 1)
2025-04-17 17:14:00 | razer.device0                  | DEBUG    | DBus call get_current_effect
2025-04-17 17:14:00 | razer.device0                  | DEBUG    | DBus call get_current_wave_dir
2025-04-17 17:14:00 | razer.device0                  | DEBUG    | DBus call get_current_effect_speed
2025-04-17 17:14:00 | razer.device0                  | DEBUG    | DBus call get_current_effect_colors
2025-04-17 17:14:00 | razer.device0                  | DEBUG    | DBus call get_brightness
2025-04-17 17:14:00 | razer.device0                  | DEBUG    | DBus call get_game_mode

z3ntu, 2025-04-19:
So you have `restore_persistence` enabled in `~/.config/openrazer/razer.conf`?

cgdiaz, 2025-04-19:
Yes, I currently have persistence and the dual boot persistence enabled.  You can actually see in the log we set the effect twice because of that.

On a whim, I tried a different effect.  Currently, I'm on ripple, random.  And that one does not have the same issue.  Once I saw the effect maintained through a reboot, I thought I inadvertently fixed my issue but when I set it back to starlight random the same thing happened.

Thanks for any support you can provide,

z3ntu, 2025-05-06:
Is it just with the starlight-random effect, or e.g. also starlight-single, starlight-dual and some other effects?

I checked the code a bit and your log, and it seems to work as expected. Is anything appearing in dmesg when you restart the daemon?

cgdiaz, 2025-05-07:
So I tested every effect, and the only one that seems to stay active after a daemon restart is ripple.  However, it will let me do ripple single or ripple random, so I guess I've got that going for me.

I ran a dmesg --follow during the daemon restarts and saw no feedback whatsoever.  I did however capture the dmesg output when I unplug the keyboard and plug it back in, in case that is of any use.

Thanks for checking in,

[image]

z3ntu, 2025-05-10:
Otherwise when setting effects, does it always work the first time you change to a different effect?

I asked you to check the kernel log in case there's some transfer errors between driver and Razer device but seems not, otherwise we would see some messages there.

@lah7 Do you maybe have an idea here?

lah7, 2025-05-11:
If I understand correctly, the keyboard doesn't change effects when the daemon starts despite persistence being applied?

Oh no, there was the Razer Huntsman V2 TKL (#2066) that did something similar, but only under certain conditions (Windows specifically). We never really got to the bottom why, except a crude workaround is to use a software effect (ripple is one).

Just for the records:
* Where is the USB keyboard connected to? (e.g. motherboard, a USB hub, a KVM switch?) 
* What firmware version is the keyboard?

I do wonder if "driver mode" (0x03) has something to do with it. After #2416 is merged, we could see if the initialisation order matters. Currently, driver mode is set **after** persistence, but this upcoming change will set it **before** anything else.  In either case, sounds like a firmware quirk.