openrazer/openrazer #2448: Can't install OpenRazer daemon, SSL certificate(?)

lmsbk, 2025-04-21:
### Affected Device / PID

N/A

### Operating System / Version

Fedora 42

### Kernel Version

6.14.2-300.fc42.x86_64

### OpenRazer Version

3.10.2

### Support Checklist

- [x] My user was added to the plugdev group.
- [x] I have rebooted and the problem persists.
- [x] I looked at the [Troubleshooting](https://github.com/openrazer/openrazer/wiki/Troubleshooting) page and it did not solve the problem.

### Symptoms

- [ ] Uninstalling OpenRazer fixes the problem
- [ ] Re-plugging the device fixes the problem
- [ ] It worked before a system/kernel update.
- [ ] Only a problem if the daemon is running.
- [ ] Only a problem from power up or a reboot.
- [ ] I'm dual booting with Windows.
- [ ] Intermittent or erratic problem.
- [ ] My device is missing a feature.

### Please describe the issue

Attempting to install OpenRazer using instructions listed for Fedora results in curl error 60, relating to ssl or ssh cert/key. Looks to be that SSL certificate is expired.

``` dnf install openrazer-meta ```

```
[12/12] openrazer-daemon-0:3.10.2-1.1.noarch                                                0% |   0.0   B/s |   0.0   B |  00m05s
>>> Curl error (60): SSL peer certificate or SSH remote key was not OK for https://mirror.firstyear.id.au/repositories/hardware:/r
>>> Curl error (60): SSL peer certificate or SSH remote key was not OK for https://mirror.firstyear.id.au/repositories/hardware:/r
>>> Curl error (60): SSL peer certificate or SSH remote key was not OK for https://mirror.firstyear.id.au/repositories/hardware:/r
>>> Curl error (60): SSL peer certificate or SSH remote key was not OK for https://mirror.firstyear.id.au/repositories/hardware:/r
>>> No more mirrors to try - All mirrors were already tried without success                                                       
---------------------------------------------------------------------------------
```

z3ntu, 2025-04-21:
Unfortunately you are not the only person having issues with the openSUSE mirrors (where we build the packages for a bunch of distros), see e.g. #2442 and #2419

Hopefully next time you try to fetch the packages again, you'll be redirected to a usable mirror. Not sure I can suggest much else right now.

lmsbk, 2025-04-21:
@z3ntu 
Cheers for your speedy response and apologies for the duplicate.
Still unavailable as of now but I guess its just a waiting game! Cheers.

z3ntu, 2025-04-22:
No worries, it's good to hear when people are having issues. Maybe you can also try a VPN if you have one available, maybe that'll pick a different mirror then.

Or manually pick a mirror which is listed here?
https://download.opensuse.org/repositories/hardware:/razer/Fedora_42/noarch/openrazer-daemon-3.10.2-1.1.noarch.rpm.mirrorlist