openrazer/openrazer #2449: Razer DeathAdder V3 falls asleep sporadically

Acumane, 2025-04-23:
### Affected Device / PID

Razer DeathAdder V3 Pro (Wireless), 1532:00B7

### Operating System / Version

Fedora 41

### Kernel Version

6.13.11

### OpenRazer Version

3.10.2

### Support Checklist

- [x] My user was added to the plugdev group.
- [x] I have rebooted and the problem persists.
- [x] I looked at the [Troubleshooting](https://github.com/openrazer/openrazer/wiki/Troubleshooting) page and it did not solve the problem.

### Symptoms

- [x] Uninstalling OpenRazer fixes the problem
- [ ] Re-plugging the device fixes the problem
- [ ] It worked before a system/kernel update.
- [x] Only a problem if the daemon is running.
- [ ] Only a problem from power up or a reboot.
- [ ] I'm dual booting with Windows.
- [x] Intermittent or erratic problem.
- [ ] My device is missing a feature.

### Please describe the issue

The mouse will occasionally fall sleep (becomes unresponsive, LED blinks). After several seconds, it wakes up and continues working normally.

I get these 2-3 minute bursts of `dmesg` entries which approximately coincide with the mouse misbehaving. Here's a recent excerpt:

```
[Tue Apr 22 21:27:02 2025] razermouse 0003:1532:00B7.0003: input,hidraw2: USB HID v1.11 Mouse [Razer Razer DeathAdder V3 Pro] on usb-0000:06:00.1-2/input0
[Tue Apr 22 21:27:03 2025] razermouse 0003:1532:00B7.0004: input,hidraw3: USB HID v1.11 Keyboard [Razer Razer DeathAdder V3 Pro] on usb-0000:06:00.1-2/input1
[Tue Apr 22 21:27:03 2025] razermouse 0003:1532:00B7.0005: input,hidraw8: USB HID v1.11 Keyboard [Razer Razer DeathAdder V3 Pro] on usb-0000:06:00.1-2/input2
[Tue Apr 22 21:27:17 2025] razermouse: Command timed out. status: 04 transaction_id.id: 1f remaining_packets: 00 protocol_type: 00 data_size: 16, command_class: 00, command_id.id: 82 Params: 00000000000000000000000000000000 .
[Tue Apr 22 21:27:17 2025] razermouse: Command timed out. status: 04 transaction_id.id: 1f remaining_packets: 00 protocol_type: 00 data_size: 16, command_class: 00, command_id.id: 82 Params: 00000000000000000000000000000000 .
[Tue Apr 22 21:27:17 2025] razermouse: Command timed out. status: 04 transaction_id.id: 1f remaining_packets: 00 protocol_type: 00 data_size: 16, command_class: 00, command_id.id: 82 Params: 00000000000000000000000000000000 .
[Tue Apr 22 21:27:17 2025] razermouse: Command timed out. status: 04 transaction_id.id: 1f remaining_packets: 00 protocol_type: 00 data_size: 16, command_class: 00, command_id.id: 82 Params: 00000000000000000000000000000000 .
[Tue Apr 22 21:27:18 2025] razermouse: Command timed out. status: 04 transaction_id.id: 1f remaining_packets: 00 protocol_type: 00 data_size: 16, command_class: 00, command_id.id: 82 Params: 00000000000000000000000000000000 .
[Tue Apr 22 21:27:18 2025] razermouse: Command timed out. status: 04 transaction_id.id: 1f remaining_packets: 00 protocol_type: 00 data_size: 07, command_class: 04, command_id.id: 05 Params: 0100c800c80000000000000000000000 .
[Tue Apr 22 21:27:18 2025] razermouse: Command timed out. status: 04 transaction_id.id: 1f remaining_packets: 00 protocol_type: 00 data_size: 01, command_class: 00, command_id.id: 05 Params: 02000000000000000000000000000000 .
[Tue Apr 22 21:27:18 2025] razermouse: Command timed out. status: 04 transaction_id.id: 1f remaining_packets: 00 protocol_type: 00 data_size: 02, command_class: 07, command_id.id: 80 Params: 00000000000000000000000000000000 .
[Tue Apr 22 21:27:19 2025] razermouse: Command timed out. status: 04 transaction_id.id: 1f remaining_packets: 00 protocol_type: 00 data_size: 02, command_class: 00, command_id.id: 81 Params: 00000000000000000000000000000000 .
[Tue Apr 22 21:27:19 2025] razermouse: Command timed out. status: 04 transaction_id.id: 1f remaining_packets: 00 protocol_type: 00 data_size: 01, command_class: 00, command_id.id: 05 Params: 02000000000000000000000000000000 .
[Tue Apr 22 21:27:28 2025] razermouse: Command timed out. status: 04 transaction_id.id: 1f remaining_packets: 00 protocol_type: 00 data_size: 02, command_class: 07, command_id.id: 80 Params: 00000000000000000000000000000000 .
[Tue Apr 22 21:27:38 2025] razermouse: Command timed out. status: 04 transaction_id.id: 1f remaining_packets: 00 protocol_type: 00 data_size: 02, command_class: 07, command_id.id: 80 Params: 00000000000000000000000000000000 .
[Tue Apr 22 21:27:48 2025] razermouse: Command timed out. status: 04 transaction_id.id: 1f remaining_packets: 00 protocol_type: 00 data_size: 02, command_class: 07, command_id.id: 80 Params: 00000000000000000000000000000000 .
[Tue Apr 22 21:27:58 2025] razermouse: Command timed out. status: 04 transaction_id.id: 1f remaining_packets: 00 protocol_type: 00 data_size: 02, command_class: 07, command_id.id: 80 Params: 00000000000000000000000000000000 .
[Tue Apr 22 21:28:08 2025] razermouse: Command timed out. status: 04 transaction_id.id: 1f remaining_packets: 00 protocol_type: 00 data_size: 02, command_class: 07, command_id.id: 80 Params: 00000000000000000000000000000000 .
[Tue Apr 22 21:28:19 2025] razermouse: Command timed out. status: 04 transaction_id.id: 1f remaining_packets: 00 protocol_type: 00 data_size: 02, command_class: 07, command_id.id: 80 Params: 00000000000000000000000000000000 .
[Tue Apr 22 21:28:29 2025] razermouse: Command timed out. status: 04 transaction_id.id: 1f remaining_packets: 00 protocol_type: 00 data_size: 02, command_class: 07, command_id.id: 80 Params: 00000000000000000000000000000000 .
[Tue Apr 22 21:28:39 2025] razermouse: Command timed out. status: 04 transaction_id.id: 1f remaining_packets: 00 protocol_type: 00 data_size: 02, command_class: 07, command_id.id: 80 Params: 00000000000000000000000000000000 .
[Tue Apr 22 21:28:49 2025] razermouse: Command timed out. status: 04 transaction_id.id: 1f remaining_packets: 00 protocol_type: 00 data_size: 02, command_class: 07, command_id.id: 80 Params: 00000000000000000000000000000000 .
[Tue Apr 22 21:28:59 2025] razermouse: Command timed out. status: 04 transaction_id.id: 1f remaining_packets: 00 protocol_type: 00 data_size: 02, command_class: 07, command_id.id: 80 Params: 00000000000000000000000000000000 .
[Tue Apr 22 21:29:09 2025] razermouse: Command timed out. status: 04 transaction_id.id: 1f remaining_packets: 00 protocol_type: 00 data_size: 02, command_class: 07, command_id.id: 80 Params: 00000000000000000000000000000000 .
[Tue Apr 22 21:29:19 2025] razermouse: Command timed out. status: 04 transaction_id.id: 1f remaining_packets: 00 protocol_type: 00 data_size: 02, command_class: 07, command_id.id: 80 Params: 00000000000000000000000000000000 .
```

FWIW, it's a resold DeathAdder I paired to a new (genuine) receiver. Might this be causing the driver some confusion?

z3ntu, 2025-04-23:
So this behavior doesn't happen when OpenRazer is not installed?

Acumane, 2025-04-23:
I've had it happen twice in 10 minutes and once in 10 hours... so it's hard to be certain. But yes, it hasn't happened since I've uninstalled, so that's a start.

I'll keep you posted

Acumane, 2025-04-26:
Seems there's no issue when daemon is disabled or OpenRazer's not installed.

z3ntu, 2025-05-10:
That's unusual. We do have some code to set some sleep timeouts but normally this shouldn't cause any random failures like you are describing. 

Could you please try patching the daemon by editing `daemon/openrazer_daemon/hardware/mouse.py` find `class RazerDeathAdderV3ProWired(__RazerDevice):` and remove the `'get_idle_time', 'set_idle_time', 'get_low_battery_threshold', 'set_low_battery_threshold'` parts from the METHODS array? Then restart the daemon and see if it still happens.