openrazer/openrazer #2455: Media Keys work if driver reloaded, but stops after replugging keyboard

HidingCherry, 2025-04-30:
### Affected Device / PID

Razer DeathStalker V2 Pro

### Operating System / Version

ArchLinux

### Kernel Version

6.14.4-zen

### OpenRazer Version

3.10.2

### Support Checklist

- [x] My user was added to the plugdev group.
- [x] I have rebooted and the problem persists.
- [x] I looked at the [Troubleshooting](https://github.com/openrazer/openrazer/wiki/Troubleshooting) page and it did not solve the problem.

### Symptoms

- [ ] Uninstalling OpenRazer fixes the problem
- [ ] Re-plugging the device fixes the problem
- [ ] It worked before a system/kernel update.
- [ ] Only a problem if the daemon is running.
- [ ] Only a problem from power up or a reboot.
- [ ] I'm dual booting with Windows.
- [x] Intermittent or erratic problem.
- [ ] My device is missing a feature.

### Please describe the issue

This might be an awkward issue.

Media keys work, if I rmmod and modprobe razerkbd driver, even if I restart openrazer-daemon.
They stop working if I unplug the keyboard and replug them.

This can be repeated a dozen times, it's always the same.

I think it's a workaround, which is not working whenever the keyboard is not plugged in while the driver loads.

Just for documentary reasons:
I thought media keys are not working, check issue #2234

z3ntu, 2025-05-10:
Can you check `cat /sys/bus/hid/drivers/razerkbd/0*/device_mode` when the media keys and when they don't, if there's a difference?

z3ntu, 2025-07-06:
That should then be caused by the wrong code being used in `razer_raw_event` (and its subfunctions) in razerkbd_driver.c

https://github.com/openrazer/openrazer/blob/master/driver/razerkbd_driver.c#L3806

I can mostly recommend trying a few different things there and see what works. It should work with device_mode = 0x03