openrazer/openrazer #2457: Razer Nommo V2 pro

mak8788, 2025-05-03:
Hi was wondering how I can get this working or when it will be supported in polychromatic? It is an unsupported device Razer Nommo V2 Pro 1532:055A.

z3ntu, 2025-05-06:
Please fill out the issue template! https://github.com/openrazer/openrazer/issues/new?template=00-new-device.yml

taker1988, 2025-05-08:
https://github.com/openrazer/openrazer/issues/2466 

Done.

closed 2025-05-09 by z3ntu

shibby2222, 2025-07-19:
I might be misunderstanding this, but I think this was closed as a duplicate of another thread. However that other thread also closed due to it being for the wrong Nommo Pro (non v2). Does Nommo V2 pro need a new request?

lah7, 2025-07-19:
Looks like [there was a confusion](https://github.com/openrazer/openrazer/issues/2466#issuecomment-2864145940)  but there is an open issue for your `1532:055A` here: #2468