openrazer/openrazer #2465: Blackwidow v4 x Pressing the macro button causes kernel panic

Robert-798, 2025-05-07:
Dear openrazer developer, when I am using Kali Linux (Debian unstable) to install the built-in openrazer-meta3.10.2-dfsg-1 and Polychromatic (0.9.5) software repository, pressing any macro button (M1-M6) will cause kernel panic. The debug information for kernel panic is shown in the figure. The kernel I am using is 6.12.25-amd64 (kali1 2025-04-30). I hope these messages can help you troubleshoot. If you need me to provide other debug messages, you can contact me heyyou3@tuta.io Thank you.

[image]

z3ntu, 2025-05-07:
Should be a duplicate of #2377, we'll have a release with this fixed within the next days!

closed 2025-05-07 by z3ntu

Robert-798, 2025-05-07:
> z3ntu>  left a comment > (openrazer/openrazer#2465) <https://github.com/openrazer/openrazer/issues/2465#issuecomment-2858532563>