openrazer/openrazer #2466: Support for Razer Nommo v2 Pro

taker1988, 2025-05-08:
### Device Name

Razer Nommo v2 Pro

### Product Number

RZ05-02470100-R371

### Model Number

RZ05-0247

### VID:PID

1532:0518

### USB Details

```shell
Bus 001 Device 027: ID 1532:0518 Razer USA, Ltd Nommo Pro
Negotiated speed: Full Speed (12Mbps)
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 [unknown]
  bDeviceSubClass         0 [unknown]
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1532 Razer USA, Ltd
  idProduct          0x0518 Nommo Pro
  bcdDevice            2.00
  iManufacturer           1 Razer
  iProduct                2 Razer Nommo Pro
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x009c
    bNumInterfaces          4
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0x80
      (Bus Powered)
    MaxPower              500mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 [unknown]
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 (null)
          wDescriptorLength      58
          Report Descriptors: 
            ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
[... 377 lines cut ...]
 31 16 00 80 26 FF 7F 75 10 95 02 81 06 C0 06 00
 FF 09 02 15 00 25 01 75 08 95 5A B1 01 C0

001:027:000:DESCRIPTOR         1746723649.711021
 05 0C 09 01 A1 01 85 01 19 00 2A 3C 02 15 00 26
 3C 02 95 01 75 10 81 00 75 08 95 0D 81 01 C0 05
 01 09 00 A1 01 85 05 09 03 15 00 26 FF 00 35 00
 46 FF 00 75 08 95 0F 81 00 C0

001:021:002:DESCRIPTOR         1746723649.716020
 05 0C 09 01 A1 01 06 00 FF 09 02 15 00 25 01 75
 08 95 10 81 00 C0

001:021:001:DESCRIPTOR         1746723649.722020
 05 0C 09 01 A1 01 06 00 FF 09 02 15 00 25 01 75
 08 96 40 01 B1 01 C0

001:021:000:DESCRIPTOR         1746723649.728021
 05 0C 09 01 A1 01 06 00 FF 09 02 15 00 25 01 75
 08 95 5A B1 01 C0
```

### Synapse Screenshots

Only using CachyOS, no Windows.

z3ntu, 2025-05-08:
Are you sure yours are "Nommo V2 Pro"? The USB ID matches the older "Razer Nommo Pro"

Also in #2457 @mak8788 wrote they have 1532:055A

taker1988, 2025-05-08:
Oh, I guess you are completely right. Sorry
I looked up my amazon order.
https://www.amazon.de/dp/B07D8X68TD?ref_=ppx_hzsearch_conn_dt_b_fed_asin_title_8 
It is the "Nommo Pro", not v2. The ones for 435€.

taker1988, 2025-05-09:
Yes, it works now. The problem was with my pc or the device itself. The USB cable was defective and therefore it was not recognized. I replaced it (replacement from Razer) and it was recognized immediately and works. :)

z3ntu, 2025-05-09:
Good to hear!