openrazer/openrazer #2469: Multi device not recognising both devices.

maxwell64, 2025-05-09:
### Affected Device / PID

Razer Blackwidow v3 Pro, Razer Deathadder v2 Pro

### Operating System / Version

Linux Mint 22.1 Cinnamon

### Kernel Version

6.8.0-59-generic

### OpenRazer Version

3.10.3

### Support Checklist

- [x] My user was added to the plugdev group.
- [x] I have rebooted and the problem persists.
- [x] I looked at the [Troubleshooting](https://github.com/openrazer/openrazer/wiki/Troubleshooting) page and it did not solve the problem.

### Symptoms

- [ ] Uninstalling OpenRazer fixes the problem
- [ ] Re-plugging the device fixes the problem
- [ ] It worked before a system/kernel update.
- [ ] Only a problem if the daemon is running.
- [ ] Only a problem from power up or a reboot.
- [ ] I'm dual booting with Windows.
- [ ] Intermittent or erratic problem.
- [ ] My device is missing a feature.

### Please describe the issue

I am running both my mouse and kb through a single dongle, only the device the dongle corresponds to is recognised in both Polychromatic and lsusb (i.e. if I use the mouse dongle the mouse is recognised, vice versa for the kb). The other device does not show up at all in lsusb, but does function, so I can use the mouse even though it is not showing up as connected. Both work fine if I use wired connection. 

lsusb output:

lsusb | grep 1532
Bus 001 Device 014: ID 1532:025c Razer USA, Ltd Razer BlackWidow V3 Pro

Not sure if relevant but dmsg returns the following:

[ 6356.678638] razerkbd: Command timed out. status: 04 transaction_id.id: 9f remaining_packets: 00 protocol_type: 00 data_size: 03, command_class: 0f, command_id.id: 04 Params: 0105bf00000000000000000000000000 .
[ 6356.690638] razerkbd: Command timed out. status: 04 transaction_id.id: ff remaining_packets: 00 protocol_type: 00 data_size: 02, command_class: 00, command_id.id: 04 Params: 03000000000000000000000000000000 .
[ 6356.778648] razerkbd: Command timed out. status: 04 transaction_id.id: 9f remaining_packets: 00 protocol_type: 00 data_size: 02, command_class: 07, command_id.id: 80 Params: 00000000000000000000000000000000 .

z3ntu, 2025-05-10:
Is this about the same as #1892 ?

maxwell64, 2025-05-11:
Yep that would be the same issue, not sure how I didn't see that when I was searching. Thanks for the pointer! Given it's been 18 months I'm guessing it was not an easy one to fix?

z3ntu, 2025-07-06:
@maxwell64 Yes it would require a pretty fundamental change in how OpenRazer is architected. So far nobody has been willing to put in the work, and I certainly don't expect I'll implement this myself anytime soon.

closed 2025-07-06 by z3ntu