openrazer/openrazer #2474: Support for Razer Pro Click V2 Vertical

JBandersson, 2025-05-16:
### Device Name

Razer Pro Click V2 Vertical Edition

### Product Number

RZ01-05250100

### Model Number

RZ01-0525

### VID:PID

1532:00c8

### USB Details

```shell
lsusb -d 1532: -v

Bus 001 Device 011: ID 1532:00c8 Razer USA, Ltd Razer Pro Click V2 Vertical Edition
Couldn't open device, some information will be missing
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 [unknown]
  bDeviceSubClass         0 [unknown]
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1532 Razer USA, Ltd
  idProduct          0x00c8 Razer Pro Click V2 Vertical Edition
  bcdDevice            1.00
  iManufacturer           1 Razer
  iProduct                2 Razer Pro Click V2 Vertical Edition
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0086
    bNumInterfaces          5
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      2 Mouse
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength     100
          Report Descriptors: 
            ** UNAVAILABLE **
      Endpoint Descriptor:
[... 218 lines cut ...]
 00 25 01 75 01 95 06 81 02 75 01 95 02 81 03 06
 00 FF 09 40 75 08 95 02 15 81 25 7F 81 02 05 01
 09 38 15 81 25 7F 75 08 95 01 81 06 09 30 09 31
 16 00 80 26 FF 7F 75 10 95 02 81 06 C0 C0 05 0C
 09 01 A1 01 85 02 19 00 2A 3C 02 15 00 26 3C 02
 95 01 75 10 81 00 75 08 95 0D 81 01 C0 05 01 09
 80 A1 01 85 03 19 81 29 83 15 00 25 01 75 01 95
 03 81 02 95 05 81 01 75 08 95 0E 81 01 C0 05 01
 09 00 A1 01 85 04 09 03 15 00 26 FF 00 35 00 46
 FF 00 75 08 95 0F 81 00 C0 05 01 09 00 A1 01 85
 05 09 03 15 00 26 FF 00 35 00 46 FF 00 75 08 95
 0F 81 00 C0 05 01 09 00 A1 01 85 08 09 03 15 00
 26 FF 00 35 00 46 FF 00 75 08 95 0F 81 00 C0 05
 01 09 00 A1 01 85 09 09 03 15 00 26 FF 00 35 00
 46 FF 00 75 08 95 0F 81 00 C0

001:011:000:DESCRIPTOR         1747409839.172846
 05 01 09 02 A1 01 09 01 A1 00 05 09 19 01 29 05
 15 00 25 01 75 01 95 05 81 02 75 01 95 03 81 03
 75 01 95 08 81 03 05 0C 0A 38 02 15 81 25 7F 75
 08 95 01 81 06 05 01 09 38 15 81 25 7F 75 08 95
 01 81 06 09 30 09 31 16 01 80 26 FF 7F 75 10 95
 02 81 06 C0 06 00 FF 09 02 15 00 25 01 75 08 95
 5A B1 01 C0
```

NuCl34R, 2025-05-22:
+1 for device support request, just got mine yesterday!

bradleygit, 2025-06-09:
PR made for this issue

closed 2025-07-15 by z3ntu

ginolegigot, 2025-08-25:
hello!
i have installed on an arch like (cachy) openrazer-meta-git and polychromatic-git with yay
```bash
openrazer-daemon 3.10.3-1
openrazer-driver-dkms 3.10.3-1
openrazer-meta-git 3.10.3.r19.ged828d23-1
polychromatic-git 0.9.5.r6.g965e76d-1
```
However when i troubleshoot i have this:

[image]
Any idea why ?

bradleygit, 2025-08-25:
@ginolegigot It's not in a release yet, you'll need to use the openrazer-daemon-git as 3.10.3 doesn't have this change in it - this was merged 15th July and 3.10.3 was from May

ginolegigot, 2025-08-26:
@bradleygit i thought openrazer was reflecting the main branch, my bad.  
Anyways, thanks for your answer and your work on this PR!