openrazer/openrazer #2477: Razer BlackWidow V3 Pro (Wireless) Not Showing

WarpedEdge, 2025-05-22:
### Affected Device / PID

Razer BlackWidow V3 Pro (Wireless) 1532:025C

### Operating System / Version

Bazzite Fedora 42

### Kernel Version

6.14.6-102.bazzite.fc42.x86_64

### OpenRazer Version

3.10.3

### Support Checklist

- [x] My user was added to the plugdev group.
- [x] I have rebooted and the problem persists.
- [x] I looked at the [Troubleshooting](https://github.com/openrazer/openrazer/wiki/Troubleshooting) page and it did not solve the problem.

### Symptoms

- [ ] Uninstalling OpenRazer fixes the problem
- [ ] Re-plugging the device fixes the problem
- [ ] It worked before a system/kernel update.
- [ ] Only a problem if the daemon is running.
- [ ] Only a problem from power up or a reboot.
- [ ] I'm dual booting with Windows.
- [ ] Intermittent or erratic problem.
- [x] My device is missing a feature.

### Please describe the issue

I wanted to add a couple things here:

1. Razer Blackwidow V3 Pro when Wired does show up in RazerGenie, however when it is wireless it is not showing up at all (2.4ghz or BT)

2. The newly added Razer Naga V2 Pro Wireless shows up both ways wired and wireless

3. Side question: Is it possible to make the battery into a widget now that its reporting through openrazer 2.4 ghz?

[image]

AJSlye, 2025-06-25:
Are they using the same dongle?
The one that came with the mouse?

WarpedEdge, 2025-06-25:
Yes same dongle.