openrazer/openrazer #2481: Unable to detect right click events

AnjanaYEAH, 2025-06-01:
### Affected Device / PID

Razer Viper Mini SE 1532:009F

### Operating System / Version

Arch Linux

### Kernel Version

6.14.7-arch2-1

### OpenRazer Version

3.10.3

### Support Checklist

- [x] My user was added to the plugdev group.
- [x] I have rebooted and the problem persists.
- [x] I looked at the [Troubleshooting](https://github.com/openrazer/openrazer/wiki/Troubleshooting) page and it did not solve the problem.

### Symptoms

- [ ] Uninstalling OpenRazer fixes the problem
- [ ] Re-plugging the device fixes the problem
- [ ] It worked before a system/kernel update.
- [ ] Only a problem if the daemon is running.
- [ ] Only a problem from power up or a reboot.
- [x] I'm dual booting with Windows.
- [ ] Intermittent or erratic problem.
- [x] My device is missing a feature.

### Please describe the issue

When I look at `libinput debug-events` and press right click I do not see any events registered
I am pretty sure the driver is working correctly
My DeathAddr V3's right click works correct
I noticed there was a difference in razor dmesg, could this be a clue to the problem:

```
[ 1069.544761] usbhid 1-4.1:1.0: Razer Device disconnected
[ 1069.578747] usbhid 1-4.1:1.1: Razer Device disconnected
[ 1069.670714] usbhid 1-4.1:1.2: Razer Device disconnected
[ 1071.102306] usb 1-4.1: Product: Razer DeathAdder V3
[ 1071.102310] usb 1-4.1: Manufacturer: Razer
[ 1071.110711] input: Razer Razer DeathAdder V3 as /devices/pci0000:00/0000:00:14.0/usb1/1-4/1-4.1/1-4.1:1.0/0003:1532:00B2.0013/input/input32
[ 1071.110951] razermouse 0003:1532:00B2.0013: input,hidraw2: USB HID v1.00 Mouse [Razer Razer DeathAdder V3] on usb-0000:00:14.0-4.1/input0
[ 1071.112217] input: Razer Razer DeathAdder V3 as /devices/pci0000:00/0000:00:14.0/usb1/1-4/1-4.1/1-4.1:1.1/0003:1532:00B2.0014/input/input33
[ 1071.162693] razermouse 0003:1532:00B2.0014: input,hidraw4: USB HID v1.00 Keyboard [Razer Razer DeathAdder V3] on usb-0000:00:14.0-4.1/input1
[ 1071.163663] input: Razer Razer DeathAdder V3 as /devices/pci0000:00/0000:00:14.0/usb1/1-4/1-4.1/1-4.1:1.2/0003:1532:00B2.0015/input/input34
[ 1071.237875] razermouse 0003:1532:00B2.0015: input,hidraw8: USB HID v1.11 Keyboard [Razer Razer DeathAdder V3] on usb-0000:00:14.0-4.1/input2
[ 1071.358771] usbhid 1-4.1:1.0: Razer Device disconnected
[ 1071.389685] usbhid 1-4.1:1.1: Razer Device disconnected
[ 1071.455657] usbhid 1-4.1:1.2: Razer Device disconnected
[ 1076.035441] usb 1-4.1: Product: Razer Viper Mini Signature Edition
[ 1076.044418] input: Razer Viper Mini Signature Edition as /devices/pci0000:00/0000:00:14.0/usb1/1-4/1-4.1/1-4.1:1.0/0003:1532:009E.0016/input/input35
[ 1076.044924] razermouse 0003:1532:009E.0016: input,hidraw2: USB HID v1.11 Mouse [Razer Viper Mini Signature Edition] on usb-0000:00:14.0-4.1/input0
[ 1076.046735] input: Razer Viper Mini Signature Edition as /devices/pci0000:00/0000:00:14.0/usb1/1-4/1-4.1/1-4.1:1.1/0003:1532:009E.0017/input/input36
[ 1076.098315] razermouse 0003:1532:009E.0017: input,hidraw4: USB HID v1.11 Keyboard [Razer Viper Mini Signature Edition] on usb-0000:00:14.0-4.1/input1
[ 1076.279642] input: Razer Viper Mini Signature Edition as /devices/pci0000:00/0000:00:14.0/usb1/1-4/1-4.1/1-4.1:1.2/0003:1532:009E.0018/input/input37
[ 1076.360901] razermouse 0003:1532:009E.0018: input,hidraw8: USB HID v1.11 Keyboard [Razer Viper Mini Signature Edition] on usb-0000:00:14.0-4.1/input2
[ 1594.898267] razermouse: Command not supported. status: 05 transaction_id.id: 1f remaining_packets: 00 protocol_type: 00 data_size: 02, command_class: 00, command_id.id: 40 Params: 00080000000000000000000000000000 .
[ 1594.959430] razermouse: Command not supported. status: 05 transaction_id.id: 1f remaining_packets: 00 protocol_type: 00 data_size: 02, command_class: 00, command_id.id: 40 Params: 01080000000000000000000000000000 .
[ 1609.573422] razermouse: Command not supported. status: 05 transaction_id.id: 1f remaining_packets: 00 protocol_type: 00 data_size: 02, command_class: 00, command_id.id: 40 Params: 00100000000000000000000000000000 .
[ 1609.635144] razermouse: Command not supported. status: 05 transaction_id.id: 1f remaining_packets: 00 protocol_type: 00 data_size: 02, command_class: 00, command_id.id: 40 Params: 01100000000000000000000000000000 .
[ 1609.899534] razermouse: Command not supported. status: 05 transaction_id.id: 1f remaining_packets: 00 protocol_type: 00 data_size: 02, command_class: 00, command_id.id: 40 Params: 00400000000000000000000000000000 .
[ 1609.960904] razermouse: Command not supported. status: 05 transaction_id.id: 1f remaining_packets: 00 protocol_type: 00 data_size: 02, command_class: 00, command_id.id: 40 Params: 01400000000000000000000000000000 .
```

Right click works fine on windows
Let me know if there are more debugging logs you would like me to post, I know very little about mouse drivers

z3ntu, 2025-07-06:
Do you see the same issue without OpenRazer installed?

AnjanaYEAH, 2025-12-27:
Yes this is still a problem with or without OpenRazor installed
It looks like this mouse does not map the Right Click event to anything

Output from: `sudo libinput list-devices`

```
Device:                  Razer Razer Viper Mini Signature Edition
Kernel:                  /dev/input/event13
Id:                      usb:1532:009f
Group:                   4
Seat:                    seat0, default
Capabilities:            pointer 
Tap-to-click:            n/a
Tap-and-drag:            n/a
Tap button map:          n/a
Tap drag lock:           n/a
Left-handed:             disabled
Nat.scrolling:           disabled
Middle emulation:        disabled
Calibration:             n/a
Scroll methods:          button
Scroll button:           BTN_MIDDLE
Scroll button lock:      disabled
Click methods:           none
Clickfinger button map:  n/a
Disable-w-typing:        n/a
Disable-w-trackpointing: n/a
Accel profiles:          flat *adaptive custom
Rotation:                0.0
Area rectangle:          n/a

Device:                  Razer Razer Viper Mini Signature Edition
Kernel:                  /dev/input/event19
Id:                      usb:1532:009f
Group:                   4
Seat:                    seat0, default
Capabilities:            keyboard pointer 
Tap-to-click:            n/a
Tap-and-drag:            n/a
Tap button map:          n/a
Tap drag lock:           n/a
Left-handed:             disabled
Nat.scrolling:           disabled
Middle emulation:        disabled
Calibration:             n/a
Scroll methods:          button
Scroll button:           BTN_MIDDLE
Scroll button lock:      disabled
Click methods:           none
Clickfinger button map:  n/a
Disable-w-typing:        n/a
Disable-w-trackpointing: n/a
Accel profiles:          flat *adaptive custom
Rotation:                0.0
Area rectangle:          n/a

Device:                  Razer Razer Viper Mini Signature Edition
Kernel:                  /dev/input/event7
Id:                      usb:1532:009f
Group:                   4
Seat:                    seat0, default
Capabilities:            keyboard 
Tap-to-click:            n/a
Tap-and-drag:            n/a
Tap button map:          n/a
Tap drag lock:           n/a
Left-handed:             n/a
Nat.scrolling:           n/a
Middle emulation:        n/a
Calibration:             n/a
Scroll methods:          none
Scroll button:           n/a
Scroll button lock:      n/a
Click methods:           none
Clickfinger button map:  n/a
Disable-w-typing:        n/a
Disable-w-trackpointing: n/a
Accel profiles:          n/a
Rotation:                0.0
Area rectangle:          n/a
```