openrazer/openrazer #2483: Daemon at user level instead of system level on Arch Linux

B-Teague, 2025-06-01:
Is there a specific reason why the daemon service is user level instead of system level on Arch Linux?

I was hoping my keyboard would light up before the SDDM login screen, because it's hard to see my keyboard in the dark.

When researching, it seems that the daemon is a user level service that runs after successful login, but I still also have to set the settings each time I login in to get the keyboard to light up.

Any plans to make the daemon a system level service since all users would probably be using the same keyboard on a system?

B-Teague, 2025-06-01:
With the help of ChatGPT, I was able to find a working solution.   But please let me know what the "correct" way to configure my keyboard to work on boot is supposed to be.

WARNING!!! I'm "okay" at working with Linux, so there may be bugs to this approach, but just wanted to share with others.
I'm sure there is a more secure/idiomatic solution, but this worked for me.

1. Created a new system level service that runs the razer-boot-setup.sh shell script.
```sudo nano /etc/systemd/system/razer-setup.service```

```
[Unit]
Description=Apply Razer keyboard lighting on boot
After=multi-user.target
Requires=multi-user.target

[Service]
Type=oneshot
ExecStart=/usr/local/bin/razer-boot-setup.sh
RemainAfterExit=true

[Install]
WantedBy=multi-user.target
```

2. Create new udev rules allowing full read/write access targeting only the lighting features of the keyboard device.  (Don't want to give full access to the entire keyboard, just the lights.)

```sudo nano /etc/udev/rules.d/99-razer.rules```

```
# /etc/udev/rules.d/99-razer-lighting.rules

ACTION=="add", SUBSYSTEM=="hid", ATTRS{idVendor}=="1532", ATTRS{driver}=="razerkbd", RUN+="/bin/sh -c 'chmod 0666 /sys/%p/matrix_effect_static || true'"
ACTION=="add", SUBSYSTEM=="hid", ATTRS{idVendor}=="1532", ATTRS{driver}=="razerkbd", RUN+="/bin/sh -c 'chmod 0666 /sys/%p/matrix_effect_wave || true'"
ACTION=="add", SUBSYSTEM=="hid", ATTRS{idVendor}=="1532", ATTRS{driver}=="razerkbd", RUN+="/bin/sh -c 'chmod 0666 /sys/%p/matrix_effect_breath || true'"
ACTION=="add", SUBSYSTEM=="hid", ATTRS{idVendor}=="1532", ATTRS{driver}=="razerkbd", RUN+="/bin/sh -c 'chmod 0666 /sys/%p/matrix_effect_spectrum || true'"
ACTION=="add", SUBSYSTEM=="hid", ATTRS{idVendor}=="1532", ATTRS{driver}=="razerkbd", RUN+="/bin/sh -c 'chmod 0666 /sys/%p/matrix_effect_starlight || true'"
ACTION=="add", SUBSYSTEM=="hid", ATTRS{idVendor}=="1532", ATTRS{driver}=="razerkbd", RUN+="/bin/sh -c 'chmod 0666 /sys/%p/matrix_effect_reactive || true'"
ACTION=="add", SUBSYSTEM=="hid", ATTRS{idVendor}=="1532", ATTRS{driver}=="razerkbd", RUN+="/bin/sh -c 'chmod 0666 /sys/%p/matrix_effect_none || true'"
ACTION=="add", SUBSYSTEM=="hid", ATTRS{idVendor}=="1532", ATTRS{driver}=="razerkbd", RUN+="/bin/sh -c 'chmod 0666 /sys/%p/matrix_effect_custom || true'"
ACTION=="add", SUBSYSTEM=="hid", ATTRS{idVendor}=="1532", ATTRS{driver}=="razerkbd", RUN+="/bin/sh -c 'chmod 0666 /sys/%p/matrix_custom_frame || true'"
```

3. Create shell script to run on boot.  Make this file executable with chmod +x.  The number 1532 below is the Razer vendor id, so the * afterwards includes keyboards, mice, headphones, etc made by Razer.  I believe the number 026B in 0003:1532:026B is specifically for my Huntsman V2 keyboard, so I should probably update below to use 0003:1532:026B* instead.

```sudo nano /usr/local/bin/razer-boot-setup.sh```

```
#!/bin/bash
# Set Razer keyboard lighting to spectrum mode at boot

# Optional: small delay to allow the kernel to fully initialize devices
sleep 2

# Loop through Razer HID devices and set the lighting mode
for dev in /sys/bus/hid/devices/0003:1532:*/matrix_effect_spectrum; do
    if [[ -w "$dev" ]]; then
        echo 1 > "$dev"
        echo "Set spectrum mode for $dev"
        exit 0
    fi
done

echo "No writable matrix_effect_spectrum found" >&2
exit 1
```

lah7, 2025-06-05:
The AI did a good job there. Except you don't need the udev rules, since it's a **_system_** systemd service, which runs as `root`, I believe. `root` will have access to the sysfs driver already.

For completeness to those instructions, mark the script as executable:

    sudo chmod +x /usr/local/bin/razer-boot-setup.sh

and also enable/start the service:

    sudo systemctl enable razer-setup.service
    sudo systemctl start razer-setup.service

I think that is a suitable solution for pre-login configuration, since it uses the driver sysfs directly. The parameters for the other files are documented here:

* https://github.com/openrazer/openrazer/wiki/Using-the-keyboard-driver
* https://github.com/openrazer/openrazer/wiki/Using-the-mouse-driver

> Is there a specific reason why the daemon service is user level instead of system level on Arch Linux?

It's mainly that the daemon serves on a D-Bus session bus - which is a user process. It's not a good idea to run either one (D-Bus or the daemon) as root. It's tricky to make it work as a logged out user, since there isn't a D-Bus session running neither. It's mainly just how OpenRazer is architected.

Thanks for raising this - I'll create a wiki page as applying changes before login would be useful.