openrazer/openrazer #2487: Razer Viper V3 Pro - Settings not persistent

Sazzels, 2025-06-09:
### Affected Device / PID

Razer Viper V3 Pro

### Operating System / Version

EndeavourOS Linux (Arch)

### Kernel Version

6.14.10-arch1-1

### OpenRazer Version

3.10.3

### Support Checklist

- [x] My user was added to the plugdev group.
- [x] I have rebooted and the problem persists.
- [x] I looked at the [Troubleshooting](https://github.com/openrazer/openrazer/wiki/Troubleshooting) page and it did not solve the problem.

### Symptoms

- [ ] Uninstalling OpenRazer fixes the problem
- [ ] Re-plugging the device fixes the problem
- [ ] It worked before a system/kernel update.
- [ ] Only a problem if the daemon is running.
- [x] Only a problem from power up or a reboot.
- [ ] I'm dual booting with Windows.
- [ ] Intermittent or erratic problem.
- [ ] My device is missing a feature.

### Please describe the issue

# Mouse settings reset to defaults on boot, device misidentified until daemon restart
Every time I boot my PC, my Razer Viper V3 Pro is not recognized correctly by OpenRazer. The device shows up with an incorrect identifier and settings revert partially to defaults.

## Expected Behavior

- Mouse should be properly identified on boot
- Custom settings (especially 1000Hz polling rate) should persist across reboots
- Device should show correct serial number immediately

## Actual Behavior
**On boot:**

```bash
Razer Viper V3 Pro (Wireless)  mouse    UNKNOWN_153200C1_0000    main
```
 
- DPI: Correctly set to 800 ✓
- Poll rate: Resets to 500Hz (should be 1000Hz) ✗
- Device identified as UNKNOWN_153200C1_0000

**After manually restarting daemon via Polychromatic dropdown:**
- Device correctly identified as PM2435H32105368
- Poll rate correctly set to 1000Hz
- All settings work as expected

## Workaround
Manually restarting the OpenRazer daemon through Polychromatic fixes the issue temporarily, but this needs to be done after every boot.

## Additional Context

- Mouse is always powered on before starting the PC
- Settings never change manually - they're static
- Only the polling rate resets; DPI persists correctly
- Would prefer a solution that doesn't require Polychromatic if possible

z3ntu, 2025-07-06:
Does the device reliably get a serial number when restarting the daemon multiple times? Also can you check what `dmesg | grep -i razer` says during the first daemon startup? There should be some error from razermouse there.

z3ntu, 2025-07-15:
Oh, on USB descriptor side the serial number that is printed with "usb 7-2: SerialNumber: 000000000000" is completely standard USB, that should not be related to the serial number that openrazer uses.

The "razermouse: Command timed out" can be relevant though, but all the ones you have here are where `razer_chroma_misc_get_battery_level` times out, which is reasonably normal when the device goes to sleep.

So still not sure what is really happening here unfortunately.

Sazzels, 2025-11-13:
This issue is still relevant and the problem persists. After one or two reboots, the polling rate is set to 500 MHz instead of the expected 1000 MHz. Restarting the OpenRazer daemon correctly sets it back to 1000 MHz. The workaround of manually restarting the daemon after each boot is still required.

lah7, 2025-12-26:
The root cause is that the device doesn't respond with its serial number correctly, so the daemon thinks its a new device and applies some default settings. That's why it sometimes works later. Tracked as #2122. 

Alternately, if the mouse (and your other devices) are able to remember its settings across power cycles, you could turn the persistence feature off too.