openrazer/openrazer #2488: Support for Razer Blade 18 2025

canezila, 2025-06-12:
### Device Name

Razer Blade 18 2025

### Product Number

BY2520N66800450

### Model Number

RZ09-0529

### VID:PID

1532:02c7

### USB Details

```shell
Bus 005 Device 006: ID 1532:0f43 Razer USA, Ltd Razer Laptop Cooling Pad
Couldn't open device, some information will be missing
Negotiated speed: Full Speed (12Mbps)
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 [unknown]
  bDeviceSubClass         0 [unknown]
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1532 Razer USA, Ltd
  idProduct          0x0f43 Razer Laptop Cooling Pad
  bcdDevice            2.00
  iManufacturer           1 Razer
  iProduct                2 Razer Laptop Cooling Pad
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x003b
    bNumInterfaces          2
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0x80
      (Bus Powered)
    MaxPower              500mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 [unknown]
      bInterfaceProtocol      0 
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 (null)
          wDescriptorLength      22
          Report Descriptors: 
            ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
[... 257 lines cut ...]
 31 16 00 80 26 FF 7F 75 10 95 02 81 06 C0 06 00
 FF 09 02 15 00 25 01 75 08 95 5A B1 01 C0

005:007:001:DESCRIPTOR         1749760129.764946
 05 01 09 06 A1 01 85 01 05 07 19 E0 29 E7 15 00
 25 01 75 01 95 08 81 02 19 00 2A FF 00 15 00 26
 FF 00 75 08 95 0E 81 00 C0 05 0C 09 01 A1 01 85
 02 19 00 2A 3C 02 15 00 26 3C 02 95 01 75 10 81
 00 75 08 95 0D 81 01 C0 05 01 09 80 A1 01 85 03
 19 81 29 83 15 00 25 01 75 01 95 03 81 02 95 05
 81 01 75 08 95 0E 81 01 C0 05 01 09 00 A1 01 85
 04 09 03 15 00 26 FF 00 35 00 46 FF 00 75 08 95
 0F 81 00 C0 05 01 09 00 A1 01 85 05 09 03 15 00
 26 FF 00 35 00 46 FF 00 75 08 95 0F 81 00 C0

005:007:000:DESCRIPTOR         1749760129.765352
 05 01 09 06 A1 01 05 07 19 E0 29 E7 15 00 25 01
 75 01 95 08 81 02 81 01 19 00 2A FF 00 15 00 26
 FF 00 75 08 95 06 81 00 05 08 19 01 29 03 15 00
 25 01 75 01 95 03 91 02 95 05 91 01 C0
```

### Synapse Screenshots

[image]

canezila, 2025-06-16:
Hi.  I have read 

https://github.com/openrazer/openrazer/blob/master/DEVELOPMENT.md  

several times.  It would be great to get things setup so I can at least try to get the laptop cooler working.  Its a bit out of my league but I am going to try and see what I can do.

z3ntu, 2025-07-06:
Generally all Blade laptops are very similar from the OpenRazer perspective, so you can probably follow a recent Blade laptop addition in the driver and just re-do these changes but for your device.