openrazer/openrazer #2493: Would openrazer be interested in maintaining / incorporating razer-cli?

LoLei, 2025-06-27:
As suggested here:

https://github.com/LoLei/razer-cli/discussions/90#discussioncomment-13595003

`razer-cli` is a CLI using `openrazer` as the main backend to control devices.

Since `razer-cli` does not currently have an active maintainer (me) it was suggested bye @4kbyte that `openrazer` could take over the project.

Any thoughts on this from `openrazer`?

z3ntu, 2025-07-06:
Personally I don't have the extra bandwidth of maintaining more software like that, I already have main OpenRazer and RazerGenie to worry about, plus a bunch of other things. So from my side it's a No unfortunately.

z3ntu, 2025-07-15:
Sure, much success! I'll close this for now :)

closed 2025-07-15 by z3ntu