openrazer/openrazer #2496: Razer Blackshark V2 PRO

MadPandaHD, 2025-06-28:
### Device Name

Razer Blackshark V2 PRO

### Product Number

RZ04-04530100

### Model Number

AC2410D27404637

### VID:PID

1532:0555

### USB Details

```shell
bLength                18
  bDescriptorType         1
  bcdUSB               1.10
  bDeviceClass            0 [unknown]
  bDeviceSubClass         0 [unknown]
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1532 Razer USA, Ltd
  idProduct          0x0555 Razer BlackShark V2 Pro 2.4
  bcdDevice            0.01
  iManufacturer           0 
  iProduct                2 Razer BlackShark V2 Pro 2.4
  iSerial                 3 O001000007
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x00e0
    bNumInterfaces          4
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0x80
      (Bus Powered)
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           0
      bInterfaceClass         1 Audio
      bInterfaceSubClass      1 Control Device
      bInterfaceProtocol      0 
      iInterface              0 
      AudioControl Interface Descriptor:
        bLength                10
        bDescriptorType        36
        bDescriptorSubtype      1 (HEADER)
        bcdADC               1.00
        wTotalLength       0x0046
        bInCollection           2
        baInterfaceNr(0)        1
        baInterfaceNr(1)        2
      AudioControl Interface Descriptor:
        bLength                12
        bDescriptorType        36
        bDescriptorSubtype      2 (INPUT_TERMINAL)
        bTerminalID             1
        wTerminalType      0x0201 Microphone
        bAssocTerminal          0
        bNrChannels             1
[... 320 lines cut ...]
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x83  EP 3 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1
```

### Device Input Paths

```shell
lrwxrwxrwx 1 root root  10 Jun 28 17:33 usb-1532_Razer_BlackShark_V2_Pro_2.4_O001000007-event-if03 -> ../event26
```

### USB HID Dump

```shell
libusb: error [get_usbfs_fd] libusb couldn't open USB device /dev/bus/usb/005/003, errno=13
libusb: error [get_usbfs_fd] libusb requires write access to USB device nodes
Failed to find and open the devices: Access denied (insufficient permissions)
```

closed 2025-07-06 by z3ntu

xela-zone, 2025-07-24:
@z3ntu see what issue you've closed this into - it's from 2 years ago and is more about correctly configring a microphone.

xela-zone, 2025-08-09:
> @xela-zone <https://github.com/xela-zone> What other things are you looking to configure with OpenRazer? But in any case, that issue is the one to discuss things in, there's no point in having multiple issues for the same device.

i'd love to be able to see battery on the headset- i'll mirror this sentiment onto the other thread though.