openrazer/openrazer #2530: Viper Ultimate not detected when used in wired mode

Y4k3d0, 2025-08-04:
### Affected Device / PID

Razer Viper Ultimate, 1532:007B

### Operating System / Version

Arch Linux

### Kernel Version

6.15.8-arch1-2

### OpenRazer Version

3.10.3-1

### Support Checklist

- [x] My user was added to the plugdev group.
- [x] I have rebooted and the problem persists.
- [x] I looked at the [Troubleshooting](https://github.com/openrazer/openrazer/wiki/Troubleshooting) page and it did not solve the problem.

### Symptoms

- [ ] Uninstalling OpenRazer fixes the problem
- [ ] Re-plugging the device fixes the problem
- [x] It worked before a system/kernel update.
- [ ] Only a problem if the daemon is running.
- [ ] Only a problem from power up or a reboot.
- [x] I'm dual booting with Windows.
- [ ] Intermittent or erratic problem.
- [ ] My device is missing a feature.

### Please describe the issue

While using it in wireless mode is a flawless experience, switching to wired causes the daemon to not recognize the device, making it unresponsive.

Here's what I can find in the log with `journalctl -b0 --grep "(polychromatic|razer)"` :
```
août 04 22:54:51 morpheus kernel: usb 5-2: Product: Razer Kraken 7.1 V2
août 04 22:54:51 morpheus kernel: usb 5-2: Manufacturer: Razer
août 04 22:54:51 morpheus kernel: input: Razer Razer Kraken 7.1 V2 Consumer Control as /devices/pci0000:00/0000:00:08.1/0000:16:00.3/usb5/5-2/5-2:1.3/0003:1532:0510.0001/input/input2
août 04 22:54:51 morpheus kernel: input: Razer Razer Kraken 7.1 V2 as /devices/pci0000:00/0000:00:08.1/0000:16:00.3/usb5/5-2/5-2:1.3/0003:1532:0510.0001/input/input3
août 04 22:54:51 morpheus kernel: hid-generic 0003:1532:0510.0001: input,hiddev96,hidraw0: USB HID v1.11 Device [Razer Razer Kraken 7.1 V2] on usb-0000:16:00.3-2/input3
août 04 22:54:51 morpheus kernel: usb 3-2.1: Product: Razer Viper Ultimate Dongle
août 04 22:54:51 morpheus kernel: usb 3-2.1: Manufacturer: Razer
août 04 22:54:51 morpheus kernel: input: Razer Razer Viper Ultimate Dongle as /devices/pci0000:00/0000:00:02.1/0000:03:00.0/0000:04:0c.0/0000:13:00.0/usb3/3-2/3-2.1/3-2.1:1.0/0003:1532:007B.0004/input/input7
août 04 22:54:51 morpheus kernel: hid-generic 0003:1532:007B.0004: input,hidraw3: USB HID v1.11 Mouse [Razer Razer Viper Ultimate Dongle] on usb-0000:13:00.0-2.1/input0
août 04 22:54:51 morpheus kernel: input: Razer Razer Viper Ultimate Dongle Keyboard as /devices/pci0000:00/0000:00:02.1/0000:03:00.0/0000:04:0c.0/0000:13:00.0/usb3/3-2/3-2.1/3-2.1:1.1/0003:1532:007B.0005/input/input8
août 04 22:54:51 morpheus kernel: input: Razer Razer Viper Ultimate Dongle as /devices/pci0000:00/0000:00:02.1/0000:03:00.0/0000:04:0c.0/0000:13:00.0/usb3/3-2/3-2.1/3-2.1:1.1/0003:1532:007B.0005/input/input9
août 04 22:54:51 morpheus kernel: hid-generic 0003:1532:007B.0005: input,hidraw4: USB HID v1.11 Keyboard [Razer Razer Viper Ultimate Dongle] on usb-0000:13:00.0-2.1/input1
août 04 22:54:51 morpheus kernel: input: Razer Razer Viper Ultimate Dongle as /devices/pci0000:00/0000:00:02.1/0000:03:00.0/0000:04:0c.0/0000:13:00.0/usb3/3-2/3-2.1/3-2.1:1.2/0003:1532:007B.0006/input/input10
août 04 22:54:51 morpheus kernel: hid-generic 0003:1532:007B.0006: input,hidraw5: USB HID v1.11 Keyboard [Razer Razer Viper Ultimate Dongle] on usb-0000:13:00.0-2.1/input2
août 04 22:54:51 morpheus kernel: usb 3-2.2: Product: Razer Mouse Dock
août 04 22:54:51 morpheus kernel: usb 3-2.2: Manufacturer: RAZER
août 04 22:54:51 morpheus kernel: input: RAZER Razer Mouse Dock as /devices/pci0000:00/0000:00:02.1/0000:03:00.0/0000:04:0c.0/0000:13:00.0/usb3/3-2/3-2.2/3-2.2:1.0/0003:1532:007E.0007/input/input11
août 04 22:54:51 morpheus kernel: hid-generic 0003:1532:007E.0007: input,hidraw6: USB HID v1.11 Mouse [RAZER Razer Mouse Dock] on usb-0000:13:00.0-2.2/input0
août 04 22:54:51 morpheus kernel: input: RAZER Razer Mouse Dock Keyboard as /devices/pci0000:00/0000:00:02.1/0000:03:00.0/0000:04:0c.0/0000:13:00.0/usb3/3-2/3-2.2/3-2.2:1.1/0003:1532:007E.0008/input/input12
août 04 22:54:51 morpheus kernel: input: RAZER Razer Mouse Dock as /devices/pci0000:00/0000:00:02.1/0000:03:00.0/0000:04:0c.0/0000:13:00.0/usb3/3-2/3-2.2/3-2.2:1.1/0003:1532:007E.0008/input/input13
août 04 22:54:51 morpheus kernel: hid-generic 0003:1532:007E.0008: input,hidraw7: USB HID v1.11 Keyboard [RAZER Razer Mouse Dock] on usb-0000:13:00.0-2.2/input1
août 04 22:54:51 morpheus kernel: input: RAZER Razer Mouse Dock as /devices/pci0000:00/0000:00:02.1/0000:03:00.0/0000:04:0c.0/0000:13:00.0/usb3/3-2/3-2.2/3-2.2:1.2/0003:1532:007E.0009/input/input14
août 04 22:54:51 morpheus kernel: hid-generic 0003:1532:007E.0009: input,hidraw8: USB HID v1.11 Keyboard [RAZER Razer Mouse Dock] on usb-0000:13:00.0-2.2/input2
août 04 22:54:51 morpheus razer_mount[569]: Driver razerkraken
août 04 22:54:51 morpheus razer_mount[573]: Driver razeraccessory
août 04 22:54:51 morpheus razer_mount[574]: Driver razermouse
août 04 22:54:51 morpheus razer_mount[575]: Modprobing razerkraken
août 04 22:54:51 morpheus razer_mount[579]: Modprobing razeraccessory
août 04 22:54:51 morpheus razer_mount[580]: Modprobing razermouse
août 04 22:54:51 morpheus kernel: input: Razer Razer Viper Ultimate Dongle as /devices/pci0000:00/0000:00:02.1/0000:03:00.0/0000:04:0c.0/0000:13:00.0/usb3/3-2/3-2.1/3-2.1:1.0/0003:1532:007B.0004/input/input17
août 04 22:54:51 morpheus kernel: input: RAZER Razer Mouse Dock as /devices/pci0000:00/0000:00:02.1/0000:03:00.0/0000:04:0c.0/0000:13:00.0/usb3/3-2/3-2.2/3-2.2:1.0/0003:1532:007E.0007/input/input18
[... 239 lines cut ...]
août 04 23:41:05 morpheus systemd-logind[785]: Watching system buttons on /dev/input/event23 (RAZER Razer Mouse Dock)
août 04 23:41:05 morpheus systemd-logind[785]: Watching system buttons on /dev/input/event7 (Razer Razer Viper Ultimate Dongle)
août 04 23:41:05 morpheus systemd-logind[785]: Watching system buttons on /dev/input/event20 (RAZER Razer Mouse Dock)
août 04 23:41:07 morpheus openrazer-daemon[1264]: 2025-08-04 23:41:07 | razer                          | INFO     | Found valid device.1: 0003:1532:007E.001A
août 04 23:41:07 morpheus openrazer-daemon[1264]: 2025-08-04 23:41:07 | razer.device1                  | INFO     | Initialising device.1 RazerMouseDock
août 04 23:41:07 morpheus openrazer-daemon[1264]: 2025-08-04 23:41:07 | razer                          | INFO     | Found valid device.2: 0003:1532:007B.0017
août 04 23:41:07 morpheus openrazer-daemon[1264]: 2025-08-04 23:41:07 | razer.device2                  | INFO     | Initialising device.2 RazerViperUltimateWireless
août 04 23:42:42 morpheus systemd[985]: Stopping Daemon to manage razer devices in userspace...
août 04 23:42:42 morpheus openrazer-daemon[1264]: 2025-08-04 23:42:42 | razer                          | INFO     | Stopping daemon on signal 15
août 04 23:42:42 morpheus systemd[985]: Stopped Daemon to manage razer devices in userspace.
août 04 23:42:42 morpheus systemd[985]: Starting Daemon to manage razer devices in userspace...
août 04 23:42:42 morpheus systemd[985]: Started Daemon to manage razer devices in userspace.
août 04 23:42:42 morpheus openrazer-daemon[48184]: 2025-08-04 23:42:42 | razer                          | INFO     | Initialising Daemon (v3.10.3). Pid: 48184
août 04 23:42:42 morpheus openrazer-daemon[48184]: 2025-08-04 23:42:42 | razer.screensaver              | INFO     | Initialising DBus Screensaver Monitor
août 04 23:42:42 morpheus openrazer-daemon[48184]: 2025-08-04 23:42:42 | razer                          | INFO     | Found device.0: 0003:1532:007B.0017
août 04 23:42:42 morpheus openrazer-daemon[48184]: 2025-08-04 23:42:42 | razer.device0                  | INFO     | Initialising device.0 RazerViperUltimateWireless
août 04 23:42:43 morpheus openrazer-daemon[48184]: 2025-08-04 23:42:43 | razer                          | INFO     | Found device.1: 0003:1532:007E.001A
août 04 23:42:43 morpheus openrazer-daemon[48184]: 2025-08-04 23:42:43 | razer.device1                  | INFO     | Initialising device.1 RazerMouseDock
août 04 23:42:43 morpheus openrazer-daemon[48184]: 2025-08-04 23:42:43 | razer                          | INFO     | Found device.2: 0003:1532:0510.0001
août 04 23:42:43 morpheus openrazer-daemon[48184]: 2025-08-04 23:42:43 | razer.device2                  | INFO     | Initialising device.2 RazerKraken71V2
août 04 23:42:43 morpheus openrazer-daemon[48184]: 2025-08-04 23:42:43 | razer                          | INFO     | Serving DBus
août 04 23:49:19 morpheus sudo[49499]:  bastien : TTY=pts/1 ; PWD=/home/bastien ; USER=root ; COMMAND=/usr/bin/pacman -Q openrazer-daemon

```
Available to provide more details if needed, thanks for your time !

z3ntu, 2025-08-08:
When you switch to wired mode, does the USB ID in `lsusb | grep 1532` change? Can you share the output before and after?

Y4k3d0, 2025-08-08:
Yes, I figured as such, what a bummer. Thank you for your time though !