openrazer/openrazer #2547: Razer DeathAdder V3 Pro with DeathRazer HyperPolling Wireless Dongle polling unaffected

Myslawek, 2025-08-28:
### Affected Device / PID

Razer DeathAdder V3 Pro & DeathRazer HyperPolling Wireless Dongle

### Operating System / Version

Fedora 42

### Kernel Version

6.15.10-200.fc42.x86_64

### OpenRazer Version

3.10.3-1.2

### Support Checklist

- [x] My user was added to the plugdev group.
- [x] I have rebooted and the problem persists.
- [x] I looked at the [Troubleshooting](https://github.com/openrazer/openrazer/wiki/Troubleshooting) page and it did not solve the problem.

### Symptoms

- [ ] Uninstalling OpenRazer fixes the problem.
- [ ] Re-plugging the device fixes the problem.
- [ ] It worked before a system/kernel update.
- [ ] Only a problem if the daemon is running.
- [ ] Only a problem from power up or a reboot.
- [ ] Dual boot with Windows (Synapse not installed).
- [ ] Dual boot with Windows (Synapse 3 installed).
- [ ] Dual boot with Windows (Synapse 4 installed).
- [ ] Intermittent or erratic problem.
- [ ] My device is missing a feature.

### Please describe the issue

I was trying to change DPI and polling rate for my mice. DPI works without any issues. However I cannot get polling rate to change to any value different than default 125. 

I tried changing it with following software:
- RazerGenie
- razer-cli
- Polychromatic

None of them worked.

Myslawek, 2025-08-29:
I tried to see some logs with `sudo dmesg -w`. Here is the log after connecting the dongle, with turned **off** mouse:
```
[  637.291114] usb 5-2.2: new high-speed USB device number 7 using xhci_hcd
[  637.379166] usb 5-2.2: New USB device found, idVendor=1532, idProduct=00c3, bcdDevice= 1.01
[  637.379171] usb 5-2.2: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[  637.379173] usb 5-2.2: Product: Razer DeathAdder V3 Pro
[  637.379174] usb 5-2.2: Manufacturer: Razer
[  637.495973] input: Razer Razer DeathAdder V3 Pro as /devices/pci0000:00/0000:00:08.1/0000:11:00.4/usb5/5-2/5-2.2/5-2.2:1.0/0003:1532:00C3.0011/input/input37
[  637.496112] razermouse 0003:1532:00C3.0011: input,hidraw6: USB HID v1.00 Mouse [Razer Razer DeathAdder V3 Pro] on usb-0000:11:00.4-2.2/input0
[  637.497253] input: Razer Razer DeathAdder V3 Pro as /devices/pci0000:00/0000:00:08.1/0000:11:00.4/usb5/5-2/5-2.2/5-2.2:1.1/0003:1532:00C3.0012/input/input38
[  637.569967] razermouse 0003:1532:00C3.0012: input,hidraw7: USB HID v1.00 Keyboard [Razer Razer DeathAdder V3 Pro] on usb-0000:11:00.4-2.2/input1
[  637.571053] input: Razer Razer DeathAdder V3 Pro as /devices/pci0000:00/0000:00:08.1/0000:11:00.4/usb5/5-2/5-2.2/5-2.2:1.2/0003:1532:00C3.0013/input/input39
[  637.651939] razermouse 0003:1532:00C3.0013: input,hidraw9: USB HID v1.00 Keyboard [Razer Razer DeathAdder V3 Pro] on usb-0000:11:00.4-2.2/input2
[  639.707023] razermouse: Command timed out. status: 04 transaction_id.id: 1f remaining_packets: 00 protocol_type: 00 data_size: 16, command_class: 00, command_id.id: 82 Params: 00000000000000000000000000000000 .
[  639.839233] razermouse: Command timed out. status: 04 transaction_id.id: 1f remaining_packets: 00 protocol_type: 00 data_size: 16, command_class: 00, command_id.id: 82 Params: 00000000000000000000000000000000 .
[  639.972528] razermouse: Command timed out. status: 04 transaction_id.id: 1f remaining_packets: 00 protocol_type: 00 data_size: 16, command_class: 00, command_id.id: 82 Params: 00000000000000000000000000000000 .
[  640.105603] razermouse: Command timed out. status: 04 transaction_id.id: 1f remaining_packets: 00 protocol_type: 00 data_size: 16, command_class: 00, command_id.id: 82 Params: 00000000000000000000000000000000 .
[  640.237859] razermouse: Command timed out. status: 04 transaction_id.id: 1f remaining_packets: 00 protocol_type: 00 data_size: 16, command_class: 00, command_id.id: 82 Params: 00000000000000000000000000000000 .
[  640.370858] razermouse: Command timed out. status: 04 transaction_id.id: 1f remaining_packets: 00 protocol_type: 00 data_size: 07, command_class: 04, command_id.id: 05 Params: 01070807080000000000000000000000 .
[  640.402579] razermouse: Command timed out. status: 04 transaction_id.id: 1f remaining_packets: 00 protocol_type: 00 data_size: 01, command_class: 00, command_id.id: 05 Params: 02000000000000000000000000000000 .
[  640.470968] razermouse: Command timed out. status: 04 transaction_id.id: 1f remaining_packets: 00 protocol_type: 00 data_size: 02, command_class: 07, command_id.id: 80 Params: 00000000000000000000000000000000 .
[  650.603141] razermouse: Command timed out. status: 04 transaction_id.id: 1f remaining_packets: 00 protocol_type: 00 data_size: 02, command_class: 07, command_id.id: 80 Params: 00000000000000000000000000000000 .
```

It seems like the dongle itself is recognized as mouse, instead of dongle. Is there a way to work around that?

z3ntu, 2025-08-31:
Could be that we need an increased timeout for this call? Can you give `lsusb | grep 1532` while it's not working? For now it could be worth trying to switch multiple times until it works, sounds like sometimes it does go through and affect the mouse.

z3ntu, 2025-10-21:
Could you please try this diff?
```diff
diff --git a/driver/razermouse_driver.c b/driver/razermouse_driver.c
index c7defb2b..605cbfc4 100644
--- a/driver/razermouse_driver.c
+++ b/driver/razermouse_driver.c
@@ -58,7 +58,6 @@ static int razer_get_report(struct usb_device *usb_dev, struct razer_report *req
     case USB_DEVICE_ID_RAZER_COBRA_PRO_WIRED:
     case USB_DEVICE_ID_RAZER_DEATHADDER_V3_PRO_WIRELESS:
     case USB_DEVICE_ID_RAZER_DEATHADDER_V3_PRO_WIRED:
-    case USB_DEVICE_ID_RAZER_DEATHADDER_V3_PRO_WIRELESS_ALT:
     case USB_DEVICE_ID_RAZER_DEATHADDER_V3_PRO_WIRED_ALT:
     case USB_DEVICE_ID_RAZER_DEATHADDER_V3_HYPERSPEED_WIRED:
     case USB_DEVICE_ID_RAZER_DEATHADDER_V3_HYPERSPEED_WIRELESS:
@@ -96,6 +95,7 @@ static int razer_get_report(struct usb_device *usb_dev, struct razer_report *req
     case USB_DEVICE_ID_RAZER_HYPERPOLLING_WIRELESS_DONGLE:
     case USB_DEVICE_ID_RAZER_VIPER_V3_HYPERSPEED:
     case USB_DEVICE_ID_RAZER_VIPER_V3_PRO_WIRELESS:
+    case USB_DEVICE_ID_RAZER_DEATHADDER_V3_PRO_WIRELESS_ALT:
         return razer_get_usb_response(usb_dev, index, request, index, response, RAZER_VIPER_MOUSE_RECEIVER_WAIT_MIN_US, RAZER_VIPER_MOUSE_RECEIVER_WAIT_MAX_US);
         break;
 

```

closed 2026-06-15 by github-actions[bot]